kizniche/Mycodo #102: Sensor Conditional suggestion

pewi81, 2016-08-28:
Hi!
I have used the PID for a while now with exellent results, great stellar job :-)
However, one thing that I miss is the abillity to have Sensor Conditionals depending on time of the day, for example; 

"If [Temperature] is [Above] the setpoint of [25] **(between time [7:00] and time [21:00])** then....etc etc."

It would be an awesome feature to implement this.
Just a thought!
Best regards
Pete

kizniche, 2016-08-28:
Daily methods for PID use was something I've been working on. It's actually complete but I haven't fully tested it yet.

You say ```miss``` like this was a feature in a previous version of Mycodo, but I can't recall this being a feature of 3.5.

The time options you describe could be tricky to code. Time is always tricky to handle when dealing with only HH:MM:SS in the day, especially at midnight when it goes from the previous day to the next. I'll think about it and come back to this when I've thought about it more and have a good conception of the most efficient way to implement this feature.

pewi81, 2016-08-29:
Guess I used the word 'miss' in the wrong way, what I meant was that it is a feature that would be nice to see in the future, if possible :-)

Nevertheless, many thanks for the  answer!

kizniche, 2018-11-30:
The new conditional system in v7.0 (yet-to-be-released) allows this functionality, due to the new method of evaluating the truth of one statement that multiple conditions can be a part of. For instance, your example above can be accomplished with the following conditional, with {eu36rhw} being replaced before the condition is tested for truth:

```{eu36rhw} > 25 and (7 <= datetime.datetime.now().hour => 21)```

closed 2018-11-30 by kizniche

kizniche, 2018-11-30:
I also just added the ability to import additional modules. Here's an example, below. This is now a very powerful evaluation tool in Mycodo.

[image]