kizniche/Mycodo #104: DHT22 Won't Work After Recent Update

berserktron3k, 2016-09-07:
My DHT22 sensor will no longer work after one of the recent updates. I can validate it works from the command line. I also get the attached errors when trying to activate or deactivate the sensor. Also tried removing and trading the sensor.
[image]
[image]

kizniche, 2016-09-07:
That's not good! Would you be able make your Pi accessible to the internet and also give me an admin login to Mycodo and SSH? You can email or message me through my website contact form for privacy.

If not, we can go through a series of steps from this issue thread.

berserktron3k, 2016-09-07:
OpenVPN work?

kizniche, 2016-09-07:
Sure

berserktron3k, 2016-09-07:
You run Mac or Windows?

kizniche, 2016-09-07:
Windows and Linux

berserktron3k, 2016-09-07:
I'll shoot you the connection info through email in about an hour. Thanks
for your help.

kizniche, 2016-09-07:
We've resolved the issue. pigpiod is required to read the DHT22 sensor. The pigpiod bootup line in cron was missing. Adding the line fixed the issue. I pushed commit 69fbfba to give a more detailed error about pigpiod instead of a generic error that doesn't indicate the issue.

kizniche, 2016-09-07:
Oops. This is the actual commit to add the detailed error: ca844fb

closed 2016-09-07 by kizniche