kizniche/Mycodo #105: Web server acces

wangine, 2016-09-07:
After last update webserver can be accesed only localy, ower internet or router cant work anymore. Everything was fine before update , can you test that , you still have my account , is nothing changed and this night , my Rpi will finish the process. Thanks
[image]

kizniche, 2016-09-07:
I'm not exactly sure how dataplicity routes HTTP traffic to create it's "wormhole" (I think that's what they call it), but one of the last updates was to force HTTP connections into HTTPS, for security. I think this may be the cause. You can test this by either commenting out or removing the [line 84 of mycodo_flask.py](https://github.com/kizniche/Mycodo/blob/master/mycodo/mycodo_flask.py#L84) and then restarting the web server with:

 ```sudo /etc/init.d/apache2 restart```

kizniche, 2016-09-07:
The common method of accessing Mycodo over the internet is to forward port 443 in your router to your Pi's IP address. This will enable HTTP connections to Mycodo. This is how I access Mycodo outside my local network.

wangine, 2016-09-07:
Ok , i will try other posibility, for now its working , i just comented the line described before
Thanks

kizniche, 2016-09-07:
I will add an option to force HTTPS from the admin panel. It will be by default to force HTTPS but you can disable it. I'll try to have an update for that tonight.

kizniche, 2016-09-07:
The latest update should give you the option in the Configuration menu to disable forced HTTPS.

closed 2016-09-07 by kizniche

wangine, 2016-09-08:
Yes , was and working . Thanks

TeaPot169, 2021-09-13:
So I am commenting on this issue as your comments are related to my issue.
Cant access mycodo through the web remotely, I type in my public IP:443.
I have my PI connected through 2 routers:
PI connected to router 1, router 2 connected to router 2, router 2 connected to internet.
I can connect on the local network, and also get email notifications on my phone that have been programmed.
I've set up port forwarding in 2 different ways.
router 1 has port 443 forwarded to my PI's IP and router 2 is in bridge mode. 
Or
router 1 has port 443 forwarded to my PI's IP and router 2 has port 443 forwarded to router 1's external IP.
With both these setups I cant seem to access mycodo remotely....
Have setting on mycodo to force HTTPS activated.
Am I doing something wrong here or could it perhaps be some simple other issue?