kizniche/Mycodo #107: Simple edge switch help

itzik16, 2016-09-16:
Hey Kyle, So I finally installed the 2 water level sensor switches which are basically a reel switches which I want to monitor and log.
I have connected them to the Pi and set them up on Mycodo, I tried it with all the types of modes that is on the the software("Rising", "Falling", "Both") and it doesn't seem to work properly, sometimes when the value is on 0 when I open the switch It turns to 1 and doesn't turn to 0 after I close it, and goes the opposite way when It set to 1, So if it gets a reading I guess that I have probably have set it up wrong on Mycodo,I don't have access to the system right now but I will post you some pictures as soon as I'll get there. Im not sure what am I doing wrong, Might be software problem or an hardware problem though I have tried it with few different sensors from the same type and none seem the differ from the other.

kizniche, 2016-09-16:
I'll comment, when I get some free time today, with a wiring diagram.

itzik16, 2016-09-16:
I have wired 10k ground that connects to the gpio and to a 1k resistor and
to the reel switch wire. And the other wire goes to 3.3v

kizniche, 2016-09-16:
This is how I have my reed switch wired. It pulls the signal low (0 volts) until the circuit is completed, which it is them high (3.3 volts).

[image]

itzik16, 2016-09-16:
Hey kyle. This is exactly how I have connected it

kizniche, 2016-09-16:
Can you verify the signal of the input pin through Mycodo's "tools/system info" under "gpio readall" when it's closed and open to determine if it's actually opening and closing? You can also execute "gpio readall" in the terminal.

Also, this wiring should also work (untested by myself):
[image]

itzik16, 2016-09-16:
Sure mate, the system is currently out of reach for me I will get to be
next to it tommorow then I will contact you again

kizniche, 2016-09-16:
Also, can you link me to the exact reed device you're using?

itzik16, 2016-09-16:
http://sell.lulusoso.com/selling-leads/192326/Liquid-Water-Level-Sensor-Horizontal-Float-Switch.html

kizniche, 2016-09-16:
I see this in the description:

6.Max Switching Voltage: 250V DC
7.Minimum Voltage: 100V DC
8.Maximum Switching Current: 0.5 A

I'm not sure if this would be incompatible with the Pi's 3,3-volt system, but it has similar specifications to the flat sensors I use: http://www.ebay.com/itm/Side-Mount-Horizontal-Water-Level-Sensor-Sensing-Liquid-Float-Switch-Tank-Pool-/291770285855?hash=item43eedd3f1f:g:s84AAOSw9etXQR0n

itzik16, 2016-09-16:
> Sensor-Sensing-Liquid-Float-Switch-Tank-Pool-/291770285855?hash=
> item43eedd3f1f:g:s84AAOSw9etXQR0n

I have few types of those switches I just cant find them online since the
place i bought them from is out of stock and they are marketed for aurduino
use

kizniche, 2016-09-16:
I see them all over ebay from vendors around the world. You should be able to find a supplier close to you from there. If you hook one of those up and it doesn't work, there's something definitely wrong that would require some more investigation.

zsole2, 2016-09-27:
I tried to setup a float sensor and run into problems with edge detection now.
My wiring is confirmed to work separately on a breadboard.
What I could find out until now that the problem is probably with the behaviour RPi.GPIIO library. After playing around with various settings and failed, I checked the daemon log. Several errors were present, most notably "RuntimeError: Conflicting edge detection already enabled for this GPIO channel". After searching, it turned out the it is a problem with other people, and I found a suggestion to solve it at https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=134229. However, I couldn't work it out, the errors returned. On my breadboard setup, I even switched the reed relay pin to a relay port and when 
turning on, it lit for about a second, then turned off!!! 
So my conclusion is that the pin settings need to looked into, and how the pin config is stored by RPi.GPIO.

As a workaround, I was able to set up my float sensor after several tries, on different pin and making sure that everything was set correctly on the first try. I don't know what to do after I run out of available pins after awhile....

kizniche, 2016-09-27:
I'll try to do some testing this afternoon. Also, I couldn't figure out what you meant in the second part of your comments, about how to perform a workaround.

zsole2, 2016-09-27:
Testing sounds great.
The workaround is to set up the edge sensor at once and making sure that every setting is correct. If you do it this way, it works. If you change any RPi.GPIO setting later (like the direction of the change), you get the error, and it may not work correctly anymore on the given GPIO pin, even after a reboot. This was even the case when I changed from input to output, somehow the Pi remembered the previous setting and turned off the LED that was connected to simulate a relay output. 
If you set up another edge sensor on another pin, ti then works again.

kizniche, 2016-09-27:
I added GPIO cleanup code in the latest commit. See if that fixes the edge detection issue. I'm still not sure what's going on with GPIO.

kizniche, 2016-09-27:
I was getting the same errors you were. I guess I never touched my Edge Sensor once I set it up so I never saw the error. Now I'm not getting any errors and my edge detection is working.

zsole2, 2016-09-28:
Thank you, I also can confirm that the edge detection now works as expected. 
On the other hand, edge conditionals don't. A far as I see, the code only checks for 'above' and 'below', instead of including  'rising' and 'falling'. Can you look into this, too?
And as a side note, I think that the direction of change cannot be 'both'. It is either falling or rising, if the debouncing and refractory period is set to meaningful values. Or am I misunderstanding edge detection?

And while I am at it, I'd like to ask you another question. My setup now includes a container for the humidifier. I planning to add the float sensor to this container, to be topped up if the level gets too low, from  backup container with an aquarium pump. This should work without any problem. However, if I got another float sensor in the backup container to check whether the level is low there, can I set up Mycodo to override the relay for the pump to avoid running it dry? If I set the first sensor sensing 'both' direction, to turn on and off the pump, that would fill up the first container. How would I set the second sensor? If I set it to turn of the the pump when the backup level is low, I believe that the first sensor would still turn on the pump if it sensing a low level. I'll need to experiment with this setup, after you will have time to fix the edge conditionals.

kizniche, 2016-09-29:
> ...edge conditionals... the code only checks for 'above' and 'below', instead of including 'rising' and 'falling'.

That's strange, because my Edge conditional hs the options "rising", "Falling", and "Both."

> And as a side note, I think that the direction of change cannot be 'both'. It is either falling or rising, if the debouncing and refractory period is set to meaningful values. Or am I misunderstanding edge detection?

I think you're misunderstanding. This is just what will be detected. A rising edge is when the voltage is going from LOW to HIGH, and a falling edge is when the voltage is going from HIGH to LOW. You are just specifying what you want to detect. If you specify "Rising", only rising edges will be detected (logged and activate conditionals). Same with Falling. If you select "Both", then you will detect rising and falling edges. For instance, if you have a float sensor that completes the circuit when the water level is high (if the float is activating the reed switch), then when the water level goes low it will drop the float from the reed switch, causing a HIGH to LOW (falling edge) which can be configured to activate a water pump ON. Then when it detects a Rising edge (LOW to HIGH) when the float activates the reed switch, it could turn the water pump off. This could be a flawed system if the switch doesn't catch the rising edge for whatever reason and keep your water pump on, which could be really bad. What I would do is set a water pump to turn on for several seconds to raise the water level (enough to complete the reed circuit again), to ensure the pump doesn't stay on too long. I should integrate an Edge conditional to be able to check the state of the pin every X seconds, so if can check the water level intermittently.

> if I got another float sensor in the backup container to check whether the level is low there, can I set up Mycodo to override the relay for the pump to avoid running it dry?

See my comment above. I'll get edge conditionals working so you can test different setups.

kizniche, 2016-09-29:
I see I misunderstood your comment that "the code only checks for 'above' and 'below', instead of including 'rising' and 'falling'". There actually was code to call the conditional actions, but it was just horribly implemented. I implemented edge conditionals right after getting edge detection working but didn't test it. I guess I forgot I didn't test it and it never got fixed. I should have disabled it from the start with a note that it wasn't tested.

zsole2, 2016-09-29:
Thanks for checking.
Reading back what I wrote, it is indeed not too clear. 
I agree that the edge detection on a pin can be set for all three option, and detect either or both rising and falling edges. On the other hand, when you set a conditional, I think you cannot set 'both'. The result what you get is either a falling or rising edge from a switch. This is exactly how it is implemented in your code, with the 'rising_or_falling' variable.

zsole2, 2016-09-29:
I confirm that the edge conditional is working.

kizniche, 2016-09-29:
> On the other hand, when you set a conditional, I think you cannot set 'both'. The result what you get is either a falling or rising edge from a switch. This is exactly how it is implemented in your code, with the 'rising_or_falling' variable. 

The conditional should be able to trigger the actions on rising, falling, or both. If you have a conditional that specifies both, you will need the edge detection to also be set to both.

kizniche, 2016-09-30:
@zsole2 Is edge detection now working on the pins that got buggered previously or are those still not working?

zsole2, 2016-09-30:
I'll try to do more tests over the weekend, but after several modifications of the settings, everything looks normal now.

zsole2, 2016-10-02:
I did some testing and the basic fill-up conditional is working. What I realized with a few quick test on the level-sensor in the backup container is that edge detection is not suitable for this. You have already mentioned it earlier:

> ... This could be a flawed system if the switch doesn't catch the rising edge for whatever reason and keep your water pump on, which could be really bad. What I would do is set a water pump to turn [...]

I think the situation could be  better handled by modifying the relay conditionals. I mean if relay conditionals are rewritten similarly to the sensor conditionals (i.e. adding the conditional within the relay's data set), that could handle several additional things. E.g. if a condition is met (either the state of the relay, or the state of another relay/sensor), then the relay could be triggered. This could be useful in many situations as a safety valve: like my scenario with water levels, or  not turning on a heater if the temperature too high, etc.

I know that there are other priorities, especially with your PhD. Also I noticed the improvements suggested by @etiology, which could look like a major boost that we got earlier from this current version. So just putting this down here for future reference.

kizniche, 2016-10-02:
I just poured a damn good cup of coffee. Let's see where this takes me this morning :)

kizniche, 2016-10-02:
Update incoming to add new functionality to the edge detection conditionals!

zsole2, 2016-10-03:
Thanks, this looks great!
I'll have to experiment it. What do you think will happen in the following scenario?
I set the first EDGE sensor conditional to turn ON the pump-relay for 5 secs if the GPIO is LOW, then check it every 60 secs. (I set up currently the float sensors to have the GPIO state and the water level to run paralel: LOW signal = low water level, and vica versa.) This would fill up my first container step-wise. 
Then in the second container, the second EDGE sensor would check for GPIO LOW, to see if the level reaches the low limit. The same pump-relay should be set then now to OFF, and stay OFF until the second sensor goes HIGH. I suspect that this second stage will not work, at least I don't now yet how to tell the pump-relay that the second container was refilled. 
I  will test it and see how it behaves in the real world. It will be splashy though...

kizniche, 2016-10-03:
I'm having a hard time visualizing your setup how you describe it, specifically discerning the relationship between the containers and pumps.

kizniche, 2016-10-03:
If I'm thinking about this correctly, you will have issues. You could have a script be executed in response to the humidifier water level being low, and if the secondary bucket is not also low, then turn on the pump to fill the humidifier.

zsole2, 2016-10-03:
Sorry for the confusion, it is trivial for me. I have two container, the first one with the humidifier, sealed except for the outlet, making difficult to fill it. Here comes the second container for topping up the first. So you've got the relationship correctly. The script might be a solution, I'll think about it.

kizniche, 2016-10-03:
You can connect your float sensors so that Mycodo will only receive the signal to fill the humidifier only if the humidifier water is low and the refill bucket is high. You can then wire just the refill bucket float sensor to a GPIO to detect when it's low and to fill it. You can use all your existing hardware and use Mycodo for it all.

kizniche, 2016-10-03:
To clarify my previous comment, this is how you could wire it:

Humidifier: Circuit is complete when the water level is low.
Fill Bucket: Circuit is complete when the water level is high.

This will produce a high signal only if the humidifier is low and fill bucket is high.

You should also be able to detect the level of just the fill bucket by using a pull-down resistor to register a Low signal until the bucket is full, and have a refill pump activate while it remains low.

zsole2, 2016-10-04:
Thanks, I'll do tests soon.

kizniche, 2016-10-04:
I made a circuit but haven't tested it. The left reed switch is for the humidifier, the right is the fill bucket. The left GPIO will only turn high when both left and right reed switches close. The right GPIO only needs the right reed switch to close to become high. Are your float sensors able to be revered by turning them upside down?

[image]

zsole2, 2016-10-04:
Thanks for your help. I'll test this and come back with the results. Yes, my  sensors are upside down.

zsole2, 2016-10-05:
A quick status update: the circuit seems to work, when checking the gpio readall output. However, the sensor conditional based on the gpio status does not seem to work, at least in my breadboard test. I'll try to play a little more tomorrow, going back to edge detection which I know it worked before in my setup.

kizniche, 2016-10-05:
Hmm. I just reviewed the code and can't see anything preventing it from working properly. My tests also work find for state conditional testing with the edge detection sensor. Perhaps you can run the daemon in debug mode to get more information about why it might not be working.

```
sudo service mycodo stop
sudo ~/Mycodo/mycodo/mycodo_daemon.py -d
```

zsole2, 2016-10-06:
It may be just the breadboard with a bad connection, I'll try to go further with this tonight.

zsole2, 2016-10-06:
I tested the circuit, and it works. It was either a bad connection, or I was not patient enough to wait for the default 60 secs for checking again the GPIO state. Changing it to 10 secs helped :smile: 
Thanks for your great help!

zsole2, 2016-10-06:
So after hijacking this issue from @itzik16, lets hope that his problem is advanced, too!

kizniche, 2016-10-06:
Nice! Glad that hardware solution worked. I started thinking about how to modify the code to make it happen, then stopped and realized it could be done more easily with the hardware you already have.

closed 2017-01-15 by kizniche

dylandn, 2020-11-05:
I have a question regarding adding the ability to use the rPi's internal pull down resistor instead of the external resistor hwne using the GPIO input, would this be the right place? 

As described here:
https://raspi.tv/2013/rpi-gpio-basics-6-using-inputs-and-outputs-together-with-rpi-gpio-pull-ups-and-pull-downs

kizniche, 2020-11-07:
I don't have plans to integrate any use of the internal resistors, as this is dangerous to use. These resistors aren't active until after bootup, which can cause significant issues. Therefore, I recommend using external resistors.

dylandn, 2020-11-07:
Very good point, thank you.