kizniche/Mycodo #110: Internal server error after upgrading to commit  No. d291c1c

zsole2, 2016-09-23:
Just to let you know...
I've upgraded on the go, later I'll investigate further

kizniche, 2016-09-23:
Does this happen for every page?

zsole2, 2016-09-23:
When I checked in terminal, it showed that the mycodo daemon failed to start, missing numpy.

kizniche, 2016-09-23:
numpy should have been installed with the update. numpy sometimes takes 10 - 30 minutes to install (depending on what Pi version you're running), so you have to give the update time to complete. If you interrupt it (I sometimes get impatient and wonder if the update failed, but it just is taking a long time), then the daemon will not be able to start.

You can induce numpy to install again by issuing the following command:

```sudo pip install --upgrade -r ~/Mycodo/requirements.txt```

kizniche, 2016-09-23:
To be clear, I believe only the initial install of numpy takes so long. Future updates (if numpy has an update) should be quicker. numpy was added to do the complex math required for the new Bezier curve method.

zsole2, 2016-09-23:
That explains some things... what was strange that the status indicator turned green but still showed the old version number (4.0.19)... and your fix for the sensor conditional still not worked then. I'll try again to make sure that everything is finishes and report back.

kizniche, 2016-09-23:
I updated two of my systems and they appear to have upgraded just fine. My Pi 1 took ***much*** longer than my Pi 2 and 3. The status indicator turning green and remaining 4.0.19 is strange. Did you load a new page (or refresh) to render the new HTML with the new version number?

zsole2, 2016-09-23:
Yes, I tried both. Probably something is screwed up seriously, because restore is also not permitted, saying 'Restore functionality is not currently enabled.' I try to reinstall Mycodo.

kizniche, 2016-09-23:
I haven't implemented the restore functionality. I can give you a few more commands to try.

zsole2, 2016-09-23:
I see.
Since there are numerous issues here, I think I better off to reinstall Mycodo.

kizniche, 2016-09-23:
Try this:

```
sudo apt-get update && sudo apt-get upgrade &&
sudo ~/Mycodo/mycodo/scripts/update_mycodo.sh initialize &&
cd ~/Mycodo &&
sudo git fetch &&
sudo git reset --hard origin/master &&
sudo ~/Mycodo/mycodo/scripts/update_post.sh

```

kizniche, 2016-09-23:
Actually... Let me amend that

kizniche, 2016-09-23:
```
~/Mycodo/mycodo/mycodo_client.py -t &&
sudo service mycodo stop &&
sudo apt-get update && sudo apt-get upgrade &&
sudo ~/Mycodo/mycodo/scripts/update_mycodo.sh initialize &&
cd ~/Mycodo &&
sudo git fetch &&
sudo git reset --hard origin/master &&
sudo ~/Mycodo/mycodo/scripts/update_post.sh
```

zsole2, 2016-09-23:
thanks, I'll try..

zsole2, 2016-09-24:
OK, I tried but something went amiss...
Anyway, reinstalled Mycodo, and everything seems OK now, the sensor conditional fix also  works...

Having been enough time waiting for numpy being installed (OK, it is a lie, I went to bed, and continued next morning), I remembered that once upon a time working on version 4 you were planning/wishing  to implement a system where the modules would only load as necessary. With that in mind, I tried the following code, and it works (BMP was randomly picked for this exercise):

`
...
import time
import pip

try:
        import Adafruit_BMP.BMP085 as BMP085
except:
        # TODO: ask user to continue, etc...
        print("installing required package...\n\n")
        pip.main(['install', 'Adafruit_BMP'])
        # TODO: add new package to requirements.txt
        import Adafruit_BMP.BMP085 as BMP085

import RPi.GPIO as GPIO

class BMP(object):
...
`
Of course, this kind of modifications would require quite a bit of refactoring to ensure that all relevant import lines are moved from the front of the files to the places where they occur in the code.

pewi81, 2016-09-25:
So, I got the same problem when I updated to latest version. Cant reach the Live page, only get "Internal Server Error. The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

However, I can reach the other pages by simply add /graph, /relay etc.
Dont know what went wrong but Im able to control all the relays and stuff, only the live page that doesnt work!

kizniche, 2016-09-25:
Well, these aren't really the same problem because zsole2's issue was not reproducible and a reinstall fixed it. He wasn't able to start the daemon and the webserver worked. Your issue sounds like one page of the web server doesn't work with no mention of the daemon.

All the Live pages on the Pis I've tested are working fine, so I'll need a little more information to diagnose the issue.

Could you go to the live page to induce the error, then immediately after, go to Tools->Mycodo Logs and see if there are any errors in the HTTP Log?

pewi81, 2016-09-25:
I had to restart my Raspberry in order to get the Daemon to work after the update.
I get a page full of errors when I logged the induced error so here we go:

 [Sun Sep 25 14:10:02.600344 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     reraise(exc_type, exc_value, tb)
[Sun Sep 25 14:10:02.600361 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1639, in full_dispatch_request
[Sun Sep 25 14:10:02.600379 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     rv = self.dispatch_request()
[Sun Sep 25 14:10:02.600396 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1625, in dispatch_request
[Sun Sep 25 14:10:02.600413 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     return self.view_functions[rule.endpoint](**req.view_args)
[Sun Sep 25 14:10:02.600431 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/var/www/mycodo/mycodo/mycodo_flask.py", line 230, in page
[Sun Sep 25 14:10:02.600448 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     Method.method_order == 0).all()
[Sun Sep 25 14:10:02.600465 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2613, in all
[Sun Sep 25 14:10:02.600482 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     return list(self)
[Sun Sep 25 14:10:02.600498 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2761, in __iter__
[Sun Sep 25 14:10:02.600517 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     return self._execute_and_instances(context)
[Sun Sep 25 14:10:02.600534 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/orm/query.py", line 2776, in _execute_and_instances
[Sun Sep 25 14:10:02.600552 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     result = conn.execute(querycontext.statement, self._params)
[Sun Sep 25 14:10:02.600569 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 914, in execute
[Sun Sep 25 14:10:02.600587 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     return meth(self, multiparams, params)
[Sun Sep 25 14:10:02.600637 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/sql/elements.py", line 323, in _execute_on_connection
[Sun Sep 25 14:10:02.600656 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     return connection._execute_clauseelement(self, multiparams, params)
[Sun Sep 25 14:10:02.600672 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1010, in _execute_clauseelement
[Sun Sep 25 14:10:02.600689 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     compiled_sql, distilled_params
[Sun Sep 25 14:10:02.600749 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1146, in _execute_context
[Sun Sep 25 14:10:02.600770 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     context)
[Sun Sep 25 14:10:02.600786 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1341, in _handle_dbapi_exception
[Sun Sep 25 14:10:02.600802 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     exc_info
[Sun Sep 25 14:10:02.600818 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
[Sun Sep 25 14:10:02.600834 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     reraise(type(exception), exception, tb=exc_tb, cause=cause)
[Sun Sep 25 14:10:02.600877 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
[Sun Sep 25 14:10:02.600894 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     context)
[Sun Sep 25 14:10:02.600910 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]   File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/engine/default.py", line 450, in do_execute
[Sun Sep 25 14:10:02.600926 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526]     cursor.execute(statement, parameters)
[Sun Sep 25 14:10:02.600948 2016] [wsgi:error] [pid 679:tid 1954755632] [remote 192.168.1.5:17526] OperationalError: (sqlite3.OperationalError) no such column: method.x0 [SQL: u'SELECT method.id AS method_id_1, method.name AS method_name, method.method_id AS method_method_id, method.method_type AS method_method_type, method.method_order AS method_method_order, method.start_time AS method_start_time, method.end_time AS method_end_time, method.duration_sec AS me [... cut ...] hod.start_setpoint AS method_start_setpoint, method.end_setpoint AS method_end_setpoint, method.amplitude AS method_amplitude, method.frequency AS method_frequency, method.shift_angle AS method_shift_angle, method.shift_y AS method_shift_y, method.x0 AS method_x0, method.y0 AS method_y0, method.x1 AS method_x1, method.y1 AS method_y1, method.x2 AS method_x2, method.y2 AS method_y2, method.x3 AS method_x3, method.y3 AS method_y3 \\nFROM method \\nWHERE method.method_order = ?'] [parameters: (0,)]

pewi81, 2016-09-25:
Same Internal server error when I try to reach Methods and PID pages!

kizniche, 2016-09-25:
It looks like your database didn't get updated. Run this in a terminal:

```
cd ~/Mycodo/databases
sudo alembic upgrade head
sudo ~/Mycodo/mycodo/update_mycodo.py initialize
sudo /etc/init.d/apache2 restart
```

pewi81, 2016-09-25:
Alright, issue fixed, many thanks for your help :)

kizniche, 2016-09-25:
Cool. It appears there may have been an error during the upgrade process.

Can you see if there was an error in the update log at Tools->Mycodo Logs?

closed 2016-09-29 by kizniche