kizniche/Mycodo #111: Cant see past 24 hours on graph chart

Time4420, 2016-09-24:
Hello Chris, first let me tell you did i use your application for some time now installed on RasperryPi 3B (raspbian) and want to say that you are doing a great job.
I have one issue that on the graph chart i cannot go back past 24hr, i am using DHT sensors.
Also it loses connectivity after approx 5-8 hours of working and only by shutting the power off and on it connects to the wifi router.

closed 2016-09-25 by Time4420

kizniche, 2016-09-28:
I meant to respond to this sooner, but saw you closed it. Have you been able to resolve your issues?

Time4420, 2016-09-28:
I kinda resolved it, what i did is expanding the x-Axis, small problem is when i put more then 10k it is very slow and takes time to show and Pi gets really hot (Obviously because he pulls logs, no? i am a beginner), i would be happy if there was an option to look at history logs even a year back and compare.
Currently i have 4 temp/humidity sensors and planning to add a small motor and float/water sensor.

Time4420, 2016-09-28:
Also the Pi looses connection to the router that i did not resolved yet, i think putting a command that auto restarts the pi.