kizniche/Mycodo #113: Future enhancement: use flask-sqlalchemy in the flask app and sqlalchemy in the daemons

etiology, 2016-10-01:
Don't do this until we get some tests up I'm here. This suggestion will need a large refactor but I think it's a good milestotone after we get some automated tests:

I've  noticed several places where code has basically duplicated the functionality of flask SQL alchemy but not as well since it's really complicated to manage in a web environment.  There is a common need to keep our db models dry and use them in daemons and in flask.  I definitely feel there has been a lack of examples of showing how to reuse sqlalchemy models with flask-sqlalchemy. 

I did find an example of how to do this [here](https://github.com/mitsuhiko/flask-sqlalchemy/issues/98). It's worth thinking about and it would cut the code in this repo down quite a bit. Plus it encapsulates and deligates much of the functionality found in flaskutils to more appropriate places. 

I really like using Flask Sqlalchemy for this because even the sqlalchemy docs [suggest using flask sqlalchemy](http://docs.sqlalchemy.org/en/latest/orm/contextual.html?highlight=flask) in these cases. 

Again I personally feel we need tests and some restructuring before jumping in to this but it feels like the right way to streamline and improve the code base.

etiology, 2016-10-01:
Sorry there are some typos. I'm walking the dog while typing on my phone. Autocorrect is having a field day.

kizniche, 2016-10-01:
Those all sound like good improvements. How would you suggest I contribute? Do you have any documentation for the types of tests you propose or some example code?

etiology, 2016-10-01:
When I get to a keyboard later today I'll write up what I'm doing for the tests. The implementation will take some time but it makes things so much easier moving forward.

When you work in an code base that has good test coverage with quality tests you can get that immediate feedback that you latest commits didn't wreck something. It's a really good feeling.

etiology, 2016-11-01:
I don't think we need this.  If I'm missing something about using flask-sqlalchemy vs what we currently have then we can reopen this

kizniche, 2017-03-23:
I recently ran into some trouble when implementing the first database change using alembic. I have messed up some peoples' databases because I never sorted out how I would initially determine if the database needed upgrading or not. Essentially, when the database is created, it has the newest schema and the alembic version should be set to whatever is the latest for that database. If the database is not newly created, the current alembic version should be compared to what the config file has as the latest version, and if it differs, run the upgrade (```alembic upgrade head```). The issue I have in front of me is determining if the database is being newly-created or if it isn't. Here is the current code in [app.py](https://github.com/kizniche/Mycodo/blob/3aa6d1143be747e1e0f9eaa9dc82cec1286541a9/mycodo/mycodo_flask/app.py#L104-L107):

```python
    with app.app_context():
        db.create_all()
        populate_db()
        alembic_upgrade_db()
```

So with this, the database is created (if it doesn't exist), then the current alembic version is added to the database (if it was newly created and is an empty table), and finally, [alembic_upgrade_db()](https://github.com/kizniche/Mycodo/blob/3aa6d1143be747e1e0f9eaa9dc82cec1286541a9/mycodo/databases/models/__init__.py#L51-L71) runs:

```python
def alembic_upgrade_db():
    """Upgrade sqlite3 database with alembic"""

    def upgrade_alembic():
        command = '/bin/bash {path}/mycodo/scripts/upgrade_commands.sh update-alembic'.format(path=INSTALL_DIRECTORY)
        upgrade_alembic = subprocess.Popen(
            command, stdout=subprocess.PIPE, shell=True)
        (_, _) = upgrade_alembic.communicate()
        upgrade_alembic.wait()

    # If row with blank version_num exists, delete it
    # Then attempt to upgrade the database
    alembic = AlembicVersion.query.first()
    if alembic:
        if alembic.version_num == '':
            alembic.delete()
            upgrade_alembic()
        if alembic.version_num != ALEMBIC_VERSION:
            upgrade_alembic()
    elif not alembic:
        upgrade_alembic()
```

This will delete the alembic row if it's blank then upgrade the database, or just upgrade it if it isn't blank but doesn't match the current version, or if the row doesn't exist just upgrade the database.

What I'd like (and if this is a bad way of doing it, please let me know), is to be able to determine if the database is created, so I can do something like this:

```python
    with app.app_context():
        if database_doesnt_exist:
            db.create_all()
            populate_db()
        else:
            alembic_upgrade_db()
```

What do you think is the most appropriate way to do this, considering the upgrade system, or any other direction this should be taken?

kizniche, 2017-03-24:
It appears the way I came up with works, so I'll stick with that until I find a better solution.

From app.py:

```python
    with app.app_context():
        db.create_all()
        populate_db()
        alembic_upgrade_db()
```

...

```python
def alembic_upgrade_db():
    """Upgrade sqlite3 database with alembic"""

    def upgrade_alembic():
        """ Run alembic database upgrade """
        command = '/bin/bash {path}/mycodo/scripts/upgrade_commands.sh update-alembic'.format(path=INSTALL_DIRECTORY)
        upgrade = subprocess.Popen(
            command, stdout=subprocess.PIPE, shell=True)
        (_, _) = upgrade.communicate()
        upgrade.wait()

    alembic = AlembicVersion.query.first()
    if alembic:  # If alembic_version table has an entry
        if alembic.version_num == '':
            alembic.delete()  # Delete row with blank version_num
            upgrade_alembic()
        elif alembic.version_num != ALEMBIC_VERSION:  # Not current version
            upgrade_alembic()
    else:
        upgrade_alembic()
```

closed 2017-03-24 by kizniche