kizniche/Mycodo #117: What about Creating a Dev branch?

etiology, 2016-10-04:
It might be worth creating a dev branch in the mycodo base and keep trusted code on the master.  You can merge pull requests that you approve to the dev branch and we can test the install and behavior before moving to master.

You can also use GitHub's version marker so that people downloading the project can grab the fully tested and marked version

kizniche, 2016-10-04:
This is something I've wanted to do for a while. I've experimented with building new features on a separate branch, but I just never got into the habit of doing that.

Github versions is also something I've wanted to use, especially if we can create a stable and experimental track that users can select which updates they want to receive (from the web-UI).

kizniche, 2016-10-05:
Just FYI, I'll be beginning an 11-day trip to California tomorrow. I'll still be in communication, but I probably won't be pushing that many commits. I'm going to attempt to get OpenVPN running on one of my Pis so I'll have better access to my dev Pis.

zsole2, 2016-10-06:
You may want to give Openmediavault a try. I have it running smoothly on a Banana Pi for about two years. It was not a smooth  install then, but it was probably due to the board. http://simplenas.com/ was the solution then, it has several ready built images to use. 
Installing OpenVPN is quite simple on it, quasi-automatic. 
I already wanted to suggest you to checkout their dashboard, it is quite a nice system for handling a lot of information in a limited space.

kizniche, 2016-10-06:
I ended up using pivpn. It was extremely easy to install, with just one command. I now have my home VPN access on my phone and laptop.

etiology, 2016-10-29:
What's the word on implementing a dev branch and release versions. It sounds like a win/win

kizniche, 2016-10-29:
Yeah. How do you propose we start?

etiology, 2016-10-29:
I would pull from the main branch to get it up to current. And then create a new branch called ```mycodo_dev``` and push it to to the repo. 

As far as versioning I say the current master is a good place to start. There is a [version button on github](https://help.github.com/articles/creating-releases/).

etiology, 2016-10-29:
After that's set up I say we implement some rules about when the dev branch gets merged to the main. At some point we should look into [Travis CI](https://travis-ci.com/plans) for automated  tests for commits and pull requests.

etiology, 2016-10-29:
Good work!

You should also look into [Codacy](https://www.codacy.com). Automatic code reviews and measurements of code quality. It's a great way to identify areas to focus on for refactoring.

kizniche, 2016-10-29:
Thanks. Those look like some nice tools. I like that they're free for open source projects.

What type of rules do you propose?

Should I merge the admin_creation branch to the mycodo_dev branch?

etiology, 2016-10-29:
I haven't ran it on a fresh install yet but I'd say it's a great one to start with.  I don't have any expertise in creating policies for merging and releasing but I can offer my thoughts on it.  Here is my ideal *pie in the sky* rules:
- Ensure that new users have a good experience with the install and configuration 
   - branch must be passing all tests
   - new code should never decrease test coverage (i.e. Adding code that does not have tests covering it)
   - new functions, methods, and packages should have tests that verify expected behavior
-  New versions must be tested from a **new user's** experience via fresh install on a clean image. The person verifying the release should run through the installer, create a sensor, view graphs, etc. 
 - New versions must also be tested for an established users experience - a system with sensors assigned and configured which goes through a patch, database migration, upgrade, or install. 

Some of this could be overkill but I do feel there is a *do no harm* check that must take place before it gets a new release. Also improving test coverage and isolation is a big deal to me. It's part of being a good developer imho. One writes tests that demonstrate their code does what they think it does under various likely conditions.

etiology, 2016-10-29:
Oops. I'm on a phone and can't seem to edit the second statement was supposed to be about established user testing to preserve their experience.

kizniche, 2016-10-29:
That all sounds good to me. I've been experimenting with Travis CI and Codacy. There may be a flood of commits while I experiment with .travis.yml settings.

I merged admin_creation to mycodo_dev and deleted admin_creation

kizniche, 2016-10-29:
Alright. I got as far as I can for now with Travis CI. I'm stuck at a [smbus requirement error](https://travis-ci.org/kizniche/Mycodo/builds/171682011). I recall having issues with smbus in the past, but not sure what I did in the setup.sh script to fix that issue. I thought I moved all the pertinent commands to pre-install of .travis.yml that would resolve that issue.

kizniche, 2016-10-29:
How do you think the update system should eventually work? Downloading of releases or from master?

etiology, 2016-10-29:
I'm not sure. I'll have to do some research on different approaches before I have any opinion on it. Seems like an area worth considering a number of options. Once you choose a path you're kind of committed.

kizniche, 2016-10-29:
It's strange that when I'm outside the virtualenv I don't receive the smbus erorr, but when I'm in the virtualenv, it happens.

Outside virtualenv:

```
~/Mycodo$ sudo pip install --upgrade -r requirements.txt
...
Requirement already up-to-date: smbus in /usr/lib/python2.7/dist-packages (from MCP342x->-r requirements.txt (line 31))
...
  Running setup.py install for MCP342x ... done
Successfully installed Adafruit-ADS1x15-1.0.2 Adafruit-BMP-1.5.2 Adafruit-GPIO-1.0.1 Adafruit-TMP-1.6.1 MCP342x-0.3.0 RPi.GPIO-0.6.2 adafruit-pureio-0.2.1 argparse-1.2.1 picamera-1.10 plumbum-1.6.2 rpyc-3.3.0 sht-sensor-16.2.0 spidev-3.2 tentacle-pi-0.6.2 w1thermsensor-0.3.1
```

In virtualenv:

```
(virtualenv) ~/Mycodo$ pip install --upgrade -r requirements.txt
...
Collecting smbus (from MCP342x->-r requirements.txt (line 31))
  Could not find a version that satisfies the requirement smbus (from MCP342x->-r requirements.txt (line 31)) (from versions: )
No matching distribution found for smbus (from MCP342x->-r requirements.txt (line 31))
```

kizniche, 2016-10-30:
Just a quick update: I found a solution. If I set --system-site-packages when creating the virtual environment, It inherits smbus that's been installed by apt (python-smbus) and can complete installing all modules form requirements.txt. I wonder if I can alter the environment in Travis CI.

kizniche, 2016-10-30:
Well, I'm stumped again after finding this under [Building a Python Project](https://docs.travis-ci.com/user/languages/python/#Travis-CI-Uses-Isolated-virtualenvs):

> CI Environment uses separate virtualenv instances for each Python version. System Python is not used and should not be relied on. If you need to install Python packages, do it via pip and not apt.

> If you decide to use apt anyway, note that Python system packages only include Python 2.7 libraries on Ubuntu 12.04 LTS. ***This means that the packages installed from the repositories are not available in other virtualenvs even if you use the –system-site-packages option***.

kizniche, 2016-10-30:
Well, that's wasn't accurate, because there's apparently an option for .travis.yml that does just that!

```
virtualenv:
    system_site_packages: true
```

kizniche, 2016-10-30:
Finally figured out Travis CI, and got the dependencies sorted and the [endpoint tests to successfully run](https://travis-ci.org/kizniche/Mycodo/builds/171704536)!

etiology, 2016-10-30:
Things are looking pretty sharp. 

You've been busy today. Good work on getting Travis and codacy up and running. I want to say codacy needs some detailed configuration for python. I think you have to tune it in order to identify areas that the project should focus on for refactoring. I think there is a language section and a place to turn on and off different evaluations.  The suggestions it makes can be really good. 

I'd say it's time to place those status badges in the read me. Show off the project status.  Class it up a little.

kizniche, 2016-10-30:
That was a lot of trial and error trying to understand the root of that smbus issue. I ended up using a virtual machine of the exact version to test in case it was a package version issue. Turns out it was a virtualenv issue that I learned a lot about its package management in the process.

I already put the badges in the mycodo_dev branch README ;) but am waiting to tweak the code to show the master until we merge mycodo_dev.

etiology, 2016-10-30:
Oh cool. I totally forgot to check the dev branch. I see it now

closed 2016-11-01 by etiology