kizniche/Mycodo #118: Need a testing fixture/function for logging a test user in and out

etiology, 2016-10-05:
#### Testing Endpoints needs to login and logout users

Just putting this on the board.  Currently the test suite doesn't log any user in so we can only test that endpoints redirect to the login page or that sensors can post data.  

##### Possible Approach
To do this we may want to use something similar to the [Loggin In and Out](http://flask.pocoo.org/docs/0.11/testing/#logging-in-and-out) section of testing flaks.

-----------

It would be a *Really* good move if as a milestone we move over to using one of these tools instead:
  - [Flask-Login](https://flask-login.readthedocs.io/en/latest/)
  - [Flask-Security](https://pythonhosted.org/Flask-Security/)

Both suggestions above do the same thing since Flask-Security uses Flask-Login.  Again this would be a very long term goal.  Streamlining the mycodo code and getting test coverage up helps prepare for this level of refactoring.  Lots of things change so we need an automated way to ensure that nothing is broken after those changes.

kizniche, 2016-10-05:
That looks good and shouldn't be terribly difficult to refactor. Does the test script need actual credentials for an admin user to operate?

etiology, 2016-10-05:
Yes we will need the ability to create admins and other users.  The tests need to be able to verify the apps behavior for all users including the daemons.  

When each test runs it builds a new clean environment, runs the test, and then destroys everything it setup.  This makes sure that each test doesn't get false results caused by previous tests and their unique setup.  

So for each user type (guest, admin, daemon, etc) we need to be able to create the user in the database, have them login (however they do that) and then visit the endpoints that we want to test.

closed 2016-10-08 by kizniche

etiology, 2016-10-08:
The tests should never need root access. Eventually it would be nice to setup automated tests that run whenever we push changes. 

We need more tests to walk through a user setting up sensors, and controllers using the web interface. The 'testapp' fixture is a WebTest app. If you look at the docs it shows how to fill out and submit forms, load pages, and post data. There are a few tests I put in that show examples of how to do this.

kizniche, 2016-10-08:
Hmm. I looked over it but am still a little perplexed how to actually run a test.

etiology, 2016-10-09:
I'm heading back from a camping trip right now so it will be a few hours before I can post something in depth. 

I had a similar reaction when first using pytest. What IDE do you use?  I use PyCharm and running tests is just a keystroke once you set it up.   I'll post some content on how to run and setup pytest.  You can also use the command line to run it.

kizniche, 2016-10-09:
I've only use Sublime to write my code, probably to the detriment of my productivity.

etiology, 2016-10-09:
Sublime is great.  It's one of my favorite text editors.

For extra code goodness you should check out [PyCharm](https://www.jetbrains.com/pycharm/download/).  They also have an [Edu](https://www.jetbrains.com/pycharm-edu/download/) version while you are still doing lab work.  Using pycharm has been amazing for me.  If you turn on the different [linters](https://en.wikipedia.org/wiki/Lint_(software)) and pep 8 inspections it really helps.

etiology, 2016-10-09:
How I run pytest from CLI (there may be better ways when using the command line interface):

#### How to run all of the tests in test_endpoints.py
 - Attach your virutal environment (mine is called ```mycodo_py2_venv```)
```bash
mac:mycodo username$ source ~/PythonVirtualEnvironments/mycodo_py2_venv/bin/activate
(mycodo_py2_venv)mac:mycodo username$ 
```

 - move into the ```Mycodo/mycodo``` directory
```bash
(mycodo_py2_venv)mac:Mycodo username$ cd mycodo
```

 - call pytest through the python command using the ```-m``` flag to set the correct python path (which avoids import errors)
```bash 
(mycodo_py2_venv)mac:mycodo username$ python -m pytest tests/software_tests/test_mycodo_flask/test_endpoints.py

============================== test session starts ==============================
platform darwin -- Python 2.7.11, pytest-3.0.3, py-1.4.31, pluggy-0.4.0
rootdir: /Users/username/PycharmProjects/Mycodo/mycodo, inifile: 
collected 7 items 

tests/software_tests/test_mycodo_flask/test_endpoints.py .......

============================== 7 passed in 1.00 seconds ==============================

``` 
 - Later have a look at pytest's ```--help`` command for more info.  The command shown above just tells pytest to run everything in the ```test_endpoints.py``` module.  

#### How to run a specific tests
 - Move into the ```Mycodo/mycodo``` directory with your virtual environment attached
 - Just like the previous command we call pytest using the ```python -m``` command, but this time we use pytest's ```-k``` flag to only run tests that match an expression.  In this case, lets run the  ```test_user_can_login``` function.
```bash
(mycodo_py2_venv)mac:mycodo username$ python -m pytest tests/software_tests/test_mycodo_flask/test_endpoints.py -k test_user_can_login
============================== test session starts ==============================
platform darwin -- Python 2.7.11, pytest-3.0.3, py-1.4.31, pluggy-0.4.0
rootdir: /Users/username/PycharmProjects/Mycodo/mycodo, inifile: 
collected 7 items 

tests/software_tests/test_mycodo_flask/test_endpoints.py .

============================== 6 tests deselected ==============================
============================== 1 passed, 6 deselected in 0.73 seconds ==============================
```

When we get more tests in here you could see how great this is.  After making changes we could be certain that our users can still login, add sensors, view graphs, and see the correct content in those graphs.  Tests could be written to make sure that when a user makes changes to their settings that these actually take place.  It starts to be really nice to see that new changes don't harm the user experience while you make small adjustments.

kizniche, 2016-10-25:
I installed PyCharm and spent some time getting familiar with it. Cloned this repo. Installed modules (though some weren't able to be installed, e.g. RPi). Read a bunch of the code change recommendations from the linters.
I read in #128 your wiki suggestion and thought it would be good to combine this stuff to a central location instead of getting lost in the closed issues, so I opened the first wiki page, [Testing](https://github.com/kizniche/Mycodo/wiki/Testing).

etiology, 2016-10-25:
That's great!  If you don't mind I'll add some tutorials on how to use the fixtures for writing new tests.  Making use of the tests while writing new code makes it easy to check on the outcome too.

etiology, 2016-10-25:
That's great!  If you don't mind I'll add some tutorials on how to use the fixtures for writing new tests.  Making use of the tests while writing new code makes it easy to check on the outcome too.  

For instance, lets say you are changing the page content of the dashboard and you want to see if it's working.  The [WebTest response](http://docs.pylonsproject.org/projects/webtest/en/latest/testresponse.html) object has the method ```showbrowser()``` which will open the browser and show you the page.  It doesn't have any styling to it but it's a neat trick.  Any of the [responses](https://github.com/kizniche/Mycodo/blob/50402abec6d484c9fb8cb8848f4755f5bdf1096d/mycodo/tests/software_tests/test_mycodo_flask/test_endpoints.py#L63) can do this as well as filling out forms and checking page content.

kizniche, 2016-10-25:
I don't mind at all. I set the wiki to be publicly-editable. Thanks!

etiology, 2016-10-25:
That’s great!  It’s a good IDE for sure.  

Have you tried running the tests yet? You may have to change your settings to use pytest:
(In Settings/Preferences) —> Tools —> Python Integrated Tools —> Default Test Runner (choose pytest from dropdown)

Now you can right click on the ‘software_tests’ package and choose ‘Run Pytest’

Also you should look at the ‘Deployment’ section under the ‘Tools’ menu bar.  Here you can configure your raspberry pi sftp settings and just push your code to it.  You can also remote run the code and monitor it.

kizniche, 2016-10-25:
I've run the tests from the Pi, but haven't yet from PyCharm. I spent most of my time in my first session trying to understand all the settings, set up the virtual env., set up/instal modules, and then went through the huge amount of code warnings.

It definitely helps to have some guidance on how to use it effectively with this project, so I'll try your suggestions later. Thanks!

kizniche, 2016-10-25:
I wasn't able to run pytest past creating a user. It fails at importing pwd. I also don't seem to have the deployment option with my PyCharm install. I believe the free version lacks the features of the pro version.

etiology, 2016-10-25:
that might be the case.  you should be able to run the tests locally without interacting with the MCU.  When the tests startup I [mock out RPI.GPIO and picamera](https://github.com/kizniche/Mycodo/blob/50402abec6d484c9fb8cb8848f4755f5bdf1096d/mycodo/tests/software_tests/test_mycodo_flask/conftest.py#L7).

I'll have to try it on the community version.  Don't forget that you have access to the educational version because of your phd work.  They need to send some emails to a edu domain address to verify that you are a staff/student but it's totally worth it.  Plus you're likely to show others in the lab.  I did this for my bioinformatics work before buying the pro version.

kizniche, 2016-10-25:
I'm using PyCharm Edu. I believe my pwd issue is that there's been no version of pwd ported to windows. You probably don't see this error on mac because it has pwd.

etiology, 2016-10-25:
Ahh that makes sense.  I was thinking you were having the the pi run the tests or something.  

maybe we make an issue for it and [try something like this](https://stackoverflow.com/questions/16529166/how-to-shadow-python-builtin-pwd-module):

```python
try:
    import pwd
except ImportError:
    import winpwd as pwd
```

kizniche, 2016-10-25:
I found that workaround a few places. I finally passed that error and am now hung up on importing flaskforms from app.py.

```
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\tests\software_tests\test_mycodo_flask\conftest.py", line 15, in <module>
    from mycodo.mycodo_flask import app as _app
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\mycodo_flask\__init__.py", line 3, in <module>
    from .app import app
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\mycodo_flask\app.py", line 65, in <module>
    import flaskforms

Process finished with exit code 1
```

kizniche, 2016-10-25:
I fixed the flaskforms import by defining my project structure in settings. Then grp gave issues importing but mimicing what was done with pwd fixed that. Now I'm getting an error importing resources.

```
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\tests\software_tests\test_mycodo_flask\conftest.py", line 15, in <module>
    from mycodo.mycodo_flask import app as _app
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\mycodo_flask\__init__.py", line 3, in <module>
    from .app import app
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\mycodo_flask\app.py", line 90, in <module>
    from utils.statistics import return_stat_file_dict
  File "C:\Users\Kyle\PycharmProjects\Mycodo\mycodo\utils\statistics.py", line 11, in <module>
    import resource

Process finished with exit code 1
```

kizniche, 2016-10-25:
Hah, I should just go into linux to use PyCharm! I guess I like a challenge, but this is getting ridiculous.

etiology, 2016-10-25:
I applaud you for trying it on windows.  That's hard to do.

Linux would seem to solve the issues and bring your dev environment closer to the production environment.  You can also try just running the code on the [raspberry pi via the remote configuration](https://www.jetbrains.com/help/pycharm/2016.1/configuring-remote-interpreters-via-ssh.html).  You can even set your virtual environment on your pi and run the code using it.

kizniche, 2016-10-25:
I would have preferred the remote config, but it doesn't appear PyCharm Edu supports that. I have a lot of different devices/OSs I use and remote config sounds ideal.

kizniche, 2016-10-26:
Also, it appears PyCharm Edu doesn't verify your .edu email to allow the download. I'm not exactly sure how it differs from the community version.

etiology, 2016-10-26:
Looks like the 2016 version changed everything.  I did the edu package with the 2015 for the lab.  It sounds like it's behaving just like the community version until you do their course work.

kizniche, 2016-10-26:
I've been looking through the documentation, but haven't found anything about unloading features. Where did you find that information about the course work?

kizniche, 2016-10-26:
Well, PyCharm in linux was cake to set up. Tests ran successfully.

```
============================= test session starts ==============================
platform linux2 -- Python 2.7.11+, pytest-3.0.3, py-1.4.31, pluggy-0.4.0
rootdir: /home/kiz/PycharmProjects/Mycodo/mycodo/tests/software_tests/test_mycodo_flask, inifile: 
collected 7 items

test_endpoints.py ....../home/kiz/MycodoVirtEnv/local/lib/python2.7/site-packages/bs4/__init__.py:146: UserWarning: You provided Unicode markup but also provided a value for from_encoding. Your from_encoding will be ignored.
  warnings.warn("You provided Unicode markup but also provided a value for from_encoding. Your from_encoding will be ignored.")
.

=========================== 7 passed in 1.25 seconds ===========================
Process finished with exit code 0
```

kizniche, 2016-10-26:
Aha! I just found their form: https://www.jetbrains.com/shop/eform/students

kizniche, 2016-10-26:
Ooooo
[image]

etiology, 2016-10-26:
Nice you got the pro version.  I feel that they nailed it on this ide. One of the best

kizniche, 2016-10-26:
Hmm. I'm setting up the remote deployment and am stuck at this, requesting I choose either *create* or *move*.

[image]

kizniche, 2016-10-26:
Okay. I got to what I think is a working remote deployment setup, however I'm getting a strange set of errors when I attempt to run the tests:

```
ssh://kiz@192.168.0.7:22/usr/bin/python -u /home/kiz/.pycharm_helpers/pycharm/utrunner.py /home/kiz/Mycodo/mycodo/tests/software_tests/ true
Testing started at 03:02 ...
/home/kiz/Mycodo/mycodo/tests/software_tests/test_mycodo_flask/test_endpoints.py:3: RuntimeWarning: Parent module 'tests.software_tests.test_mycodo_flask' not found while handling absolute import
  import mock
/home/kiz/Mycodo/mycodo/tests/software_tests/test_mycodo_flask/test_endpoints.py:4: RuntimeWarning: Parent module 'tests.software_tests.test_mycodo_flask' not found while handling absolute import
  from mycodo.tests.software_tests.factories import UserFactory
/home/kiz/Mycodo/mycodo/tests/software_tests/test_mycodo_flask/test_endpoints.py:5: RuntimeWarning: Parent module 'tests.software_tests.test_mycodo_flask' not found while handling absolute import
  from mycodo.tests.software_tests.test_mycodo_flask.conftest import login_user
28.4205185895

Process finished with exit code 0
Empty test suite.
```

kizniche, 2016-10-26:
Alright! Success! I'm going to call it a night for working with PyCharm. I think that's a good note to end on for now :)

```
ssh://kiz@192.168.0.7:22/usr/bin/python -u /home/kiz/.pycharm_helpers/pycharm/pytestrunner.py -p pytest_teamcity /home/kiz/Mycodo/mycodo/tests/software_tests/test_mycodo_flask
Testing started at 03:04 ...
28.5052330844
============================= test session starts ==============================
platform linux2 -- Python 2.7.9, pytest-3.0.3, py-1.4.31, pluggy-0.4.0
rootdir: /home/kiz/Mycodo/mycodo/tests/software_tests/test_mycodo_flask, inifile: 
collected 7 items 

test_endpoints.py ....../usr/local/lib/python2.7/dist-packages/bs4/__init__.py:146: UserWarning: You provided Unicode markup but also provided a value for from_encoding. Your from_encoding will be ignored.
  warnings.warn("You provided Unicode markup but also provided a value for from_encoding. Your from_encoding will be ignored.")
.

========================== 7 passed in 12.14 seconds ===========================

Process finished with exit code 0
```

PR #121 (2016-10-08): Endpoint separation
Closes #118 
Closes #112

#### Isolated Login and Logout
The ```/login``` and ```/logout``` endpoints were pulled out of app.py and moved into its own blueprint under the ```authentication``` package
```
│   ├── mycodo_flask - HTTP server files (Flask)
│   │   ├── static - Static files reside (images, css, js, etc.)
│   │   ├── static - Static files reside (images, css, js, etc.)
│   │   └── authentication - Flask HTML templates
│   │       ├── views.py  <--- Contains a blueprint for login/logout routes
```

#### Created a factory for generating Users
A ```factory.py``` fil [... cut]