kizniche/Mycodo #119: Error in the Edge or State field - Not a valid choice

filipdc, 2016-10-05:
When adding a Sensor Conditional, I get stuck on this error all the time-Error in the Edge or State field - Not a valid choice.
I can't figure out where the mistake is.
Can someone please point me in a direction.
This is a awesome project.

kizniche, 2016-10-05:
Can you first verify for me two things?

1) You are running the latest version, by going to Admin->Upgrade

2) Go to Tools->Mycodo Logs->Update Log, and see if the most recent upgrade ended successfully with the following message (note the time will be different):

```
#### Update Finished 10-04-2016 17:14:55 ####
```

kizniche, 2016-10-05:
My first guess is you may have upgraded to a version where there was a bug (your issue). My second guess is you upgraded and interrupted the upgrade before it could properly update the database. For instance, during an update that included updating numpy Python module, it could take 10 minutes (Pi3) to 2 hours (PiZero) to finish the update.

filipdc, 2016-10-05:
It was a fresh install, done today on a B+.
I can’t find any information about an upgrade.
You you advise a clean reinstall?

kizniche, 2016-10-05:
Try this before reinstalling:

1. Rename, move, or delete your current mycodo.db from ~/Mycodo/databases/
2. Execute ```~/Mycodo/init_databases.py -i all``` to recreate your mycodo.db
3. Add your sensor again, then try to create a conditional.

kizniche, 2016-10-05:
And just to be clear, are you receiving this error a) trying to add an Edge conditional, or b) trying to save the configuration of an Edge conditional?

kizniche, 2016-10-05:
Note that after the 3 steps above, you will need to restart the daemon (or restart the Pi).

kizniche, 2016-10-06:
I just reinstalled Mycodo on one of my Pis but could not reproduce your issue. Perhaps you have a bad install for some reason. Let me know if my suggestions above worked. If not, I would suggest reinstalling Mycodo.

kizniche, 2016-10-06:
I just pushed a small update that may fix your issue. Update from Admin->Upgrade and let me know if you still get that error.

filipdc, 2016-10-06:
> Op 6 okt. 2016, om 00:14 heeft Filip De Clerck <filipdeclerck@me.com> het volgende geschreven:

I did a fresh install, everything is working perfect now.
I guess something went wrong during the initial install, no worries.
Thanks, such a great project!

kizniche, 2016-10-06:
Glad to hear!

closed 2016-10-06 by kizniche