kizniche/Mycodo #120: Unable to modify sensor settings

Ryobidelacroix, 2016-10-06:
Hi! 

Sorry to bombard you with messages, I know you're busy and working very hard on what is a fantastic project. I thought I would raise this issue that i've been having recently. 

Whenever I attempt to change _ANY_ sensor settings I receive this message **"Sensor settings were not able to be modified: invalid literal for int() with base 10: 'RPi'"**. I have attempted a fresh install of Mycodo on a new SD card (and followed your steps precisely) , however this does not seem to resolve the issue. 

Also, I love the ability to track how much power a certain relay is consuming, but I haven't used it due to differences in our AC voltage. Is there any way to change the AC voltage in Mycodo from 110v to 240v for us Australians?

kizniche, 2016-10-06:
What type of sensor are your trying to save the settings for?

Yes, I did have a setting to change the AC voltage! This setting was in version 3.0, before refactoring everything for version 4.0. Sorry for not bringing it over sooner. I should be able to bring that setting in shortly.

Ryobidelacroix, 2016-10-06:
As this is a fresh install I haven't added many sensors just yet. I was attempting to change settings on the "Raspberry Pi CPU Temperature" and a TSL2561 with no luck.

kizniche, 2016-10-06:
I just updated the settings (Admin->Configuration) to include the option to change the voltage, and I'm currently reinstalling Mycodo fresh to test if I receive the same error.

kizniche, 2016-10-06:
I was able to reproduce the issue. I'm trying to fix it now.

kizniche, 2016-10-06:
Found the issue. Let me know if you still have issues after you update.

closed 2016-10-06 by kizniche

Ryobidelacroix, 2016-10-06:
All fixed! Thank you!

kizniche, 2016-10-06:
Glad to hear! Thanks for submitting the issue :)

commit 58e8a9e3 (2016-10-06): Fix inability to save sensor settings. Fixes #120.