kizniche/Mycodo #123: Relay conditions not actived when PID issue relay raise or lower

lbechet, 2016-10-06:
Hi,

Using Mycodo 4.022 I am able to set a "conditional" statement for the relays stating the following :  
- when  relay 1 on  (0 secs) -> relay 2 on (0 secs)
- when relay 1 off (0 secs) -> . relay 2 off(0 secs)
This works like a charm within the web application clicking on the on or off for relay 1

Unfortunately, PID controller, when raising or lowering relay 1 ignore the conditional relay statements.

To my point this is an issue but maybe I should setup two identical PIDS with the two relays. 

Sincerely Yours,

Laurent.

kizniche, 2016-10-06:
That's strange, because I have the exact same setup for two relays, with one being controlled by a PID, and it works as expected. I may not be able to test right away, because I'm not near a computer.

Regarding creating two PIDs, if they will be regulating the same environmental condition in the same syatem/container, then there's a good change they will interfere with themselves. Without knowing your setup (system, sensors, relays) it's difficult to determine if it will work.

kizniche, 2016-10-06:
Alternatively, you can use a splitter so your single relay powers both devices. If you don't need to use the second device separately from this one PID, then there's really no use in using two relays when one will suffice.

lbechet, 2016-10-07:
Hi,
As you say it works for you I will do a new fresh install and retry everything.

The fact is I can't use same relay as my goal is to use one relay to provide AC to a DC 12 volt and the second relay to provide this 12 volts to the heater. (This is done in order to reduce power consumption)

If it don't work perhaps I can solder the two data pins to force these to relays to work together. This would be my last resort option. It would be better to activate the 1st relay with the power before the second and shut it down after ..

Sincerely Yours,

Laurent.

lbechet, 2016-10-08:
Hi,

After complete reinstall (noobs then mycodo then 18b20 1wire setup) I still find the same issue with the relay conditional rules.

It works fine from the web interface relay "on" "off" buttons but if PID raise or lower the relay the relay conditional rules are all ignored.

kizniche, 2016-10-08:
I just pushed an update. Update Mycodo and see if works.

lbechet, 2016-10-08:
It works perfectly now !.

Thank you a lot.

closed 2016-10-08 by lbechet