kizniche/Mycodo #125: Not an issue. Kudos and dev. requests?

montreuilg, 2016-10-18:
What a fantastic piece of work, bravo!

Is there a particular place/forum where we discuss development, or we just open a new issue, like I just did... ;-)

I would like to help add/define new sensors and new devices(rfm95 enabled).

I'm returning to programming after a 20 year hiatus, learning Python in the process, so do not put your hopes too high... I should be able to grasp the concepts and at the very least, help with the testing!

Thanks for this intermission and once again, kudos everyone involved for such a quality product. Installation was a breeze and flawless!

_
gabriel

kizniche, 2016-10-18:
Thanks, Gabriel.

So far this is the only forum for Mycodo.

At the moment it is not too difficult to add sensors, but there are a few parts of the system that need to be edited in order for both the backend and frontend to recognize the sensor module and be able to pull data from the database. I plan to make this easier to integrate sensor modules, with a minimal number of edits required to add a new sensor.

If you want to begin testing sensor code, you can go to [Mycodo/mycodo/sensors/](https://github.com/kizniche/Mycodo/tree/master/mycodo/sensors) and check out each sensor module to get an idea of how they work. I used the same model for each sensor, so they should all look similar and [hopefully] be somewhat easy to understand. Check out [raspi.py](https://github.com/kizniche/Mycodo/blob/master/mycodo/sensors/raspi.py) and execute it with ```python ~/Mycodo/mycodo/sensors/raspi.py``` to see it retrieve the CPU and GPU temperatures.

Also check out commit 2252bcf that includes all the edits necessary to add a new sensor, and feel free to ask any questions.

montreuilg, 2016-10-18:
Perfect, thanks a lot!

montreuilg, 2016-10-19:
Thanks for the roadmap, it will help! Here is the first sensor I will work on:
https://github.com/Miceuz/i2c-moisture-sensor/blob/master/README.md

À bientôt,

_
gabriel

kizniche, 2016-10-19:
Is that using the chirp board? https://wemakethings.net/chirp/ That looks pretty nice.

> If any I2C communication is received during that time, Chirp will switch into sensor mode - it will not chirp, just respond to I2C requests. The default address of the Chirp is 0x20.

Great design all around!

kizniche, 2016-10-19:
I just found the bare sensor board for those who would just like to use it as a sensor: https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/

montreuilg, 2016-10-19:
I have the sensor only version, it works with either spi or i2c.

montreuilg, 2016-10-19:
... closed by mistake...

kizniche, 2016-10-19:
No prob. Lets keep this open until we have the sensor supported.

kizniche, 2016-10-19:
I just ordered the sensor only version. I have one soil moisture probe that I use with my analog to digital converter and would like to try this more "intelligent" version.

etiology, 2016-10-19:
At some point it would be worth discussing what sort of design pattern should be used to implement sensors.  This is an area of the app that changes the most and it would benefit from being abstracted away from the rest of the system.  Right now it seems that although it changes the most, it also requires a good deal of effort to implement.  Some of which spilling over into other logically separate areas of the app.

I've only looked at the sensor implementation briefly but I can see that the UI knows too much about each sensor, and each sensor knows too much about the business logic and UI.  There needs to be some sort of middleware that keeps sensor implementation away from the business logic.

I like the idea of [having a middleware](http://home.labs.mediatek.com/media/apis/2523/sensor_subsys_architecture.png) that the sensors interact with.  Maybe this is a class or package that handles the communication.

kizniche, 2016-10-19:
I've been thinking about how to redesign the sensor modules so all information is contained in the module itself, meaning a new sensor module could be dropped into the sensor directory and be immediately picked up by the backend and frontend. Right now, in addition to adding the sensor module, the frontend/backend code needs to be edited to determine how the sensor class is accessed, and how many and what type of measurements the sensor provides.

Are you referring to a separate class/function that the frontend and backend interact with to read sensor information and measurements?

montreuilg, 2016-10-19:
Something in the way of node-red?  https://github.com/node-red/node-red

kizniche, 2016-10-19:
I realized that a redesign of the sensor code will be necessary, as I was trying to integrate the BME280 sensor. For instance, I use a classification for sensor types (temperature sensor, humidity sensor pressure sensor, etc.) that provide structure for displaying information on the frontend. The BMP pressuresensor sensor can sense temperature, pressure, and altitude. The BME280 pressuresensor can sense temperature, humidity, dew point, pressure, and altitude. This does not fit with the pressuresensor type I had built in, and consequently had to add more conditionals to account for the additional measurements that are outside the typical pressure sensor type. If we contained the number of measurements within the sensor module class, then the frontend could read that instead of hard-coding it into the Flask template.

montreuilg, 2016-11-03:
Well as it turns out, the learning curve is much steeper than what I anticipated (Electronics, linux, sql dbs, c++, python and then Mycodo). I will not be able to help as you probably guessed. This is a very interesting project, keep up the excellent (re) work!

_
gabriel

etiology, 2016-11-03:
I hear you. There is a whole lot to take on all at once. It's something that I think everyone gets hit with at some point. 

If you have the time I'd like to get your thoughts on what resources we could add to make it easier. For people starting a project, where do you see barriers?  Would example QuickStart projects help?  Are there pages/configurations that need attention?

I have a few areas of my own that I'm struggling to fully grok that I feel some documentations and guide could help.  I want to make a list of these problem areas so we can address them once the back end dev settles down.

kizniche, 2016-11-14:
I pushed an update today that added the Chirp moisture sensor. I received it a few days ago but haven't had a chance to plug it in to test until today. Even so, I haven't tested it in a real-world environment, but it does appear to be measuring properly. I figured I'd push the update and let you play around with it and see if there are any errors you can find, before I get to testing it later this week.

closed 2017-07-07 by kizniche