kizniche/Mycodo #127: Sensor addition: LiFePO4wered/Pi

montreuilg, 2016-10-20:
What do you all think of adding this one:

http://lifepo4wered.com/lifepo4wered-pi.html

kizniche, 2016-10-21:
Do you have one to test?

What do you propose be added for measuring? I see voltage may be the only measure.

montreuilg, 2016-10-21:
I'm supposed to get it early week. USB charge voltage and output voltage will be enough.

MurderLabz, 2017-01-20:
montreuilg - Did you ever get your LiFePO4wered? I  like you can use 18650 batt's. I look into getting one of these(or a clone)

kizniche, 2017-01-20:
How long can it keep the Pi running after a power loss?

MurderLabz, 2017-01-21:
I don't see any math penned out. But a 18650 Batt 3500mah 3.7v stepped up to 5v at 2.5 Amps (Max load) with -20% efficiency loss.... Rough guess over 1 hr? At idle I would think about 1/2 a day.

Here is the project blog https://hackaday.io/project/18041/logs

kizniche, 2017-10-18:
Did you ever receive this and still have interest in adding it as a sensor in Mycodo?

closed 2018-04-06 by kizniche