kizniche/Mycodo #130: Create a controller package with a common setup

etiology, 2016-10-23:
Part of #128

The controller modules seem like a good place for a package.   The ```controller``` package can store all of the individual controller modules and contain a single point for each module to keep it's setup code.  Currently there are multiple places where simple resource setup is repeated over and over again [ [1](https://github.com/kizniche/Mycodo/blob/c0df5dc7a9c06255eceb54f8a28c1ad84da39279/mycodo/scripts/initialize_relays.py#L32), [2](https://github.com/kizniche/Mycodo/blob/40f832904dfb93107edece9a2783dc448e528ab9/mycodo/controller_log.py#L33), [3](https://github.com/kizniche/Mycodo/blob/840401471efc94ceca30908108ff030df89802e8/mycodo/controller_timer.py#L23), [4](https://github.com/kizniche/Mycodo/blob/58e8a9e3773fb6be67b6963f0c2a18d26c6b37e0/init_databases.py#L42), [5](https://github.com/kizniche/Mycodo/blob/40f832904dfb93107edece9a2783dc448e528ab9/mycodo/controller_lcd.py#L74), [6](https://github.com/kizniche/Mycodo/blob/840401471efc94ceca30908108ff030df89802e8/mycodo/controller_pid.py#L54), [7](https://github.com/kizniche/Mycodo/blob/840401471efc94ceca30908108ff030df89802e8/mycodo/mycodo_daemon.py#L87), [8](https://github.com/kizniche/Mycodo/blob/3d3f0d61d39e1382fb9160cb94466b98ed3a7b84/mycodo/controller_relay.py#L37), [9](https://github.com/kizniche/Mycodo/blob/4a4d1c10c3a8304ab9478266451be2941c748fcb/mycodo/controller_sensor.py#L59)].  

The implementation is open to discussion but let me describe what I am seeing.  Each of the ```controller_x.py``` modules has a single class that inherrits from ```Thread```:

controller_lcd.py
```python
 class LCDController(threading.Thread)
```

controller_log.py
```python
class LogController(threading.Thread)
```

controller_pid.py
```python
class PIDController(threading.Thread)
```

controller_relay.py
```python
class RelayController(threading.Thread)
```

controller_sensor.py
```python
class SensorController(threading.Thread)
```

controller_timer.py
```python
 class LCDController(threading.Thread)
```

I think I would benefit from some discussion on this because of how this whole part is implemented.  I could see placing the database setup that occurs right above these class definition lines in a parent class that can accept a configuration object just like the one in #129.  Each of these subclasses could let the parent class setup the environment through a new ```config(config_obj=ProdConfig)``` method.

I consider this area to be difficult to mess with.  The goal is to make it so that we can test this area of the code base without mock patching a bunch of setup code that is implemented in each module.  So having one area that creates controllers that use different configurations would be easy to create test fixtures for.  Maybe we just make the constructor of each controller take a config object.  Again I would suggest we keep it dry by creating a parent class to implement this.

kizniche, 2016-10-26:
That sounds reasonable. I suppose I didn't design this area too well because I started with just one controller, and before long, I had a half dozen designed on the same model.

kizniche, 2019-10-07:
I've recently created a base controller class and moved all the controllers into it's own directory. This has also enabled me to implement (in a currently-unreleased version, on master) a custom controller system where users can upload their custom controller in a single-file. These are now located (as of writing), at [Mycodo/mycodo/controllers/](https://github.com/kizniche/Mycodo/tree/2adc882a2d4a65b97d0b277fe10f80f776763ddf/mycodo/controllers)

As far as integrating testing into this area, I'm not sure. If you still have any interest in this project, and would like to offer any advice, that would be nice.

closed 2020-07-23 by kizniche