kizniche/Mycodo #131: Improve installation behavior

etiology, 2016-10-26:
I find myself wanting this installation routine to be smoother so I'm just making an issue to acknowledge it and provide space for open discussions on how it should proceed.  Here are a few areas that where we could improve the user experience:
 - Check for requirements and only install what isn't already installed.  The installer may be ran multiple times so we should skip the requirements that are previously satisfied.  The lengthy install of ```numpy``` comes to mind...
 - Check the permissions on the individual scripts making sure that they are set correctly and can be executed.  Incorrect permissions on the scripts causes premature install failures.  Perhaps we just manually pass the shell scripts to the shell: ```sudo /bin/sh mycodo/scripts/crontab.sh``` vs ```sudo mycodo/scripts/crontab.sh```
 - Group the files ran by the installer into their own package.  One location for all things used by the installer
 - Are there any others I'm missing that make the install cumbersome or less than easy?

zsole2, 2016-10-26:
Yeah, numpy is a pain... it takes 2 hours on my PiZero to install. And if there is new version, upgrade is the same...
I had been playing around to circumvent it a few weeks ago, but failed. Not much time, not much knowledge of the inner workings of Mycodo, not much experience in programming...
Anyway, in my tests, the check of requirements did not add too much time to the install, if a package is already installed. Even if an upgrade is bearable, except for numpy... 
Kyle was mentioning earlier (I think when version 4 was being developed) that one of the goals was that modules to be loaded on demand. In this vein, I tried to offload  loading numpy only when it is needed. A diff file from my test system is attached,
[numpy_test.txt](https://github.com/kizniche/Mycodo/files/552384/numpy_test.txt)
I had to do other things, so after failing with this I moved on. I remember that one of my tries led to me to 
https://thraxil.org/users/anders/posts/2008/03/13/Subprocess-Hanging-PIPE-is-your-enemy/. It is quite old, from 2009, but comments indicate that even in late 2014 it was relevant. In any case, it may still be the problem: checking with top, the subprocess failed after about 15 seconds... 
Maybe you guys can use this...

kizniche, 2016-10-26:
I thought the behavior currently is to only install what's necessary.

from setup.sh:
```apt-get install -y libav-tools libffi-dev libi2c-dev python-dev python-setuptools python-smbus sqlite3```
```pip install -r requirements.txt --upgrade```
```dpkg -i $INSTALL_DIRECTORY/influxdb_1.0.0_armhf.deb```

These commands *should* only be installing packages if there are updates or if they aren't already installed (and up-to-date).

closed 2018-02-26 by kizniche