kizniche/Mycodo #133: UI Enhancement: responsive display

montreuilg, 2016-10-26:
I just installed an 800x480 display on the RPI3 running Mycodo, and the graph is hardly visible because Chromium itself takes almost a third of the screen, and the menu do not quite fit on smaller screens.

As a developper (...) i do not really see a problem because i access the server through vnc or ssh, but for a production installation, small screens or even phones will be used to check upon Mycodo?

etiology, 2016-10-26:
Yeah I think the ```.navbar-collapse``` class is set to 768px or less to collapse the nav.  Not sure what screen sizes are the target behaviors but narrowing the window a small amount will trigger it to collapse.

could play with some custom css rules

montreuilg, 2016-10-26:
A kiosk mode would be nice, no Chromium menus, Mycodo screen only.

montreuilg, 2016-10-26:
You are right, there is a point at which the menu bar changes to small action list button (i do not know how to name that icon). Thanks for pointing it out.

kizniche, 2016-10-26:
What I did for my 1024x600 touchscreen display was install a kiosk plugin that starts the browser in fullscreen mode and directs to the graph page. You should also be able to zoom out to get the nav menu to fit properly. I also enabled auto login to the GUI and a script to be executed after that starts the browser. So, in the event of a reboot or power failure, it always returns to the graph page showing my main top graph. I can get the name of the exact plugin when I get home.

kizniche, 2016-10-26:
Alternatively, if your display won't work well with zooming out, you can zoom in to activate the third style that's used on small devices, that just shows the daemon status and an icon for the menu drop down.

montreuilg, 2016-10-26:
I was going to post this mockup for a ui proposition, before you described this behavior. Thanks!

https://cag.zone/Mycodo.html

_
gabriel

Reply: kizniche/Mycodo <reply@reply.github.com>
Date: 26 October 2016 at 01:33:19
To: kizniche/Mycodo <mycodo@noreply.github.com>
Cc: Gabriel Montreuil <montreuilg@sympatico.ca>, Author <author@noreply.github.com>

kizniche, 2016-10-26:
The magic of Bootstrap! Check out the screenshots of the early Mycodo version UIs and you'll see I'm no CSS wizard.

closed 2016-11-25 by kizniche