kizniche/Mycodo #136: LCD Error

Ryobidelacroix, 2016-10-29:
Hey Kyle!

Once again sorry for bombarding you with messages! I've been having an issue with my 20x4 LCD display. As soon as the display activates, it will immediately deactivate itself. 

Mycodo shows the LCD as 'Active' however, when trying to deactivate I receive an error stating: **Cannot deactivate LCD controller with ID E1RwgFHJ: It's not active.**

I have copied the error log, below: 

2016-10-29 12:17:20,748 - INFO - [Daemon] Anonymous statistics enabled
2016-10-29 12:17:20,913 - INFO - [Relay] Relays Initialized
2016-10-29 12:17:20,915 - INFO - [Relay] Relay controller activated in 37.4 ms
2016-10-29 12:17:20,915 - INFO - [Daemon] All activated timer controllers started
2016-10-29 12:17:21,023 - INFO - [Sensor aZneaaif] Activated in 76.9 ms
2016-10-29 12:17:21,123 - INFO - [Sensor 9uzBHFdC] Activated in 68.7 ms
2016-10-29 12:17:21,357 - INFO - [Sensor Lv82ZBPs] Activated in 122.0 ms
2016-10-29 12:17:21,363 - INFO - [Daemon] All activated sensor controllers started
2016-10-29 12:17:21,364 - INFO - [Daemon] All activated log controllers started
2016-10-29 12:17:21,364 - INFO - [Daemon] All activated PID controllers started
2016-10-29 12:17:21,650 - INFO - [LCD E1RwgFHJ] Activated in 262.1 ms
2016-10-29 12:17:21,652 - INFO - [Daemon] All activated LCD controllers started
2016-10-29 12:17:21,653 - INFO - [Daemon] Mycodo v4.0.24 started in 1.219 seconds
2016-10-29 12:17:21,658 - INFO - [Daemon] 30.684 MB ram in use
2016-10-29 12:17:26,655 - ERROR - [LCD E1RwgFHJ] Exception: 'id'
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/controller_lcd.py", line 269, in run
    self.get_lcd_strings()
  File "/home/pi/Mycodo/mycodo/controller_lcd.py", line 309, in get_lcd_strings
    if self.lcd_line[i]['id']:
KeyError: 'id'
2016-10-29 12:17:26,887 - INFO - [LCD E1RwgFHJ] Deactivated in 1477705646887.5 ms

I have attempted a fresh install, however no change. 

Thanks for your great work. I'm looking froward to seeing how far this project can go :)

kizniche, 2016-10-29:
Would you be able to give me a Mycodo admin login and ssh login to your Pi to diagnose the issue? You can contact me at my contact form at kylegabriel.com so we can talk privately.

kizniche, 2016-10-29:
I think I figured out the offending code. Expect a fix to be pushed to github shortly!

closed 2016-10-30 by kizniche