kizniche/Mycodo #137: Celsius and Fahrenheit toggle in settings.

MisterMeeseeks, 2016-10-30:
I would really appreciate a toggle for metric and imperial temperatures in the settings. I could keep converting the numbers, but it would save a lot of time just to see them that way on the graphs.

Thank you very much for all your incredible work on this great program.

kizniche, 2016-10-30:
I've had this feature requested by a few people. One of the reasons it's a low priority is that it primarily is something those in the US desire, and is a nonstandard set of units (opposed to SI). Currently, we're working on structural changes within Mycodo and I'm hesitant to begin adding a feature as involved and far-reaching as this before we get our new system implemented, or I at least get other higher-priority changes completed first.

MisterMeeseeks, 2016-11-23:
Just a followup... any way to add it (easily) to the live page, at least in parenthesis like the graph?

etiology, 2016-11-23:
I think a quick way to do this without involving a toggle button or settings is that we show Fahrenheit values in parentheses.  Similar to component data sheets showing primary units of mm and (inches)

This could be handled in Jinja2 template by calling a python function to transform the value so that we keep the presentation details in the presentation area.  We could write a [python function and call it from the jinja2 template](https://stackoverflow.com/questions/6036082/call-a-python-function-from-jinja2)

etiology, 2016-11-23:
Also javascript

kizniche, 2016-11-23:
I figured it would be easier to accomplish this through javascript than a function in Flask, at least for now, I didn't want to complicate the data response from Flask to the browser with more data about converted values, until I've thought about this more and can figure out the best way to implement this system. Ideally, I would like to be able to provide a way to enable the selection from a common set of units for each condition, for display purposes. This would require first setting what unit the value is returned from the sensor, then selecting what value should be displayed.

closed 2017-10-24 by kizniche

frodus17, 2018-02-21:
Any chance you could put F in parenthesis on the Dashboard widgets as well? I'm trying to get more used to measurements in C, but this would really help.

kizniche, 2018-02-21:
I'll look into whether there's a good way to add the F conversion to the Output, Measurement, and PID Dash elements. It takes up a lot of room, so if I can't make it subtle, it may not work out.

frodus17, 2018-02-21:
Otherwise, maybe have a C output sensor and an F output sensor?

dylandn, 2019-12-13:
Have it so when temperature conversion is desired, it's enabled within the Data input setting (as it is now). When enabled, create a new channel within the input that is the converted value?

So a temperature probe would have:
[CH0] Temperature C
[CH1] Temprature F