kizniche/Mycodo #140: Potential issue with 'before_blueprint_request'

etiology, 2016-11-01:
#### Potential for unexpected behavior from this area.  

The [```before_blueprint_request```](https://github.com/kizniche/Mycodo/blob/7d0edced7f2776288638de7522691012fe4fd0c0/mycodo/mycodo_flask/general_routes.py#L115) needs to be updated to [```before_app_request```](http://flask.pocoo.org/docs/0.11/api/#flask.Blueprint.before_app_request) because the former is only called for endpoints in that blueprint.

Also [this database check](https://github.com/kizniche/Mycodo/blob/7d0edced7f2776288638de7522691012fe4fd0c0/mycodo/mycodo_flask/general_routes.py#L107-L109) is now redundant because the application factory already handles the database creation before the app starts:
```python
def before_blueprint_request():
    """
    Ensure databases exist and at least one user is in the user database.
    """
    ca = current_app
    if (not os.path.isfile(current_app.config['SQL_DATABASE_MYCODO']) or
        not os.path.isfile(current_app.config['SQL_DATABASE_USER'])):
        return "Error: Cannot find databases. Run \"init_databases.py --install_db all\" to generate them."
    with session_scope(current_app.config['USER_DB_PATH']) as new_session:
        user = new_session.query(Users).filter(
            Users.user_restriction == 'admin').first()
        if user == None:
            return "Error: No admin user found. Run \"sudo ~/Mycodo/init_databases.py --addadmin\" to add one"
blueprint.before_request(before_blueprint_request)
```

etiology, 2016-11-01:
The problem I see with using the ```before_app_request``` is that it would cause an infinite redirect since even going to the create admin page would redirect to the create admin page

kizniche, 2016-11-01:
The following could be moved from before_blueprint_request() to home()

```python
    if not admin_exists():
        return redirect(url_for("authentication.create_admin"))
```

commit 7182ca4b (2016-11-01): Merge pull request #141 from etiology/before_blueprint_request_cleanup

closes #140.  Removed redundant db check
commit 722b28f5 (2016-11-01): closes #140.  Removed redundant db check
closed 2016-11-23 by kizniche