kizniche/Mycodo #145: Almost every sensor implement a different read() return value

etiology, 2016-11-03:
All of the sensors appear to have a ```read()``` method and ```stopSensor()``` (incorrect camel case name), but with respect to the ```read()``` method they return different values.  Some return ```None``` others return ```str```, or ```int``` values, and others return either an ```int``` or ```None```.

This needs to be standardized.  We could use a parent class to ensure that these methods are present and overwritten in each sensor but we need some guides on what the return values should be.

kizniche, 2016-11-03:
stopSensor() isn't implemented, and I'm not sure it's even necessary. It was a part of the template that @Cabalist used when he made the first sensor class, and I just duplicated that template without ever using or removing it.

For the most part, most sensors should return floats. Rarely will it be an int (altitude) because a decimal is unnecessary.

I agree we need standardization, or at least better documentation what the return value is at every stage it's manipulated in the class.

etiology, 2016-11-03:
Ok.  It might be worth creating a parent class that all sensors inherit from.  It could both document the structure and ensure that certain methods are overwritten

```python
class AbstractSensor(object):
    """ 
    Base class for all sensors.  

    All sensors need to be created from the base class and should overwrite all of the provided methods.

    Maybe show some examples or explain more about how sensors are called and used.
    """"
   def read():
        """
        A very good description of what the read() method does
        and what it is required to return

        :returns: int, None  
        """"
        raises NotImplementedError  # overwrite is required in the child class

```

etiology, 2016-11-03:
What about [this part](https://github.com/kizniche/Mycodo/blob/57e32b4f737266ee6e1c955cdec1e5b1ac70c12e/mycodo/sensors/raspi.py#L38-L42)?
```python
  def __iter__(self):
        """
        Support the iterator protocol.
        """
        return self
```

This isn't write.  The ```__iter__``` special method should yield a generator but here it's returning a non-iterator.  From what I can tell, the only reason it's in here is because the main method in this module using izip() to create an continual sensor update loop.  Since the series is infinite, the whole for loop thing with izip could be replaced with a while(True).

Also the ```next()``` method is basically acting like a ```__str__``` method.

So the question I have is do you want the object behavior to use the ```__iter__``` or the ```cpu['temperature']``` (which would require implementing the ```__getitem__``` special method) or should it just be:
```python
class RaspberryPiCPUTemp(object):
    def __init__(self):
        self._temperature = None
        self.read()  # update the temperature

    def __repr__(self):
        """  Representation of object """
        return "<RaspberryPiCPU(temperature={temp})>".format(self._temperature())
 
    def __str__(self):
        """ Return temperature information """
        return "temperature: {value}".format(float("{0:.2f}".format(self.temperature))

    def read(self):
        try:
            self._temperature = self.get_measurement()
        except IOError:
            return 1

    @staticmethod
    def get_measurement():
        with open('/sys/class/thermal/thermal_zone0/temp') as cpu_temp_file:
            raw_t = cpu_temp_file.read()
        return float(raw_t) / 1000

    @property
    def temperature(self):
        return self._temperature
```

etiology, 2016-11-03:
There is some typos in that code above and I have a commit with tests for this new version if you think it's ok

kizniche, 2016-11-03:
That looks good to me.

etiology, 2016-11-03:
ok i'll post something here shortly

etiology, 2016-11-03:
Oops.  We need next.  It [gets called](https://github.com/kizniche/Mycodo/blob/a390496f5f0e129ae94fcb7f6d862876f4cb7f2c/mycodo/controller_sensor.py#L618)

I'll add it back

kizniche, 2016-11-04:
With the above code, there can be a situation that there is a proper reading that sets temperature, then the next reading may fail, yet still return a temperature. There should be a check every read that there was an actual value read from the sensor.

etiology, 2016-11-04:
Here is my current one.  I'm still writing all the tests though.  I want 100 coverage before I post it:
```python
class RaspberryPiCPUTemp(object):
    """ A sensor support class that monitors the raspberry pi's cpu temperature """

    def __init__(self):
        self._temperature = 0

    def __repr__(self):
        """  Representation of object """
        return "<{cls}(temperature={temp})>".format(cls=type(self).__name__, temp="{0:.2f}".format(self._temperature))

    def __str__(self):
        """ Return temperature information """
        return "temperature: {}".format("{0:.2f}".format(self._temperature))

    def __iter__(self):  # must return an iterator
        """ RaspberryPiCPUTemp iterates through live temperature readings """
        return self

    def next(self):
        """ Get next temperature reading """
        if self.read():  # raised an error
            raise StopIteration  # required
        return dict(temperature=float('{0:.2f}'.format(self._temperature)))

    @staticmethod
    def get_measurement():
        """ Gets the Raspberry pi's temperature in celsius by reading the temp file and div by 1000"""
        with open('/sys/class/thermal/thermal_zone0/temp') as cpu_temp_file:
            return float(cpu_temp_file.read()) / 1000

    @property
    def temperature(self):
        """ CPU temperature in celsius """
        return self._temperature

    def read(self):
        """
        Takes a reading from the CPU and updates the self._temperature value

        :returns: None on success or 1 on error
        """
        try:
            self._temperature = self.get_measurement()
        except IOError as e:
            logger.error("CPU temperature reading returned IOError: {err}".format(err=e))
            return 1
```

etiology, 2016-11-04:
right now I'm just logging errors on the read(), and any error beyond an IOError gets propagated up.  I think i'm going to catch it and log it.  Unless crashing the daemon thread is the correct behavior

kizniche, 2016-11-04:
I'd rather it just log the event (occurs in controller_sensor.py) instead of stopping the thread. I've seen a lot of sensors eventually cause an error, from bad wiring, faulty hardware, stray neutrons. Most of the time it's infrequent and should be ignored.

etiology, 2016-11-04:
Ok i'll make sure it gets logged then

kizniche, 2016-11-04:
A lot of people that use the system are inexperienced with sensors, so knowing when the sensor isn't working and maybe why is important.

PR #146 (2016-11-04): codacy_grade_improvement_raspi_sensor
closes #145

First rework of the sensor classes starting with the RaspberryPiCPUTemp and RaspberryPiGPUTemp classes.  Created AbstractSensor Class to enforce consistancy and provide a place to document the Sensor class methods.
closed 2016-12-03 by kizniche