kizniche/Mycodo #149: I'm Back!

kizniche, 2016-11-16:
Not an issue, but I just wanted to say that my lack of activity for the past few weeks has payed off. I successfully defended my PhD dissertation this morning and became Dr. Gabriel! My presentation title slide is below.

[image]

etiology, 2016-11-16:
# Congratulations!!

Can you link your paper?  I'd love to read it and pay walls are annoying

kizniche, 2016-11-16:
Thank you! I would, but there's intellectual property and a current patent application discussed in it, so I need to first check if my committee wants to put an embargo on it. If not, I'll upload it. I'll find out soon.

etiology, 2016-11-16:
No worries.  When it's time though I'd love to read it.  A link to the published work would be fine.  That way you're in the clear

berserktron3k, 2016-11-16:
Congrats!

Brady Hummel

Myke974, 2016-11-16:
Congratulations ! ! Dr. Gabriel ! ! ! :D Respect !

zsole2, 2016-11-17:
Congratulations!
Be careful what you tell us even here... Being a patent attorney,  I could tell you a few stories about accidentally disclosing an invention before filing a well written patent application. It could be a major headache later...

kizniche, 2016-11-17:
This is our current PCT application, so I'll link this before I find out if I should link my dissertation. I'm also performing minor revisions on the dissertation, so I haven't submitted the final copy yet.

[Patent Application PCT/US2015/044807](https://www.google.com/patents/WO2016025572A1?cl=en&dq=ininventor:%22Kyle+T.+GABRIEL%22&hl=en&sa=X&ved=0ahUKEwjbz6H91bDQAhVW1WMKHdeJA3YQ6AEIHTAA)

zsole2, 2016-11-18:
OK, if you have the PCT application already published, then you're safe. It is certainly interesting stuff, although the search opinion is not too good. It is always tough to get patents in computer implemented inventions, even in the US it is getting harder recently.

kizniche, 2016-12-03:
My committee wants to put a 2-year embargo on my dissertation, so unfortunately I can't publicize it until that expires.

closed 2017-01-15 by kizniche