kizniche/Mycodo #150: Create an Issues Template

etiology, 2016-11-17:
GitHub has a  template that can appear whenever a new issue is created.  It can be a nice way to standardize basic details about the project version, problem description, and how to reproduce the errors.  See the [Helping people contribute to your project](https://help.github.com/articles/helping-people-contribute-to-your-project/) page for other templates

kizniche, 2016-11-17:
That's a good idea. I don't see the template when I try to create an issue, but I suppose that's normal for the repo owner.

etiology, 2016-11-17:
That's strange actually.  You should see it when making issues.  I'll look into it

etiology, 2016-11-17:
The pull request was to the dev branch.  I guess it will wait until it's merged with master.

I'll close this for now

closed 2016-11-17 by etiology

kizniche, 2016-11-17:
I started issue #148 to discuss the merging of mycodo_dev. Do you think it's ready?

commit f247aa0a (2016-11-17): closes #150.  Created a GitHub issue template