kizniche/Mycodo #152: tring to reinstall

ltore123, 2016-11-18:
Enabling mycodo startup script
Failed to execute operation: File exists

kizniche, 2016-11-18:
Can you please provide a little more detail? I'm not exactly sure what this means. Here's a nice template that can be used:

## Mycodo Issue Report:
  - Specific Mycodo Version:
  - git commit hash:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

etiology, 2016-11-18:
>Enabling mycodo startup script
Failed to execute operation: File exists

Does this mean running the ```setup.py``` file twice?

kizniche, 2016-11-18:
It depends when the setup.sh script exited. The part where it says aborted can't indicate the issue, but what happened before it can. Can you please provide the entire setup.log found in the Mycodo directory?

ltore123, 2016-11-18:
there is no setup.log in the Mycodo diroctory 

writing new private key to 'privkey.pem'
-----
Generating RSA private key, 1024 bit long modulus
................................................++++++
.................................++++++
e is 65537 (0x10001)
Signature ok
subject=/C=US/ST=Georgia/L=Atlanta/O=mycodo/OU=mycodo/CN=mycodo
Getting CA Private Key
#### Enabling mycodo startup script
Failed to execute operation: File exists

kizniche, 2016-11-18:
Well, you mentioned an issue was displaying on a 20x4 LCD. Reinstalling by rerunning the setup.sh script was never an intended method to reinstall the system. If you want to reinstall, I would suggest recloning Mycodo and running the setup.sh script fresh.

However, it sounds like this issue should really be about your LCD issue. What is the hardware you're using? Have you tried executing the [LCD test script](https://github.com/kizniche/Mycodo/blob/master/mycodo/tests/manual_tests/test_i2c_LCD.py)?

etiology, 2016-11-18:
Could be in [this section of the setup.sh file](https://github.com/kizniche/Mycodo/blob/1550c5ce0e1b6e47a4a1afa2d98b5220fc6abc7b/setup.sh#L93-L101)

```bash

openssl req \
    -new \
    -x509 \
    -sha512 \
    -days 365 \
    -nodes \
    -out cert.pem \
    -keyout privkey.pem\
    -subj "/C=US/ST=Georgia/L=Atlanta/O=mycodo/OU=mycodo/CN=mycodo"
```

kizniche, 2016-11-18:
@etiology, It's an issue with systemctl. I believe disabling mycodo.service first, with ```sudo systemctl disable mycodo/service``` should fix the issue that's happening in the setup.sh

ltore123, 2016-11-18:
sudo systemctl disable mycodo/service
Failed to execute operation: Invalid argument

kizniche, 2016-11-18:
Oops, the / is supposed to be a .

ltore123, 2016-11-18:
sudo systemctl disable mycodo service
Removed symlink /etc/systemd/system/multi-user.target.wants/mycodo.service.
Removed symlink /etc/systemd/system/mycodo.service.

ltore123, 2016-11-18:
my lcd    https://tr.aliexpress.com/item/IIC-I2C-2004-blue-screen-LCD-module-provides-libraries-For-Arduino/1968055629.html?spm=2114.13010608.0.0.pzUylV
address for this lcd is not 0X27 it is 0X3F

ltore123, 2016-11-18:
May be you shoul add this address to your database

kizniche, 2016-11-18:
Strangely, this has already been done with commit 77e581f

kizniche, 2016-11-19:
But it looks like it was taken out in commit 46df9a2, [here](https://github.com/kizniche/Mycodo/commit/46df9a2ca2f1b6a01c317fd23cb7a6404cb71bc4#diff-d951bd74b59b1bd02ebe047b8d3e35eaL103-L110)

ltore123, 2016-11-19:
ok setup finished succesfully after sudo systemctl disable mycodo service command but getting
403 Forbidden

You don't have permission to access /live on this server. error

kizniche, 2016-11-19:
Reinstallation was not necessary for the error you were having the the LCD. Most likely attempting the reinstall without disabling the web server and shutting down the daemon, etc. has caused these new issues. It may be more difficult to get your system working than to just recommend you go through the install instructions again, but refrain from trying to reinstall. I'll push an update shortly for the LCD address options to restore it to what it was with commit 77e581f

kizniche, 2016-11-19:
However, I would always recommend a restart with all the systemctl changes that have been going on your system, and see if you still get the 404 error

etiology, 2016-11-19:
Maybe we should add new use case behavior to the install script to address these types of things.  Require behaviors like:
  - Create a setup.sh log file that each step appended to
  - Shutdown the webservice (if needed)
  - Ensure that script permissions are set to executable
  - etc

kizniche, 2016-11-19:
@etiology There is a directive to create a setup.log. My systems create it. The other points are good.

etiology, 2016-11-19:
@kizniche Good to know.  I couldn't remember

ltore123, 2016-11-19:
I am really sorry ıam pain in the neck but ı am trying use this program for an egg incubator
I am not getting 404 getting 403 forbidden error ıam sure ı can solve this problem by nyself thankyou so much for your help whenever ı get spare time ı would like to translate Mycodo to my native language if you give me permission to do it

kizniche, 2016-11-19:
@ltore123, You're not a pain in the neck. These address options should have been there, but somehow I reverted a previous commit I made that added them. You also have shown the setup.sh script could use a little TLC.

kizniche, 2016-11-19:
With the commit I just pushed, you should be able to select your address. For now, I would recommend reinstalling raspbian. That will probably be quicker than trying to diagnose what happened to your first install. If you ever need to update to the latest version, you can use the Admin/upgrade menu from within Mycodo.

ltore123, 2016-11-19:
Ok I will do that thanks again

etiology, 2016-11-19:
@ltore123 posting issues is really helpful.  When issues get addressed it makes the project stronger. 

The ```403``` is interesting though.  It's almost like your user isn't logged in.  Have you tried using a private browser mode?  It will make sure all of the browser cache and stuff is clean.  If you get this again please post more info.

kizniche, 2016-11-19:
It could also be cookie/session data from a previous instance of apache/Mycodo screwing with the authentication.

kizniche, 2016-11-19:
Could try going to https://yourMycodoAddress/logout to attempt to log out

ltore123, 2016-11-19:
https://localhost/logout
Forbidden

You don't have permission to access /logout on this server.
Apache/2.4.10 (Raspbian) Server at localhost Port 443

kizniche, 2016-11-19:
Haha! Probably best to just reinstall Raspbian.

ltore123, 2016-11-19:
Symbolic link not allowed or link target not accessible: /var/www/mycodo

ltore123, 2016-11-19:
apache2 error log 
Symbolic link not allowed or link target not accessible: /var/www/mycodo

ltore123, 2016-11-19:
SOLVED ı got ıt working ı deleted everything under /root/.local/share/Trash/files diroctory amd reinstalled Mycodo
Thanks again

ltore123, 2016-11-19:
sorry to bother you again one more question ı have seen you had added lcd address 0X3F
do ı have enter any value for Multiplexer I2C Address and Multiplexer Channel for this type of lcd ?

ltore123, 2016-11-19:
do ı have to enter any value for Multiplexer I2C Address and Multiplexer Channel for this type of lcd ?

ltore123, 2016-11-19:
ı am not using Multiplexer I2C

kizniche, 2016-11-19:
Glad to hear it's working.

Leave the multiplexer options blank if you're not using an I2C multiplexer.

ltore123, 2016-11-19:
today something strange happening    i2cdetect -y 1  do not detect anything

i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

ltore123, 2016-11-19:
yesterday it was showing the address 0X3F

kizniche, 2016-11-19:
It could be a hardware, wiring, or software issue. Since it was working earlier, I'd guess it's your hardware/wiring.

closed 2017-01-15 by kizniche