kizniche/Mycodo #153: Apache Server Error 500

MisterMeeseeks, 2016-11-23:
I've been having an intermittent problem with Apache returning only "Internal Server Error" when trying to access the webpage. Best I can tell, the mycodo server is running correctly (e.g. still measuring and storing data points), but the web page is inaccessable.

tail -f /var/log/apache2/error.log

[Tue Nov 22 23:19:19.584280 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950] Traceback (most recent call last):
[Tue Nov 22 23:19:19.587190 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]   File "/var/www/mycodo/mycodo_flask.wsgi", line 5, in <module>
[Tue Nov 22 23:19:19.588374 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]     from mycodo.start_flask_ui import app as application
[Tue Nov 22 23:19:19.591087 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]   File "/var/www/mycodo/mycodo/start_flask_ui.py", line 10, in <module>
[Tue Nov 22 23:19:19.592199 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]     from mycodo.mycodo_flask.app import create_app
[Tue Nov 22 23:19:19.594798 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]   File "/var/www/mycodo/mycodo/mycodo_flask/app.py", line 29, in <module>
[Tue Nov 22 23:19:19.596742 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]     from flask import Flask
[Tue Nov 22 23:19:19.597501 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]   File "/usr/local/lib/python2.7/dist-packages/flask/__init__.py", line 21, in <module>
[Tue Nov 22 23:19:19.598292 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950]     from .app import Flask, Request, Response
[Tue Nov 22 23:19:19.600129 2016] [wsgi:error] [pid 26316:tid 3028775984] [remote 192.168.1.100:32950] ValueError: bad marshal data (string ref out of range)
[Tue Nov 22 23:44:48.339811 2016] [wsgi:error] [pid 26316:tid 3003610160] [remote 192.168.1.100:32950] mod_wsgi (pid=26316): Target WSGI script '/var/www/mycodo/mycodo_flask.wsgi' cannot be loaded as Python module.
[Tue Nov 22 23:44:48.344865 2016] [wsgi:error] [pid 26316:tid 3003610160] [remote 192.168.1.100:32950] mod_wsgi (pid=26316): Exception occurred processing WSGI script '/var/www/mycodo/mycodo_flask.wsgi'.

That shows up every time I attempt to access the page. I've tried restarting mycodo, apache, and the entire system several times in several combinations, but to no avail. I also ran the update script and am at the latest build. I don't particularly care too much about losing the data from before, but I'd like to be able to access it going forward.

Any ideas? (Thanks)

etiology, 2016-11-23:
[Maybe this?](https://stackoverflow.com/questions/30861493/how-to-fix-python-valueerrorbad-marshal-data)

etiology, 2016-11-23:
If that doesn't work then try using *pip* to uninstall flask and reinstall it.

MisterMeeseeks, 2016-11-23:
I tried doing both... now the mycodo daemon won't start...

./mycodo_daemon.py

gives me:

Traceback (most recent call last):
  File "./mycodo_daemon.py", line 37, in <module>
    import rpyc
  File "/usr/local/lib/python2.7/dist-packages/rpyc/__init__.py", line 44, in <module>
    from rpyc.core import (SocketStream, TunneledSocketStream, PipeStream, Channel,
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/__init__.py", line 1, in <module>
    from rpyc.core.stream import SocketStream, TunneledSocketStream, PipeStream
  File "/usr/local/lib/python2.7/dist-packages/rpyc/core/stream.py", line 11, in <module>
    from rpyc.lib.compat import select, select_error, BYTES_LITERAL, get_exc_errno, maxint
ValueError: bad marshal data (unknown type code)

kizniche, 2016-11-23:
The daemon is completely separate from the Flask UI. The error you're now seeing from the daemon is indicating corruption with the rpyc Python module. Try reinstalling rpyc with pip.

kizniche, 2016-11-23:
Also, you can backup all sensor data with the following commands (metastore and database):

```
influxd backup <path-to-metastore-backup>
influxd backup -database mycodo_db <path-to-database-backup>
```

Then the metastore and database may be restored with the following commands:

```
influxd restore -metadir /var/lib/influxdb/meta <path-to-metastore-backup>
influxd restore -datadir /var/lib/influxdb/data <path-to-database-backup>
```

zsole2, 2016-11-23:
That info is very useful! 
 Is it only the data or the settings, too?  Can we use this to copy over the full setup to a new install, even onto another RasPi? And would it cause any problem later on when the remote admin feature will be fully functional? I mean it looks that the unique IDs could be copied over...

kizniche, 2016-11-23:
If you wanted to move an entire Mycodo install with the data and settings, where it would be unrecognizable from the previous, this is what I would do:

Backup the influx database (above) and the Mycodo databases (*.db files located in Mycodo/databases).

Then, on the new system, assign the same IP as the old system (if using the remote admin), restore the influx databases (above), clone the latest Mycodo from github, copy the *.db files back to Mycodo/databases, then, to ensure your databases are up-to-date and compatible with the latest version of Mycodo, run:

```bash
cd ~/Mycodo/databases
alembic upgrade head
```

Finally, execute the Mycodo setup.sh script. The order of these events are important, because databases will be created and services started when the setup.sh script is executed, so certain commands need to be done before this happens.

zsole2, 2016-11-23:
Thanks, I will certainly try with my next rebuild... which will not come very soon hopefully...

MisterMeeseeks, 2016-11-23:
Well, after reinstalling rpyc, I get this:

./mycodo_daemon.py

Traceback (most recent call last):
  File "./mycodo_daemon.py", line 44, in <module>
    from utils.statistics import add_update_csv
  File "/home/pi/Mycodo/mycodo/utils/statistics.py", line 14, in <module>
    from sqlalchemy import func
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/__init__.py", line 9, in <module>
    from .sql import (
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/sql/__init__.py", line 8, in <module>
    from .expression import (
  File "/usr/local/lib/python2.7/dist-packages/sqlalchemy/sql/expression.py", line 53, in <module>
    from .dml import Insert, Update, Delete, UpdateBase, ValuesBase
EOFError: EOF read where object expected

I'm beginning to think the whole installation got borked somehow. Any more ideas before I just wipe it all out and reinstall?

kizniche, 2016-11-23:
Not sure. Could be an issue with the Python module sqlalchemy. Reinstalling may fix it.

etiology, 2016-11-23:
I'm beginning to think the file system is corrupt on the drive, or the flash drive itself is corrupted. 

That's the third file that is either was corrupted or became corrupted since the first. I don't think it would be worth the effort to [force a file system repair](https://www.cyberciti.biz/faq/linux-force-fsck-on-the-next-reboot-or-boot-sequence/) because reimagine the drive is not too bad. It's easier if you made a restore image after getting the system set up prior to installing Mycoco. If you have to rebuild from scratch then take a snapshot of it once you get everything set up.

kizniche, 2016-11-23:
Are these errors during test preventing the tests from completing?

```ERROR:mycodo.mycodo_flask.general_routes:URL for 'inject_mycodo_version' raised and error: Connection refused. Is the daemon running?```

kizniche, 2016-12-03:
I'm going to close this issue because it appears this is a widespread SD card corruption and not an actual issue with Mycodo. If any new information can be provided or if using anew SD card, or reformatting/installing the OS and Mycodo on the current SD card doesn't fix the issue, please reopen it.

closed 2016-12-03 by kizniche