kizniche/Mycodo #155: Start Stream Gives 500 Internal Server Error

berserktron3k, 2016-11-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.0.26
  - git commit hash: 2fd91bb
 
#### Problem Description
Please list:
When clicking Start Stream, I receive:

Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2016-11-25:
At the moment my camera is in use, so I can't test this right now. I'll be able to pause it soon, then I'll run some tests.

kizniche, 2016-12-03:
I didn't anticipate taking so long to test this, but I found the issue and pushed an update.

closed 2016-12-03 by kizniche