kizniche/Mycodo #159: SHT7x not work after update

wangine, 2016-12-02:
## Mycodo Issue Report:
  - Specific Mycodo Version:4,.0.26
  
#### Problem Description
After last update the SHT75 sensor stop working. 
 
### Errors
No any error when i try to activate/ desactivate the sensor , but sensor not respond , I try to change the sensor with SHT71 , same result . I try to force reinstall and the same problem . I think is update problem because the sensor is working on the same configuration on other PI board but that PI wasn't update 3 months ago. I didn't try a new fresh install , my computer broken ,and i didn't has time to fix it.
 
### Additional Notes
Measurement | Timestamp
0 °C (0.0 °F) | No Data Last 30m 
0 % | No Data Last 30m 
0 °C (0.0 °F) DP | No Data Last 30m

etiology, 2016-12-02:
@wangine I wouldn't bother reinstalling anything just yet.  Would you be able to run the sensor directly using the SHT1x7xSensor class?  I'd love to see more info about what's happening but I don't have that sensor to work with.  So with your sensor, here is how I would approach doing that:

1. Move into the ```Mycodo/mycodo``` folder.  Assuming that the *Mycodo* folder is in your home directory: ```cd ~/Mycodo/mycodo```
2. Start a python console and enter the following lines:
```python
from sensors.sht1x_7x import SHT1x7xSensor
from time import sleep

# Important: you must use your input pin number, clock pin number, and voltage value
# In this example I'm using input pin 4, clock pin 99 (because I have no idea), and a voltage of '5.0'
sensor = SHT1x7xSensor(pin=4, clock_pin=99, voltage='5.0')

# take reading with a 2 second sleep in between
for result in sensor:
    print("Dew Point".format(result.dew_point))
    print("humidity".format(result.humidity))
    print("temperature".format(result.temperature))
    print("-" * 10)  # prints a line to visually separate out each loop
    sleep(2)
```

Let me know if you run into any issues with the code above.  You'll need to put in the correct pin numbers.  For instance I have no idea what your clock pin is or your voltage.  A good idea is to use the same values you have in your configuration so that we are testing the correct setup.

etiology, 2016-12-02:
To start the python console you can just type ```python```

kizniche, 2016-12-03:
I just pushed an update (commit 6477798) that may fix this issue. See if it works after updating your system.

wangine, 2016-12-03:
Thank you Kyle , it's working now. The sensor work normal , have a great day.

kizniche, 2016-12-03:
@wangine, Great!

@etiology, sorry to jump in with that commit before getting a chance to see the console output, but I saw what I thought would be causing the error and wanted to commit it before I forgot about it.

etiology, 2016-12-03:
@kizniche damn you're fast!  Good job pushing the fix. I've never played with these sensors before so if you have the fix I'd say ship it.

kizniche, 2016-12-03:
@etiology, I've never used an SHT sensor. @wangine has been the one who's helped to bring support for the two SHT classes to Mycodo. I just happened to notice uses of self.temperature and self.humidity when they haven't been set yet :)

closed 2016-12-03 by kizniche