kizniche/Mycodo #160: PID Not Working

wasitim, 2016-12-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.0.26
  - git commit hash:
 
#### Problem Description
Unable to get PID to control temperature using DS18B20 sensor and relay board.

I have a Sainsmart relay board connected to GPIO17 to turn on a refrigerator for a cheese cave project. I have a the sensor set up in the sensors section and the relay set up in the relays section.

I can turn the relay on in the web app and the frig runs. I can set conditions on the relay and the conditions will control the relay and the frig. 

I set up a PID for finer control. 
- selected the correct sensor
- set the type to Temperature
- set direction to lower
- set period to 15 seconds
- set setpoint to 12 (current temp is 16)
- set PID to 1, 0, 0
- left integrator min and max to defaults
- didn't set anything for the raise values
- set lower relay id to the one mentioned above that works
- set lower min duration to 20 and max duration to 240
- didn't create any methods to track setpoint
 
### Errors
  - Nothing particular to list for errors
 
### Additional Notes
I had originally installed mycodo back in September and had been playing around with it for various tasks. As I mentioned, it works fine for simple relay control using conditionals but the temp over/undr shoot was kind of large and since I'm familiar with PID in general, I thought it a better option. The PID control wouldn't turn in that version, so I upgraded to 4.0.26. Upgrade went mostly well except for some issues with the camerstill table. Deleted the mycodo.db and re-ran the init python script and all was well.
What could I be missing?

[image]

[image]

kizniche, 2016-12-03:
The lower min duration is the minimum duration the relay can turn on for. The period is how often the PID controller will check the temperature (or latest measurement in database), apply the PID function, and turn the relay on for a duration. Having such a low period with such a high min duration may not work well. For a fridge, I would set the period to at least 60 seconds.

Also, your image shows a setpoint of 20 degrees C, yet you say the air is 16 C. If that is so, then the PID would never turn the cooler on because the temperature is already below your setpoint.

wasitim, 2016-12-03:
Yeah, that screenshot was a different attempt. Same results setpoint at 12. I've lowered the setpoint to 12, set the period to 60, and the lower min duration to 5 and still nothing.

I'm monitoring the relay page expecting to see the relay turn on. Would that be incorrect?

[image]

kizniche, 2016-12-03:
That's correct. You can also create a graph that displays the relay data so you don't have to monitor the relay page, you can just look at the graph to see if there has been any relay activation.

This is why your relay isn't activating:
Your PID settings are Kp=1, Ki=0, Kd=0
Essentially, you are only using the proportional gain, and with a temperature difference (error) of 4 and a Kp=1, this would equate to a 4*1=4 second relay duration. Since your min duration is 20 (or 5 in the above settings), your relay won't turn on, You would need an error of greater than 5 for it to turn on with thee PID gains.

What I would recommend is to increase your Ki gain. This will steadily raise the relay output in situations like you're experiencing, where the low relay duration (PID output) is not affecting the environment. Essentially, if the PID output continues to not affect the environment, the output will steadily start to rise until it does. Using the Kp gain alone won't allow this because your regulator is only a proportional controller.

wasitim, 2016-12-03:
Awesome!. I kicked the i value to .5 for now and I see the relay kick and the duration is increasing. I'll leave it there for a while until I get to some (hopefully) steady state. 

Great project and awesome work. I wasn't sure if this was the correct place to post an issue like this so if it's not, just let me know.

kizniche, 2016-12-03:
Great! And yes, post any and all issues here. This is the only forum for Mycodo, so issues, ideas, thoughts, etc. can all go here.

I would recommend using a large min relay duration. A fridge compressor isn't designed for rapid power cycling or even frequent power cycling. I would maybe set the min duration to something like 180 second and the period to something like 360 second.

kizniche, 2016-12-03:
To add to the above recommendation, You may want to try an even longer period than 360 seconds if using a min duration of 180 seconds. 180/360 = 0.5 = 50% duty cycle. This will be the lowest possible duty cycle with this configuration. If your steady state can be achieved with say a 10% duty cycle, then you would need a period of 180/0.1=1800 seconds (30 minutes). But, 30 minutes may be too long of a period. I would start smaller, say 5 minutes (300 seconds). With 300 seconds as the period, a 10% duty cycle would be 300*0.1=30 seconds.

closed 2016-12-03 by kizniche