kizniche/Mycodo #162: SHT1x sensor - no readings available

misudo, 2016-12-14:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.0.26
  - git commit hash: not sure where to find it (still learning)
 
#### Problem Description
Please list:
Clean install of Mycodo on my Rpi Zero, connected via USB. 
I am trying to connect [SHT1x](http://robotdyn.com/catalog/temperature/temperature_and_humidity_sensor_sht1x/) sensor, using GPIO 17 (wiring pin 11) for data and GPIO 21 (wiring pin 40) for Clock with 5v voltage. 
With all connected right, I set up the sensor in Mycodo, but get no value readings. I would appreciate any help. Thank you!
 
### Errors
2016-12-14 12:36:21,650 - ERROR - [Sensor 3EXzEpZI] Error while attempting to read sensor: 
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
    measurements = self.measure_sensor.next()
  File "/home/pi/Mycodo/mycodo/sensors/sht1x_7x.py", line 47, in next
    raise StopIteration  # required
StopIteration

### Steps to Reproduce the issue:
How can this issue be reproduced?
1. connect SHT1x sensor
2. set up sensor in 'Sensors' tab in Mycodo
3. no sensor readings available. 
 
### Additional Notes
When trying to get sensor reading from CLI, I twice get the following error:
pi@raspberrypi:~ $ sht -v -trd 21 17
Traceback (most recent call last):
  File "/usr/local/bin/sht", line 9, in <module>
    load_entry_point('sht-sensor==16.2.0', 'console_scripts', 'sht')()
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 264, in main
    sht = Sht(opts.pin_sck, opts.pin_data)
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 211, in __init__
    super(Sht, self).__init__(pin_sck, pin_data, **sht_comms_kws)
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 67, in __init__
    self._init()
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/sensor.py", line 72, in _init
    self.gpio.set_pin_value(pin, k='direction', v='low')
  File "/usr/local/lib/python2.7/dist-packages/sht_sensor/gpio.py", line 65, in set_pin_value
    with open(get_pin_path(n, k), 'wb', 0) as dst:
IOError: [Errno 13] Permission denied: '/sys/class/gpio/gpio17/direction'

However, the third and all subsequent times I get a good result, for example:
pi@raspberrypi:~ $ sht -v -trd 21 17
temperature: 22.96
rh: 52.2128920405
dew_point: 12.6366759655

etiology, 2016-12-15:
The part that catches my eye is the ```Permission denied```

mk-fg, 2016-12-15:
As linked by github above, indeed had similar issue [reported](https://github.com/mk-fg/sht-sensor/issues/3) and addressed [here](https://github.com/mk-fg/sht-sensor/commit/c27d283f) (recent commit from Nov 10th).

I'm not exactly sure if EPERM (Permission Error) is from the kernel's gpio module initializing sysfs knobs before they can be used (EPERM is a bit odd choice for that), or maybe permissions from some udev rules not being applied in time (don't think there are any for gpio knobs on ArchLinuxARM out of the box, but not sure about Raspbian, which seem to be used here).

@misudo
You have `with open(get_pin_path(n, k), 'wb', 0) as dst:` in the last backtrace, which indicates that you're using an older version of sht-sensor module, so simply updating it might resolve the issue, I think.

mk-fg, 2016-12-15:
Also, on a very loosely related note, I've recently bumped into another recent sht1x python wrapper module - https://github.com/drohm/pi-sht1x/
It seem to be a bit more full-featured than mine "sht-sensor" one with two key diffs aside from that: uses RPi.GPIO for gpio access, Python3-only.

Former might be a good feature to steer clear of potential issues like this one, as that module should be very well-maintained (better than gpio.py stub in sht-sensor).
And py3 thing might also be very desirable for any future development.

From what I can tell, code there is very similar to sht-sensor, so should be really easy to use as a drop-in replacement, and shouldn't cause any additional issues.

misudo, 2016-12-15:
Thank you, @mk-fg, the issue with permission has been resolved and I removed it from the "Additional Notes" section. Nevertheless, the problem with Sht-sensor readings still exists in Mycodo and is still visible in the logs:

> 2016-12-15 21:25:01,302 - ERROR - [Sensor cckWsTZq] Error while attempting to read sensor: 
> Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
>     measurements = self.measure_sensor.next()
>   File "/home/pi/Mycodo/mycodo/sensors/sht1x_7x.py", line 47, in next
>     raise StopIteration  # required
> StopIteration

Please, could you help?

mk-fg, 2016-12-15:
I think you should have message like `... raised an exception when taking a reading: ...` in the same log, immediately preceding `Error while attempting to read sensor: ...` that you've pasted above, which might be a bit more helpful.

misudo, 2016-12-15:
No, unfortunately it's just this, nothing else. 
I've just tried to verify Mycodo Daemon status by `$ sudo service mycodo service` and got the following reply

> mycodo: unrecognized service

Trying to launch the Daemon manually by 'sudo ~/Mycodo/mycodo/mycodo_daemon.py', I get the following reply 

> 28.8793690569
> Lock file present. Ensure the daemon isn't already running and delete /var/lock/mycodo.pid

I'll try to reinstall Mycodo anew and will report if anything changes.

kizniche, 2016-12-15:
> 2016-12-14 12:36:21,650 - ERROR - [Sensor 3EXzEpZI] Error while attempting to read sensor:

The error you are receiving indicates the Mycodo sensor module cannot communicate with the sensor.

> I am trying to connect SHT1x sensor, using GPIO 17 (wiring pin 11) for data and GPIO 21 (wiring pin 40) for Clock with 5v voltage.

You mention two different pins for each sensor connection but do not mention which you use to configure the sensor in Mycodo. All pins used in Mycodo should use BCM numbering.

> I've just tried to verify Mycodo Daemon status by ```sudo service mycodo service```

This should be ```sudo service mycodo status```

kizniche, 2016-12-15:
Note that in #159 we fixed the only known issue with the sensor and it has been confirmed to work.

kizniche, 2016-12-15:
I jumped the gun with my last message (I deleted it). Let me check out the code and see if I can see any glaring issues.

kizniche, 2016-12-15:
Okay, I was correct that in #159 the most recent issue with the SHT1x_7x sensor module was resolved. That was with the SHT75, however the SHT1x uses the same module. I'm not sure how to diagnose your issue.

misudo, 2016-12-15:
Thank you for your support, @kizniche. 
Here are my settings for the sensor: `GPIO 17, Clock Pin 21, SHT Input Voltage 5.0V`, I trust that's how it should be. 

Below is the latest Daemon log from Mycodo: 

> 2016-12-15 22:43:42,008 - INFO - [Daemon] All activated timer controllers started
> 2016-12-15 22:43:42,010 - INFO - [Daemon] All activated sensor controllers started
> 2016-12-15 22:43:42,011 - INFO - [Daemon] All activated log controllers started
> 2016-12-15 22:43:42,013 - INFO - [Relay] Relay controller activated in 165.6 ms
> 2016-12-15 22:43:42,015 - INFO - [Daemon] All activated PID controllers started
> 2016-12-15 22:43:42,017 - INFO - [Daemon] All activated LCD controllers started
> 2016-12-15 22:43:42,019 - INFO - [Daemon] Mycodo v4.0.26 started in 1.726 seconds
> 2016-12-15 22:43:42,035 - INFO - [Daemon] 30.068 MB ram in use
> 2016-12-15 22:55:15,502 - INFO - [Sensor esT0RAoL] Activated in 375.3 ms
> 2016-12-15 22:55:15,511 - ERROR - [Sensor esT0RAoL] Error while attempting to read sensor: 
> Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
>     measurements = self.measure_sensor.next()
>   File "/home/pi/Mycodo/mycodo/sensors/sht1x_7x.py", line 47, in next
>     raise StopIteration  # required
> StopIteration
> 2016-12-15 22:55:30,574 - ERROR - [Sensor esT0RAoL] Error while attempting to read sensor: 
> Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
>     measurements = self.measure_sensor.next()
>   File "/home/pi/Mycodo/mycodo/sensors/sht1x_7x.py", line 47, in next
>     raise StopIteration  # required
> StopIteration
> 2016-12-15 22:55:45,615 - ERROR - [Sensor esT0RAoL] Error while attempting to read sensor: 
> Traceback (most recent call last):
>   File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
>     measurements = self.measure_sensor.next()
>   File "/home/pi/Mycodo/mycodo/sensors/sht1x_7x.py", line 47, in next
>     raise StopIteration  # required
> StopIteration 
`sudo service mycodo service` still returns `mycodo: unrecognized service`.
I'll see if it's a Daemon issue as suggested in your post.

kizniche, 2016-12-15:
I don't believe it's a daemon issue. The daemon is running as expected and the errors are expected if it cannot communicate with the sensor. As in #159 the module is confirmed to work with the SHT75, therefore it must be an issue specific to the SHT1x.

kizniche, 2016-12-15:
> ```sudo service mycodo service``` still returns ```mycodo: unrecognized service```.

This indicates systemctl was not configured properly during the Mycodo install. My installs return:

```
kiz@tango:~/Mycodo$ sudo service mycodo status                                          12/15/16  5:57 PM
● mycodo.service - Mycodo server
   Loaded: loaded (/home/kiz/Mycodo/install/mycodo.service; enabled)
   Active: active (running) since Thu 2016-12-15 17:35:19 EST; 36min ago
  Process: 32648 ExecStart=/var/www/mycodo/mycodo/mycodo_daemon.py (code=exited, status=0/SUCCESS)
 Main PID: 32663 (mycodo_daemon.p)
   CGroup: /system.slice/mycodo.service
           └─32663 /usr/bin/python /var/www/mycodo/mycodo/mycodo_daemon.py

Dec 15 17:35:19 tango systemd[1]: Started Mycodo server.
Dec 15 17:35:19 tango Mycodod[32663]: Starting daemon.
```

misudo, 2016-12-16:
@kizniche, thanks again for your input. 
It looks like I've finally figured out a workaround. It looks like the SHT1x sensor does not work in Mycodo if 5v setting is chosen - even though the sensor is connected to 5v pin and functions as desired. I tested by `sht -vtrd 21 17` from CLI. 
Thus, changing the setting (in Mycodo interface) to 3v or 3.5v makes it all work. 
I am not sure if that's a Mycodo or SHT-sensor driver problem.
Hope this helps.

Myke974, 2016-12-16:
@misudo : "...changing the setting (in Mycodo interface) to 3v or 3.5v..."
Where do you find that ?

kizniche, 2016-12-16:
@misudo, Great work! I'm not sure of why this is an issue, but It's something related to the sht-sensor module and not Mycodo (as far as I can tell). @mk-fg, do you have any ideas as to why this could be? The 3, 3.5 and 5 are all [handled the same way](https://github.com/kizniche/Mycodo/blob/647779847a8d0bdd1c2233725f71de0c287cd85c/mycodo/sensors/sht1x_7x.py#L23) by Mycodo when being used with the sht module.

kizniche, 2016-12-16:
@Myke974, All SHT sensors have a voltage option.

Myke974, 2016-12-16:
@kizniche : as suspected ! :) thanks.

mk-fg, 2016-12-16:
> do you have any ideas as to why this could be? The 3, 3.5 and 5 are all handled the same way

Not really, just picked from the dict by whatever key is provided.
I suspect that `self.voltage = "{}V".format(voltage)` formatting might produce something like 5.000001 though, due to nature of floating point values - https://docs.python.org/3/tutorial/floatingpoint.html

Should be easy to figure out what exactly goes wrong there if exceptions were being logged though, not sure why there aren't lines from that logger.error call in read() func in the logs (also not sure why it's not logger.exception there - traceback would be even nicer).

Seeing how "logger" seem to be initialized as a global for the whole module there, wonder if maybe it's the issue with weird disable_existing_loggers=True defaults in python's logging configuration - e.g. https://docs.python.org/3/library/logging.config.html#logging.config.fileConfig

kizniche, 2016-12-16:
I just missed your post here and started an issue on your repo. I'll investigate the logging issue to see if we can get something more verbose. We just moved to a new sensor module format, so things may not be working as expected (logging-wise).

kizniche, 2016-12-16:
@mk-fg, Looking at the actual form ([here](https://github.com/kizniche/Mycodo/blob/647779847a8d0bdd1c2233725f71de0c287cd85c/mycodo/flaskforms.py#L1060) and [here](https://github.com/kizniche/Mycodo/blob/647779847a8d0bdd1c2233725f71de0c287cd85c/mycodo/mycodo_flask/templates/pages/sensor_options/SHT1x_7x.html#L19-L23)), the voltages are stored as strings, not int or float, so it is going in as "5V".

kizniche, 2016-12-16:
Ah, it is being stored as 5.0 instead of 5. This seems like it may be the issue.

kizniche, 2016-12-16:
However, @misudo mentions the 3-volt option (technically "3.0V") works, which still confuses me why "5.0V" won't work.

kizniche, 2016-12-16:
Reviewing your code, @mk-fg, there's no reason why 3.0V or 5.0V options in Mycodo would work. If you can confirm this, I'll push an update for Mycodo and @misudo can test the new options.

mk-fg, 2016-12-16:
Yeah, neither '3.0V' nor '5.0V' should work, I'd suspect that mention of former one working is just a mistake.

mk-fg, 2016-12-16:
I think I'll add enum for that value, to maybe avoid that kind of bugs with string constants in the module, as they inevitably crop up due to typos and poorly-documented interfaces over time.

kizniche, 2016-12-16:
Well, even more confusing is I created the database with the [value as a float](https://github.com/kizniche/Mycodo/blob/647779847a8d0bdd1c2233725f71de0c287cd85c/mycodo/databases/mycodo_db/models.py#L180).

Looks like I'll have to figure out how to perform a database upgrade to change the type and values to a string.

misudo, 2016-12-16:
You are right, Guys, about the volt setting. It works well on 3.5v setting while when switching to 3v, I could have seen the history reading from 3.5v setup. I'll be able to test whatever changes you implement. 
Btw, I'm using the latest version of the Sht-sensor driver.

mk-fg, 2016-12-16:
As of sht-sensor==16.12.0, recommended way would be to do `sht = Sht(21, 17, voltage=ShtVDDLevel.vdd_5v)` as the updated README suggests.

That way, you'd explicitly have to have some translation mapping like `{'5.0V': ShtVDDLevel.vdd_5v, ...}`, and if that enum value is mistyped or goes away at any point (it won't, purely hypothetical), interpreter will barf as soon as possible.
These values are glorified strings though, so should be fine to keep passing e.g. '3.5V' and other strings for compatibility with older module versions.

> Well, even more confusing is I created the database with the value as a float.

Quite surprised that SQLAlchemy even accepts stringified float into REAL column, though guess maybe you do the manual type-casting somewhere along the way.

kizniche, 2016-12-16:
Hmm. I was unable to figure out how to change the column type with alembic, but I thought of a clever way to strip the decimal if it's a whole number float value:

```python
if voltage.is_integer():
    self.voltage = "{}V".format(int(voltage))
else:
    self.voltage = "{}V".format(voltage)
```

Update incoming, @misudo, if you would like to test it.

kizniche, 2016-12-17:
@misudo, I didn't mean to close the issue but that's what happens when a commit with the "fixes #issue" format. If you could report back after you upgrade mycodo to say whether it fixed the issue, it would be appreciated. Thanks.

mk-fg, 2016-12-17:
> thought of a clever way to strip the decimal if it's a whole number float value:

I think it misses the point a bit, and quite misleading, as sht-sensor doesn't accept any random '{}V' value, just like 5 specific ones from the datasheet table.
I.e. imagine random person looking at that a year down the line - why the formatting, why voltage.is_integer(), etc...

Plus it's generally a bad idea to rely on floats having any exact value (e.g. 0.1 is actually 0.10...055-infinity) and formatting to something specific without e.g. `{:.3f}`, though python takes care of these things, mostly.

Imho more obvious and less error-prone solution would've been:

```python
from sht_sensor import ShtVDDLevel

sht_sensor_vdd_value = {
  2.5: ShtVDDLevel.vdd_2_5,
  3.0: ShtVDDLevel.vdd_3,
  3.5: ShtVDDLevel.vdd_3_5,
  4.0: ShtVDDLevel.vdd_4,
  5.0: ShtVDDLevel.vdd_5 }
...
self.voltage = sht_sensor_vdd_value[voltage]
```

But hey, whatever works.

misudo, 2016-12-17:
@kizniche, all volt settings (2.5v; 3v; 3.5v; 4v; 5v) for the SHT1x sensor in Mycodo return the right values from the sensor. Works like a charm - I'm happy you've fixed it now. Great work together with @mk-fg on this case.

kizniche, 2016-12-17:
@mk-fg, thank you for explaining that. I really just spent so much time trying to understand if I could alter a sqlite3 column type with alembic that I really just wanted to get it working, and this is what I came up with. Your method is definitely superior.

Strangely, I can't seem to import ShtVDDLevel:

```
>>> from sht_sensor import ShtVDDLevel
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name ShtVDDLevel
```

I've verified I'm using the latest [sensor.py](https://github.com/mk-fg/sht-sensor/blob/526b66f0c9ed42e7dc853079d8bbc6f88d642c9a/sht_sensor/sensor.py#L63-L64) from sht_sensor 16.12.0. What else am I missing?

kizniche, 2016-12-17:
Aha! I believe it should be ```from sht_sensor.sensor import ShtVDDLevel```

kizniche, 2016-12-17:
Strange that it's still giving an "unresolved reference" error in pycharm. The sensor tests are passing, though, so I'll go ahead and push the update.

@misudo, could you check if the latest version works when you have time?

mk-fg, 2016-12-17:
> Strangely, I can't seem to import ShtVDDLevel:

Indeed, my bad, forgot to add import for that new enum to top-level module, fixed in 16.12.1.
Should be ok using `from sht_sensor.sensor import ShtVDDLevel` though, without that shortcut.

mk-fg, 2016-12-17:
Not sure about what that pycharm error might refer to though, never really used it myself.

kizniche, 2016-12-17:
Thanks for the help. I'll wait for @misudo to chime in, but appears it should work.

misudo, 2016-12-18:
@kizniche & @mk-fg, another successful update - great team work!
I've tested all 5 voltage settings and all works great with SHTx1 sensor.

In the meantime I have a quick question: the sensor is connected to a 3.3v GPIO, but there are only 3.0v and 3.5v settings. Though the difference in measurements is only 0.2C, why isn't there a 3.3v setting?

kizniche, 2016-12-18:
Thanks for the update.

The voltage options were just what the chip-maker decided. The values were taken directly from the datasheet.

closed 2016-12-18 by kizniche

mk-fg, 2016-12-19:
> there are only 3.0v and 3.5v settings

Extrapolated correction for 3.3V as opposed to 3.5V should be something like 0.03, which is 10-20 times smaller than precision of the sensor (+/- 0.3C), so probably not worth worrying about.

Which is probably why datasheet has formula with rigid coefficients and not voltage value.
Note that it also explicitly suggests using 3.3V as calibrated-for VDD level for best accuracy, despite that weird omission `¯\_(ツ)_/¯`

There's more info on the thing here - https://github.com/mk-fg/sht-sensor/issues/1

kizniche, 2016-12-19:
@misudi, yeah, @mk-fg and I had the same discussion a year ago ;)

misudo, 2016-12-19:
@kizniche & @mk-fg  - acknowledged and agreed. Many thanks for your detailed clarification.

commit fd558b80 (2016-12-16): Fix SHT1x_7x sensor module voltage parameter (fixes #162).