kizniche/Mycodo #163: Travis CI Tests Failing

kizniche, 2016-12-16:
Current;y there are two errors that cause the [Travis CI tests to fail](https://travis-ci.org/kizniche/Mycodo/builds/184630913):

```
==================================== ERRORS ====================================

_____ ERROR collecting tests/software_tests/test_fixtures_and_factories.py _____

ImportError while importing test module '/home/travis/build/kizniche/Mycodo/mycodo/tests/software_tests/test_fixtures_and_factories.py'.

Original error message:

'The ``fake-factory`` package is now called ``Faker``.

Please update your requirements.

'

Make sure your test modules/packages have valid Python names.

__ ERROR collecting tests/software_tests/test_mycodo_flask/test_endpoints.py ___

ImportError while importing test module '/home/travis/build/kizniche/Mycodo/mycodo/tests/software_tests/test_mycodo_flask/test_endpoints.py'.

Original error message:

'The ``fake-factory`` package is now called ``Faker``.

Please update your requirements.

'

Make sure your test modules/packages have valid Python names.

!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!

=========================== 2 error in 0.39 seconds ============================
```

This doesn't occur when I run the tests from the command line.
I haven't been able to figure out a solution.

kizniche, 2016-12-17:
> The `fake-factory` package was deprecated on December 15th, 2016.

It appears factory_boy uses fake-factory and may be what is causing the errors.

kizniche, 2016-12-17:
Odd, because [factory_boy #322](https://github.com/FactoryBoy/factory_boy/pull/322/commits/dba31e3866cdefde1d7f37fc7f0553a12feeab9b) should have fixed this. I'll update the requirements.txt to the latest version of factory_boy (2.8.1) to see if that fixes, though I thought this was included in 2.7.0.

etiology, 2016-12-17:
I was just looking at the new update and you beat me to it. You're always on top of it

kizniche, 2016-12-17:
commit ad70a35 fixed the test errors.

closed 2016-12-17 by kizniche

kizniche, 2016-12-17:
Hah, that was an interesting issue to creep up. I finally put everything together after coming back to it a few times.