kizniche/Mycodo #167: Differents languages support

matatasb, 2016-12-29:
Hello my name is Matias and I'm from Argentina.

I'm starting with Mycodo and I have to say that I'm impressed with the work you have made. I want to implement Mycodo to use on a project with my friends but I would like to have an spanish interface. 

Searching a little I've found: 
[https://pythonhosted.org/Flask-Babel/](https://pythonhosted.org/Flask-Babel/).
[https://blog.miguelgrinberg.com/post/the-flask-mega-tutorial-part-xiv-i18n-and-l10n](url)

I think I could use that tool for translating the webpages and will be useful for others languages.

I wait for your comments.

kizniche, 2017-01-04:
Just writing to let you know I'm seeing this. I'm currently on the last few days of a vacation and I'll take a closer look at this info after I return home.

matatasb, 2017-01-04:
Thank you Kyle. Evaluate if this feature could bring more solutions than problems. If you think that I could help I could work on this. Enjoy your vacation.

Myke974, 2017-01-04:
Just my 2 cents about that :

1] I think, generally speaking, that this is not the *Softwares *that have
to change but the *Users *that have to learn english.
In fact, for me, and seems the passing years had prove me right, *english *is
going to be the *Universal Human Language* in the future, when humanity
will conquer space with starships, we'll speak English aboard thoses
vessels.

2] @matatasb : What to you want to translate in the web interface ?
The words like : Live, Graph, Sensor, Relay, Method,PID,Timer
Relay Id, state, activate, desactivate, ON(sec)......
all thoses are the same in every language, or very similar(like only 1
letter change), aren't they?
Thoses words were absorbed from english ! like T-shirt, Basket.....
I'm french and this absorption of the english words by/in the french
language is called "franciser" ;)

I would understand if you were speaking about translating the manual (
readme.md)

My apologises if this did seem to be aggressive : this was not intended to
be.

Hasta luego ! :) Au revoir ! Byebye !

kizniche, 2017-01-04:
I think there's a lot to gain from adding translations. This would be most useful for the form input tooltips (when hovering) and the manual. From what I've used in the past, translations are easily implemented once the framework is put in place. Then translations can be created for each block of text.

Myke974, 2017-01-04:
> @Myke974 <https://github.com/Myke974> you weren't aggresive and I think
> the same way, but when I saw Babel I thought that with some minors changes
> Mycodo could be used in any language.
> I would like to work on this task but only if all of you think that this
> would be an inprovement for Mycodo and not a headache in the future.

@kyle : tooltips when hovering ! !  agreed ! !
"easy implemented" : okk !!
Lucky you to know someone well versed in french's grammar, orthograph and
traps ;)

@matatasb : ;=)

kizniche, 2017-01-05:
With the latest mycodo_dev commit (referenced above), I was able to get translations working and initially there is a very limited amount of text able to be translated (navigation bar only, just to demonstrate it's working). It was surprisingly easy to set up and generate the translations. I added a [Translation section to the README](https://github.com/kizniche/Mycodo/blob/d0eb6a423212a8830ed4383d8cf00535f1ce93e3/README.md#translations) on the mycodo_dev branch to explain how to generate a new language translation with the little text I enabled for translation. I'll continue to add text that can be translated, but I won't be committing any large effort to get all text able to be translated until I merge the mycodo_dev branch.

[image]

matatasb, 2017-01-05:
Wow! You are really fast implementing things.. I'm starting with Rasp and everything takes me hours... Very nice!

kizniche, 2017-01-05:
The nice thing about translations is once I have all the code in for each piece of text, anyone can contribute to the translations, in any language. It's tedious going through every piece of text and modifying the code to work with Flask-Babel, but it's also satisfying how streamlined the babel package is. I should have the majority of the text able to be translated by this weekend, if not today.

kizniche, 2017-01-05:
I just finished a big section. I'm going to tackle the manual at some point, but I'd like to review and edit it to be more comprehensible before doing that. I wouldn't want to waste anyone's time who would be translating.

There are now 312 translatable text objects. I've translated 43 of them to Spanish.

@matatasb, If you would like to translate some to Spanish, that would be great! I can walk you through the exact steps for checking out the dev branch, getting the editor open to translate, and saving the translation file you can send to me. Let me know.

matatasb, 2017-01-05:
@kizniche  of course I would like to do that. I have been reading some translation on .po file. 

If you have some guide I will thank you, still I don't find myself very confident with github.

kizniche, 2017-01-05:
This should be everything needed to get going. In a linux terminal, execute these commands:

Create new directory separate from the non-dev Mycodo:
```
cd ~
mkdir mycodo_dev
git clone -b mycodo_dev https://github.com/kizniche/Mycodo.git
cd mycodo_dev/Mycodo
```

Make sure all pip requirements are installed:

```pip install -r install/requirements.txt --upgrade```

Go to root of mycodo directory:

```cd mycodo```

Extract all translatable texts to messages.pot

```pybabel extract -F babel.cfg -k lazy_gettext -o messages.pot .```

Update Mycodo/mycodo/mycodo_flask/translations/es/LC_MESSAGES/messages.po file (as well as other translation files, including the french folder already present):

```pybabel update -i messages.pot -d mycodo_flask/translations```

Download [poedit](https://poedit.net/) or similar software and then open the Mycodo/mycodo/mycodo_flask/translations/es/LC_MESSAGES/messages.po with it.

This should bring up 300+ English words or phrases. Enter the translation for each word or phrase, save the messages.po file, then send it to me. You can contact me at my website [contact form](http://kylegabriel.com/contact/) for us to exchange email addresses.

kizniche, 2017-01-05:
The last commit brought the number of translatable text/phrases to 505 items! The majority of Mycodo is now able to be translated (excluding the manual).

matatasb, 2017-01-06:
Thank you for the tutorial! I'm going to start the translation now.

Myke974, 2017-01-06:
/mycodo_dev/mycodo/mycodo_flask/translations/fr/LC_MESSAGES/messages.po
Cannot be open by Poedit Windows 10
error : corrupt file or not reconised by Poedit.

kizniche, 2017-01-06:
@myke974, It works for me. It also appears on github to be complete (not corrupt). It's probably an issue on your end. Alternatively, with the instructions I provide above, you can generate your own messages.po file.

matatasb, 2017-01-06:
It also worked for me. I use Poedit on Windows 10.

Myke974, 2017-01-06:
Seriously ? :O
I just tryed again and again and the one in spanish too and same error :O
[image: Images intégrées 1]

@Kyle : yes but downloading the file from the web and launching poedit was
the quickest and lazyest  way for me ;)
Worst case I'll do it a shot directly from Rpi command line as you
explained.
But still ! I don't see why it won't open ! size and file's content are ok
! :O

kizniche, 2017-01-06:
If you can't get the software to work, I'm fairly sure you could just open the file and insert your text into the translated quotes. There are also other po editors (virtaal).

Myke974, 2017-01-06:
yes ! saw that through web github ;)

Myke974, 2017-01-06:
I found out my error !
On the webpage
https://github.com/kizniche/Mycodo/tree/mycodo_dev/mycodo/mycodo_flask/translations/fr/LC_MESSAGES
I right clicked on message.po and "save link as"  and it registered the
html of the page :
https://github.com/kizniche/Mycodo/blob/mycodo_dev/mycodo/mycodo_flask/translations/fr/LC_MESSAGES/messages.po
I use edit then CTRL+A to select all and copyed it in replacement of my
already downloaded message.po
and PoEdit opened it ;)

2017-01-07 2:50 GMT+04:00 Myke974 <myke974@gmail.com>:

kizniche, 2017-01-06:
It's looking good! Now that there's an option in the settings I can change the language and then search for texts I missed that will appear in English (notice a couple in the image below).

[image]

kizniche, 2017-01-07:
I'd like to recommend holding off on translating everything, until I push the next commit. I'm currently vastly reducing the number of translatable phrases.

Myke974, 2017-01-07:
I don't understand how you will reduce the number of translatable phrases !
This means you'll keep some words in english ?

kizniche, 2017-01-07:
If there are two phrases, such as:

  - "Guests are not allowed to activate PIDs"
  - "Guests are not allowed to deactivate PIDs"

And I change them both to: "Guests are not allowed to do that"

Then instead of two phrases to translate, there is only one.

Myke974, 2017-01-07:
OK ! yeah I thought of that when I began the translation yesterday !

kizniche, 2017-01-14:
Well, with the latest commits/releases, translations are in. I'm going to close this issue, but if you ever want to add translations, you can make a pull request or just send me the translation file (messages.po) after following the [translation instructions](https://github.com/kizniche/Mycodo#translations).

closed 2017-01-14 by kizniche