kizniche/Mycodo #170: enhancement suggestion: Telegram bot

itzik16, 2017-01-13:
####Description
Hey Kyle, I had an Idea to integrate  Mycodo with wit.ai(Optinal) to form a telegram bot to make an easy solution to check the status of the sensors and Etc on the go.
I don't think it should be too complex and I'd like to hear what you will say about it 

 
### Additional Notes
Here is a sample bot simple guide for reference with code: 
https://www.hackster.io/Salman_faris_vp/telegram-bot-with-raspberry-pi-f373da?ref=part&ref_id=19713&offset=3

kizniche, 2017-01-14:
Thanks for the suggestion. I'm not too familiar with this. How were you envisioning it being implemented? Could you give a brief example(s)?

itzik16, 2017-01-15:
Yeah sure, Basically telegram is an encrypted free IM service like messenger that has bots and "Bots are simply Telegram accounts operated by software."
So I thought it would be nice to have it in Mycodo.
It could be used to check the temps and to turn on and off relay and such, there is many ways to implement it you can either create simple functions for the bot to read or export graphs and turn on/off relays.
here is an example i found online for a basic bot on RPI
[image]

[image]

or it could be used with AI so you could talk to the bot instead and ask it do what  you need. but that is a little more complicated. Wit.ai is an AI made for Bots by Facebook's messenger its open source and free to use, it could be trained to recognize what you are saying and then can reply back.
I am not an expert Im an extreme hobbyist with broken english so I can't really explain that to you all the way so Im sorry if thats a vague explanation
 
thanks for your great work mate 👍 
here is some of the sources i used:
http://www.instructables.com/id/Set-up-Telegram-Bot-on-Raspberry-Pi/?ALLSTEPS
https://youtu.be/ejnE6DM-hqU
https://www.youtube.com/watch?v=VhaKG0QYXck

DrArkaneX, 2017-11-13:
I could second this, however, I wouldn't want the world to see my stats.  I'm wondering how this can be effected on mycodo to ensure only you or selected people can see this data.

BTW Kyle, Thank you for your hard work in this.  soon as i get all my sensors and stuff, i'll try to lend a hand with this project.  You are awesome

Theoi-Meteoroi, 2017-12-26:
I think I would rather have a MQTT server that can be configured.  The server could then forward to many other types of endpoints rather than being bound to an implementation, as this suggests.  Any consideration for data egress must have a security stance and path to being audited as secure.  Standard implementations like MQTT at least have a large enough user base to have that traceability and test cycles to find the always present "ah shit" vulnerabilities.

closed 2018-03-04 by kizniche