kizniche/Mycodo #171: HUMI-01(DHT11) Not working in WebUI

MurderLabz, 2017-01-14:
Mycodo 4.1.4
Pi 2 -> model name	: ARMv7 Processor rev 5 (v7l) BogoMIPS	: 38.40
Raspbian GNU/Linux 8 \n \l
Linux raspberrypi 4.4.41-v7+ #942 SMP Mon Jan 9 15:00:25 GMT 2017 armv7l GNU/Linux

**Sensor** -> HUMI-01-OSEPP (DHT11)
https://www.jameco.com/z/HUMI-01-OSEPP-DHT11-Humidity-and-Temperature-Sensor_2237116.html
##################################################
DHT11 shows green and activates fine but displays no data in WebUI.
RPi Device displays and works fine.
Moved from GPIO pin  4 then 6. Both Pins past GPIO test below
##########################################
**Sensor passes test >>**
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ sudo python test_gpio_DHTx.py -g 6 DHT11
Temperature: 21.0
Humidity: 37.0
###########################################
**/var/log/mycodo/mycodo.log**
Traceback (most recent call last):
  File "/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
    measurements = self.measure_sensor.next()
  File "/Mycodo/mycodo/sensors/dht11.py", line 75, in next
    raise StopIteration  # required
StopIteration
2017-01-14 06:58:11,754 - INFO - [Sensor n0b7NXQl] Activated in 130.2 ms
2017-01-14 06:58:11,756 - ERROR - [Sensor n0b7NXQl] Error while attempting to read sensor: 
Traceback (most recent call last):
  File "/Mycodo/mycodo/controller_sensor.py", line 623, in updateMeasurement
    measurements = self.measure_sensor.next()
  File "/Mycodo/mycodo/sensors/dht11.py", line 75, in next
    raise StopIteration  # required
StopIteration

kizniche, 2017-01-14:
Thanks for the bug report. I'm taking a look at this now.

kizniche, 2017-01-14:
I think I found the bug. I'm going to push an update soon.

kizniche, 2017-01-14:
Actually, I'm not so sure that was the bug you experienced. I'm going to paste some test code for you to run and let me know if it returns temperature/humidity data.

MurderLabz, 2017-01-14:
Thanks for your time and efforts on this great project.

kizniche, 2017-01-14:
Try running this code with the command ```sudo python dht11_test.py```

```python
#!/usr/bin/python
#  coding=utf-8
import logging
import sys
import time
import pigpio

logging.basicConfig(level=logging.INFO)
logger = logging.getLogger(__name__)

class DHT11Sensor:
    def __init__(self, pi, gpio, power=None):
        self.pi = pi
        self.gpio = gpio
        self.power = power
        self.high_tick = 0
        self.bit = 40
        self.either_edge_cb = None
        self._humidity = 0.0
        self._temperature = 0.0

    def get_measurement(self):
        if self.power is not None:
            logger.debug("Turning on sensor at GPIO {}...".format(self.gpio))
            self.pi.write(self.power, 1)  # Switch sensor on.
            time.sleep(2)
        try:
            self.setup()
            self.pi.write(self.gpio, pigpio.LOW)
            time.sleep(0.017)  # 17 ms
            self.pi.set_mode(self.gpio, pigpio.INPUT)
            self.pi.set_watchdog(self.gpio, 200)
            time.sleep(0.2)
        except Exception as e:
            logger.error("{cls} raised an exception when taking a reading: "
                         "{err}".format(cls=type(self).__name__, err=e))
        finally:
            self.close

    def read(self):
        try:
            self.get_measurement()
            # self_humidity and self._temperature are set in self._edge_rise()
            return  # success - no errors
        except:
            return 1

    def get(self):
        return self._temperature, self._humidity

    def setup(self):
        self.pi.set_pull_up_down(self.gpio, pigpio.PUD_OFF)
        self.pi.set_watchdog(self.gpio, 0)
        self.register_callbacks()

    def register_callbacks(self):
        self.either_edge_cb = self.pi.callback(self.gpio,
                                               pigpio.EITHER_EDGE,
                                               self.either_edge_callback)

    def either_edge_callback(self, gpio, level, tick):
        level_handlers = {
            pigpio.FALLING_EDGE: self._edge_fall,
            pigpio.RISING_EDGE: self._edge_rise,
            pigpio.EITHER_EDGE: self._edge_either
        }
        handler = level_handlers[level]
        diff = pigpio.tickDiff(self.high_tick, tick)
        handler(tick, diff)

    def _edge_rise(self, tick, diff):
        val = 0
        if diff >= 50:
            val = 1
        if diff >= 200:  # Bad bit?
            self.checksum = 256  # Force bad checksum
        if self.bit >= 40:  # Message complete
            self.bit = 40
        elif self.bit >= 32:  # In checksum byte
            self.checksum = (self.checksum << 1) + val
            if self.bit == 39:
                # 40th bit received
                self.pi.set_watchdog(self.gpio, 0)
                total = self._humidity + self._temperature
                # is checksum ok ?
                if not (total & 255) == self.checksum:
                    raise Exception
        elif 16 <= self.bit < 24:  # in temperature byte
            self._temperature = (self._temperature << 1) + val
        elif 0 <= self.bit < 8:  # in humidity byte
            self._humidity = (self._humidity << 1) + val
        self.bit += 1

    def _edge_fall(self, tick, diff):
        self.high_tick = tick
        if diff <= 250000:
            return
        self.bit = -2
        self.checksum = 0
        self._temperature = 0
        self._humidity = 0

    def _edge_either(self, tick, diff):
        self.pi.set_watchdog(self.gpio, 0)

    def close(self):
        self.pi.set_watchdog(self.gpio, 0)
        if self.either_edge_cb:
            self.either_edge_cb.cancel()
            self.either_edge_cb = None

if __name__ == '__main__':
    try:
        gpio = raw_input("DHT11 Sensor GPIO Pin? ")
        sensor = DHT11Sensor(pigpio.pi(), gpio)
        if sensor.read():
            sys.exit(1)
        temp, hum = sensor.get()
        logger.info("Temp: {tmp}, Hum: {hum}".format(tmp=temp, hum=hum))
    except KeyboardInterrupt:
        sys.exit(1)
```

kizniche, 2017-01-14:
I just made a slight tweak to the code above, so if you happened to have already copied it, can you please try it again? Thanks.

MurderLabz, 2017-01-14:
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ sudo python test_gpio_DHTx.py -g 6 DHT11
Temperature: 20.0
Humidity: 38.0
Temperature: 16.0
Humidity: 23.0
^CTraceback (most recent call last):
  File "test_gpio_DHTx.py", line 53, in <module>
    menu()
  File "test_gpio_DHTx.py", line 46, in menu
    time.sleep(2)
KeyboardInterrupt

pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ sudo python issue_171.py 
DHT11 Sensor GPIO Pin? 6
ERROR:__main__:instance raised an exception when taking a reading: cannot convert argument to integer
INFO:__main__:Temp: 0.0, Hum: 0.0
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $

MurderLabz, 2017-01-14:
Just to verify I has the updated code

pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ sudo python issue_171.v2.py 
DHT11 Sensor GPIO Pin? 6
ERROR:__main__:instance raised an exception when taking a reading: cannot convert argument to integer
INFO:__main__:Temp: 0.0, Hum: 0.0
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $

MurderLabz, 2017-01-14:
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ which python
/usr/bin/python
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ python --version
Python 2.7.9

kizniche, 2017-01-14:
Let me back up a minute.

Did you have a previous version of Mycodo installed that the DHT11 did work with? If so, what version was that?
Was this an upgrade or a fresh install of Mycodo?

MurderLabz, 2017-01-14:
fresh install with 4.1.4

kizniche, 2017-01-14:
I just updated the code above. Try it and let me know the output.

kizniche, 2017-01-14:
But you have never had the DHT11 working with any version of Mycodo?

MurderLabz, 2017-01-14:
woohoo!

pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $ sudo python issue_171.v3.py 
DHT11 Sensor GPIO Pin? 6
INFO:__main__:Temp: 19, Hum: 37
pi@raspberrypi:/Mycodo/mycodo/tests/manual_tests $

MurderLabz, 2017-01-14:
But you have never had the DHT11 working with any version of Mycodo? = no

kizniche, 2017-01-14:
Great, it's working! That code is actually completely different than the test script you were using. I believe it is more stable, so I was dreading thinking I might have to go back to using that older code to get the sensor working. Let me see what I changed and see if I can get an update to the module pushed to github so you can upgrade.

MurderLabz, 2017-01-14:
Thanks again and apologies for having you resurrect old code. Not sure how many people will be using DHT11's I plan to use AM2302's in my setups.

kizniche, 2017-01-14:
You're welcome. I wasn't resurrecting old code. I was saying I'm glad I was able to figure out the issue and get it working with the new code so I wouldn't have to go back to using the old (the original test script you had it working with).

kizniche, 2017-01-14:
I just created release v4.1.5, so it should be available for upgrade in Mycodo.

MurderLabz, 2017-01-14:
I am missing something.. Fresh install of Raspbian and mycodo 4.1.5 still no WebUI out put.
No errors.. 

2017-01-14 14:31:16,543 - INFO - [Sensor orkVSG3y] Activated in 135.7 ms
2017-01-14 14:31:16,784 - INFO - [Sensor Cguhk3d2] Activated in 128.4 ms
2017-01-14 14:31:16,789 - INFO - [Daemon] All activated sensor controllers started
2017-01-14 14:31:16,790 - INFO - [Daemon] All activated log controllers started
2017-01-14 14:31:16,791 - INFO - [Daemon] All activated PID controllers started
2017-01-14 14:31:16,792 - INFO - [Daemon] All activated LCD controllers started
2017-01-14 14:31:16,793 - INFO - [Daemon] Mycodo v4.1.5 started in 1.047 seconds
2017-01-14 14:31:16,801 - INFO - [Daemon] 26.912 MB ram in use
pi@raspberrypi:~ $

kizniche, 2017-01-14:
Try restarting the pi. If that doesn't fix it, I'll give you some code to try.

MurderLabz, 2017-01-14:
its alive! this crappy sensor is alive! It was pigpio, I think? Thanks Doc!

pigpiod.service - Daemon required to control GPIO pins via pigpio
   Loaded: loaded (/lib/systemd/system/pigpiod.service; disabled)
   Active: active (running) since Sat 2017-01-14 14:53:01 MST; 2min 35s ago
  Process: 943 ExecStart=/usr/bin/pigpiod -l (code=exited, status=0/SUCCESS)
 Main PID: 944 (pigpiod)
   CGroup: /system.slice/pigpiod.service
           └─944 /usr/bin/pigpiod -l

kizniche, 2017-01-14:
Nice! So, it's working in Mycodo? Let me know and we can lay this issue to rest. Thanks for the help in diagnosing the problem.

kizniche, 2017-01-14:
I'll add starting pigpiod to the setup.sh script. That will alleviate a lot of head-scratching for those who haven't rebooted.

MurderLabz, 2017-01-14:
Yes done!

closed 2017-01-15 by kizniche