kizniche/Mycodo #172: PID controller stopped after upgrade

zsole2, 2017-01-14:
## Mycodo Issue Report:
Mycodo Version: 4.1.4
 
#### Problem Description
Upgrade went fine, then my HTU21D sensor went blank, both on the Live and Graph screens.

### Errors
The Demon log showed:
2017-01-14 09:21:09,245 - WARNING - [PID cl1V5Wr0] No data returned from influxdb
2017-01-14 09:22:48,524 - WARNING - [PID CKwFRAWB] No data returned from influxdb
(these are the 2 PIDs for temp and hum)

### Additional Notes

Kyle,
Nice work again with the new version, as usual, lot of improvement that you have accomplished here!!!! Congratulations!
To get me going, instead of troubleshooting the new version with the structural changes, I decided to restore the previous version. I'll certainly check it again as soon as my secondary device will be rebuilt.

kizniche, 2017-01-14:
Thanks. I was a bit nervous getting this ready since it was such a drastic change to the upgrade system, but after spending a few hours ironing out the issues, I'm glad it's working for other people and not just myself.
I'll look into the differences between the sensor module in each version.

kizniche, 2017-01-14:
I think I found the problem. Can you run this code and see if it returns your temp/hum from the sensor? It's a distilled version of the sensor module code with a fix to one variable.

```python
#!/usr/bin/python
# coding=utf-8
import pigpio
import time

class HTU21DSensor:
    def __init__(self):
        self.I2C_bus_number = 1
        self.address = 0x40  # HTU21D-F Address
        self.pi = pigpio.pi()

    def get_measurement(self):
        """ Gets the humidity and temperature """
        # wtreg = 0xE6
        # rdreg = 0xE7
        rdtemp = 0xE3
        rdhumi = 0xE5

        handle = self.pi.i2c_open(self.I2C_bus_number, self.address)  # open i2c bus
        self.pi.i2c_write_byte(handle, rdtemp)  # send read temp command
        time.sleep(0.055)  # readings take up to 50ms, lets give it some time
        (_, byte_array) = self.pi.i2c_read_device(handle, 3)  # vacuum up those bytes
        self.pi.i2c_close(handle)  # close the i2c bus
        t1 = byte_array[0]  # most significant byte msb
        t2 = byte_array[1]  # least significant byte lsb
        temp_reading = (t1 * 256) + t2  # combine both bytes into one big integer
        temp_reading = float(temp_reading)
        temperature = ((temp_reading / 65536) * 175.72) - 46.85  # formula from datasheet

        handle = self.pi.i2c_open(self.I2C_bus_number, self.address)  # open i2c bus
        self.pi.i2c_write_byte(handle, rdhumi)  # send read humi command
        time.sleep(0.055)  # readings take up to 50ms, lets give it some time
        (_, byte_array) = self.pi.i2c_read_device(handle, 3)  # vacuum up those bytes
        self.pi.i2c_close(handle)  # close the i2c bus
        h1 = byte_array[0]  # most significant byte msb
        h2 = byte_array[1]  # least significant byte lsb
        humi_reading = (h1 * 256) + h2  # combine both bytes into one big integer
        humi_reading = float(humi_reading)
        uncomp_humidity = ((humi_reading / 65536) * 125) - 6  # formula from datasheet
        humidity = ((25 - temperature) * -0.15) + uncomp_humidity
        return humidity, temperature

    def htu_reset(self):
        reset = 0xFE
        handle = self.pi.i2c_open(self.I2C_bus_number, self.address)  # open i2c bus
        self.pi.i2c_write_byte(handle, reset)  # send reset command
        self.pi.i2c_close(handle)  # close i2c bus
        time.sleep(0.2)  # reset takes 15ms so let's give it some time

    def read(self):
        try:
            self.htu_reset()
            (humidity,
             temperature) = self.get_measurement()
            print("Temp: {tmp}, Hum: {hum}".format(tmp=temperature, hum=humidity))
            return  # success - no errors
        except Exception as e:
            print("{cls} raised an exception when taking a reading: "
                  "{err}".format(cls=type(self).__name__, err=e))
        return 1

if __name__ == '__main__':
    htu = HTU21DSensor()
    for _ in range(3):
        htu.read()
        time.sleep(2)
```

kizniche, 2017-01-14:
Can you also confirm there is a line in your crontab that starts pigpio at bootup?

```sudo crontab -e```

zsole2, 2017-01-14:
Thanks for this test code. Tomorrow  I'll try to test it, I'll have to find another SD card with an older install to upgrade... 
Pigpio should have been OK, other than the sensor,  everything else worked fine (i.e.  relays, on-off button, etc.) ... all connected to various GPIO pins.

kizniche, 2017-01-14:
I'm not sure if I was clear about how to run the code. It can be ran by itself, not put into Mycodo, if that's what you were thinking. ```sudo python test_script.py```
It appears the sensor module logging isn't working, and that's not good for debugging. I'm also working on that now.

zsole2, 2017-01-14:
OK, I run it already, in a separate script file, but right now my RasPi has  the older version of Mycodo (which I restored back, 4.0.24).
It worked, printing out three pairs of values.
What I meant trying is   to upgrade another old install to 4.1.4 the same way it failed earlier, and run it there.

kizniche, 2017-01-14:
Great. That means the fix worked. I'll push an update later after I've tested a few other fixes.

If it's a really old version, it might just be best to install it new from the install instructions, then copy your databases over, and run these commands to upgrade the databases:

```cd Mycodo/databases && alembic upgrade head```

kizniche, 2017-01-14:
I just created release v4.1.5, so it should be available for upgrade in Mycodo.

zsole2, 2017-01-15:
Kyle,
I've tried to upgrade to this new version, but did not succeed. The issue is that after yesterday's restore (following the old instructions, i.e. copying back the backup from /var/Mycodo_backups), I get the message that my system is up to date. Which is not, it is v4.0.24.
After trying to run the update script from the terminal, it finishes with the same message, but here is the output:

pi@chz-113:~ $ sudo Mycodo/mycodo/scripts/update_mycodo.sh upgrade
#### Update Initiated 01-15-2017 13:02:10 ####
#### Checking for Update ####
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
Your version of Mycodo is already the latest version.

I then realized that the .git directory is not copied over by the script at the time of backup. Since I not a git.wizard, I stopped here.

kizniche, 2017-01-15:
What I would do in this case, is move your Mycodo install directory to Mycodo-old, then follow the new instructions for fresh install, then copy your databases from Mycodo-old/databases/*.db to the new Mycodo install, then update the databases with ```cd Mycodo/databases && alembic upgrade head```
That is essentially what the upgrade should have done if it had the .git directory. You could try to rebuild your .git directory, but that would probably be more difficult than just moving and upgrading the databases to the latest version.

kizniche, 2017-01-15:
Your .git directory should have stayed intact in the ~/Mycodo directory. Only the backups don't contain the .git directory. So, you must have deleted or overwrote your ~/Mycodo directory at some point to lose it? In any case, this is one reason why I moved from using git commits to releases, because keeping track of commits and a .git directory is difficult.

zsole2, 2017-01-16:
The version I have (4.0.24) has the following code in update_mycodo.sh:

```
case "${1:-''}" in
    'backup')
        NOW=$(date +"%Y-%m-%d_%H-%M-%S")
        CURCOMMIT=$(git rev-parse --short HEAD)
        printf "#### $INSTALL_DIRECTORY Creating backup /var/Mycodo-b$
        mkdir -p /var/Mycodo-backups
        mkdir -p /var/Mycodo-backups/Mycodo-$NOW-$CURCOMMIT
        rsync -ah --stats --exclude old --exclude .git --exclude src $
    ;;
```
Apparently the .git directory is omitted at backup as you said. On the other hand, the old restore instructions started like:

> sudo mv ~/Mycodo ~/Mycodo_old
sudo cp -a /var/Mycodo-backups/Mycodo-TIME-COMMIT ~/Mycodo

Therefore the .git directory got lost in the process...

In any  case, I plan to rebuild the system as you suggested,  just have to find the time for it...

kizniche, 2017-01-16:
Sorry for the lack of current documentation. It's a lot to keep track of. The whole backup/restore part of Mycodo (and docs) could use some work. I was holding off on that until I got the new upgrade system in place.

The new upgrade system is looking pretty good so far. I can push updates and not have to worry about users upgrading to the latest commit, until I make a release. This saves a lot of headaches when I find an issue and need to quickly fix it before someone upgrades and breaks their system.

zsole2, 2017-01-16:
I did not want to complain, just wanted to find out what the problem was...

kizniche, 2017-01-16:
I wasn't trying to insinuate that. I was just thinking aloud and realizing I should update the docs before the next release.

kizniche, 2017-01-19:
I think everything in this issue has been resolved. If so, let me know, and I'll close it.

zsole2, 2017-01-21:
I believe so, too. You can close it.

closed 2017-01-21 by kizniche