kizniche/Mycodo #174: dht22 acting oddly

Drakthule, 2017-01-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.6
 
#### Problem Description
Please list: daemon is not automatically starting after install, dht22 is failing to initialize

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

> 2017-01-19 00:14:46,460 - mycodo - INFO - Mycodo v4.1.6 starting
2017-01-19 00:14:46,843 - mycodo - INFO - Anonymous statistics enabled
2017-01-19 00:14:48,272 - mycodo - DEBUG - Starting relay controller
2017-01-19 00:14:48,277 - mycodo.relay - DEBUG - Initializing Relays
2017-01-19 00:14:48,346 - mycodo.relay - INFO - Relays Initialized
2017-01-19 00:14:48,347 - mycodo.relay - INFO - Relay controller activated in 73.8 ms
2017-01-19 00:14:48,348 - mycodo - DEBUG - Starting all activated timer controllers
2017-01-19 00:14:48,349 - mycodo - INFO - All activated timer controllers started
2017-01-19 00:14:48,349 - mycodo - DEBUG - Starting all activated sensor controllers
2017-01-19 00:14:48,673 - mycodo - ERROR - Could not activate Sensor controller with ID wQnGiG8M: DHT22 could not initialize. Check if gpiod is running.
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 393, in activate_controller
    ready, cont_id)
  File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 241, in __init__
    self.measure_sensor = DHT22Sensor(pigpio.pi(), int(self.location))
  File "/home/pi/Mycodo/mycodo/sensors/dht22.py", line 69, in __init__
    raise Exception('DHT22 could not initialize. Check if gpiod is running.')
Exception: DHT22 could not initialize. Check if gpiod is running.
2017-01-19 00:14:48,686 - mycodo - INFO - All activated sensor controllers started
2017-01-19 00:14:48,687 - mycodo - DEBUG - Starting all activated PID controllers
2017-01-19 00:14:48,689 - mycodo - INFO - All activated PID controllers started
2017-01-19 00:14:48,689 - mycodo - DEBUG - Starting all activated LCD controllers
2017-01-19 00:14:48,689 - mycodo - INFO - All activated LCD controllers started
2017-01-19 00:14:48,690 - mycodo - INFO - Mycodo v4.1.6 started in 2.230 seconds
2017-01-19 00:14:48,700 - mycodo - INFO - 27.328 MB ram in use
2017-01-19 00:16:47,695 - mycodo.stats - DEBUG - Sent anonymous usage statistics

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

so a couple nights ago i felt ambitious and decided to clean up my bread board, since i just upgraded and all to what i think was 4.1.5. So I re-wired everything and to my dismay the dht22 kept reporting 0 for all readings. "well damn guess i burnt up the sensor or something" 2 days later i got a new sensor and still same issue. so i yanked out the sd and slapped it in my backup pi3B and low and behold same issue so i decided that i would look at the wiring. i originally had the + and data tied with a 10k resistor and it worked fine, even went to radio shack and picked up a 4.7k after i did some snooping in the support tickets. this also didnt work. so i decided its time to wipe it out, fresh reload of raspian, fresh install and after a reboot the daemon wouldnt start on its own, i figured i screwed up. reload and still the daemon is having a bad day and its either im wiring something wrong or ive encoutnered some bug thats preventing the pi from utilizing the sensor.

your project is pretty amazing kyle, i hope to recieve your sagely advice!

kizniche, 2017-01-19:
First, can you check if gpiod is running?

```ps aux | grep gpiod```

kizniche, 2017-01-19:
Just so I'm understanding correctly, the issue first presented itself when you rewired the sensor and could no longer get measurements?

kizniche, 2017-01-19:
Then, you wiped and made a fresh install of 4.1.6, and the Mycodo daemon wouldn't automatically start after the install?

I notice you're running it in debug mode, above.

isoflux, 2017-01-19:
Hi,
First of all, I would like to thank kizniche for doing a great job.
I am also facing the similar problem during clean installation on RPI 2 model B.  I first tried with Mycodo 4.1.5 a few day ago and reinstall 4.1.6 today with the similar issues.  I also did 

`sudo pip install -r ~/Mycodo/install/requirements.txt --upgrade --force-reinstall --no-deps`

but the issues still persist.

The issues are;

-Daemon is up running after finish installation, but not running/start when reboot.  
I started the mycodo service manually to solve the problem. 

-Error message when added sensors (DHT22) " Check if gpiod is running."  
I added the line

`@reboot /usr/local/bin/pigpiod &` 

at the end of crontab. Problem solved.

-Live page read 0oC with correct timestamp 
Once I tested the sensor with 

`sudo  python Mycodo/mycodo/tests/manual_tests/test_gpio_DHTx.py -g 4 DHT22`

On Live page showed the correct temperature for 5 seconds.  However, it wouldn’t update the temperature until you run testing sensor again.  It seems that a reading temperature script didn't work properly.

Hope these give you guys some clues.
Cheers,

kizniche, 2017-01-19:
Hmm, it may seem that gpiod isn't being properly added to crontab!

The test script is getting measurements but not the Mycodo sensor module.

Hmmmm. Let me investigate a bit.

kizniche, 2017-01-19:
@isoflux, are there any errors in your daemon log (Tools->Mycodo Logs->Daemon)?

isoflux, 2017-01-19:
Here are the error message.
`pi@raspberrypi:~ $ cat /var/log/mycodo/mycodo.log`
2017-01-18 17:25:10,956 - Mycodo - INFO - Anonymous statistics enabled
2017-01-18 17:25:11,344 - Mycodo.Relay - INFO - Relays Initialized
2017-01-18 17:25:11,347 - Mycodo.Relay - INFO - Relay controller activated in 12
4.9 ms
2017-01-18 17:25:11,348 - Mycodo - INFO - All activated timer controllers starte
d
2017-01-18 17:25:11,350 - Mycodo - INFO - All activated sensor controllers start
ed
2017-01-18 17:25:11,351 - Mycodo - INFO - All activated PID controllers started
2017-01-18 17:25:11,351 - Mycodo - INFO - All activated LCD controllers started
2017-01-18 17:25:11,357 - Mycodo - INFO - Mycodo v4.1.6 started in 0.622 seconds
2017-01-18 17:25:11,366 - Mycodo - INFO - 25.268 MB ram in use
2017-01-18 17:33:34,709 - Mycodo.stats - ERROR - Could not send anonymous usage 
statistics: HTTPConnectionPool(host='fungi.kylegabriel.com', port=8086): Max ret
ries exceeded with url: /write?db=mycodo_stats (Caused by NewConnectionError('<r
equests.packages.urllib3.connection.HTTPConnection object at 0x75096970>: Failed
 to establish a new connection: [Errno 110] Connection timed out',))
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/utils/statistics.py", line 263, in send_stats
    client.write_points(formatted_stat_dict)
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 391, in
 write_points
    tags=tags)
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 436, in
 _write_points
    expected_response_code=204
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 278, in
 write
    headers=headers
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 241, in
 request
    raise e
ConnectionError: HTTPConnectionPool(host='fungi.kylegabriel.com', port=8086): Ma
x retries exceeded with url: /write?db=mycodo_stats (Caused by NewConnectionErro
r('<requests.packages.urllib3.connection.HTTPConnection object at 0x75096970>: F
ailed to establish a new connection: [Errno 110] Connection timed out',))
2017-01-18 17:38:30,456 - Mycodo.Sensor-0FB5MaNC - INFO - Activated in 228.2 ms
2017-01-18 17:38:37,165 - Mycodo.Sensor-0w9lilcn - INFO - Activated in 229.3 ms
2017-01-18 17:46:42,483 - Mycodo.Sensor-0w9lilcn - INFO - Deactivated in 19.6 ms
2017-01-18 17:46:49,535 - Mycodo.Sensor-0w9lilcn - INFO - Activated in 237.3 ms
2017-01-18 17:58:00,263 - Mycodo - INFO - Received command to terminate daemon
2017-01-18 17:58:00,472 - Mycodo - INFO - All LCD controllers stopped
2017-01-18 17:58:00,475 - Mycodo - INFO - All Timer controllers stopped
2017-01-18 17:58:00,479 - Mycodo.Sensor-0FB5MaNC - INFO - Deactivated in 9.3 ms
2017-01-18 17:58:00,560 - Mycodo.Sensor-0w9lilcn - INFO - Deactivated in 88.2 ms
2017-01-18 17:58:00,565 - Mycodo - INFO - All Sensor controllers stopped
2017-01-18 17:58:00,567 - Mycodo - INFO - All PID controllers stopped
2017-01-18 17:58:00,573 - Mycodo.Relay - INFO - Relay controller deactivated in 
3.5 ms
2017-01-18 17:58:00,577 - Mycodo - INFO - Mycodo terminated in 0.314 seconds
2017-01-19 00:07:47,672 - Mycodo - INFO - Anonymous statistics enabled
2017-01-19 00:07:48,010 - Mycodo.Relay - INFO - Relays Initialized
2017-01-19 00:07:48,012 - Mycodo - INFO - All activated timer controllers starte
d
2017-01-19 00:07:48,013 - Mycodo.Relay - INFO - Relay controller activated in 97
.8 ms
2017-01-19 00:07:48,427 - Mycodo - ERROR - Could not activate Sensor controller 
with ID 0FB5MaNC: DHT22 could not initialize. Check if gpiod is running.
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 394, in activateControlle
r
    ready, cont_id)
  File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 241, in __init__
    self.measure_sensor = DHT22Sensor(pigpio.pi(), int(self.location))
  File "/home/pi/Mycodo/mycodo/sensors/dht22.py", line 69, in __init__
    raise Exception('DHT22 could not initialize. Check if gpiod is running.')
Exception: DHT22 could not initialize. Check if gpiod is running.
2017-01-19 00:07:48,695 - Mycodo - ERROR - Could not activate Sensor controller 
[... 82 lines cut ...]
 write
    headers=headers
  File "/usr/local/lib/python2.7/dist-packages/influxdb/client.py", line 248, in
 request
    raise InfluxDBClientError(response.content, response.status_code)
InfluxDBClientError: 400: {"error":"partial write:\nunable to parse 'country,ano
nymous_id=nUP5EF5qzhKB ': invalid field format"}
2017-01-19 00:40:27,176 - Mycodo - INFO - Received command to terminate daemon
2017-01-19 00:40:27,428 - Mycodo - INFO - All LCD controllers stopped
2017-01-19 00:40:27,430 - Mycodo - INFO - All Timer controllers stopped
2017-01-19 00:40:27,431 - Mycodo - INFO - All Sensor controllers stopped
2017-01-19 00:40:27,431 - Mycodo - INFO - All PID controllers stopped
2017-01-19 00:40:27,441 - Mycodo.Relay - INFO - Relay controller deactivated in 8.4 ms
2017-01-19 00:40:27,443 - Mycodo - INFO - Mycodo terminated in 0.267 seconds
2017-01-19 00:46:04,640 - Mycodo - INFO - Anonymous statistics enabled
2017-01-19 00:46:04,876 - Mycodo.Relay - INFO - Relays Initialized
2017-01-19 00:46:04,878 - Mycodo.Relay - INFO - Relay controller activated in 69.1 ms
2017-01-19 00:46:04,879 - Mycodo - INFO - All activated timer controllers started
2017-01-19 00:46:05,174 - Mycodo.Sensor-WvIw4EeE - INFO - Activated in 130.2 ms
2017-01-19 00:46:05,350 - Mycodo.Sensor-VMdC8Zxc - INFO - Activated in 122.1 ms
2017-01-19 00:46:05,353 - Mycodo - INFO - All activated sensor controllers started
2017-01-19 00:46:05,355 - Mycodo - INFO - All activated PID controllers started
2017-01-19 00:46:05,355 - Mycodo - INFO - All activated LCD controllers started
2017-01-19 00:46:05,356 - Mycodo - INFO - Mycodo v4.1.6 started in 0.932 seconds
2017-01-19 00:46:05,363 - Mycodo - INFO - 25.956 MB ram in use

kizniche, 2017-01-19:
It looks like I missed adding the crontab entry in the setup.sh script. I'm fixing that now. I'm also fixing that horrible statistics connection timeout error to be smaller and less obnoxious.

It appears this may all be due to gpiod not running. The error is at least indicating that and that would definitely prevent the sensor from being measured. I'll try to get a new release out after lunch to test.

isoflux, 2017-01-19:
@kizniche Thank you for your quick response.

kizniche, 2017-01-19:
I just released version 4.1.7. I found an issue with the upgrade system form the web interface. If you cannot upgrade, I made some notes in the changelog to instruct how to upgrade from the terminal (copied below).

This update fixes an error in the upgrade system, but it may mean that you cannot upgrade from the web interface (only for this version). If you cannot upgrade, execute the following command from the terminal to manually upgrade:

sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_mycodo_release.sh upgrade

kizniche, 2017-01-19:
Actually. I pulled that release while I'm working on a database issue. Sorry for th econfusion. It should be back soon.

kizniche, 2017-01-19:
v4.1.7 is live again, but there may be an issue with your database. I have no idea how this happened, but it's only affecting some of my test systems. If you get through the upgrade and see this message:

```
#### Upgrading database ####
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
ERROR [alembic.util.messaging] Requested revision cd014c0d61a0 overlaps with other requested revisions e4f984cd01d4
  FAILED: Requested revision cd014c0d61a0 overlaps with other requested revisions e4f984cd01d4
```

Then you are struck with this issue. You will need to open Mycodo/databases/mycodo.db with an SQL editor and go to the table 'alembic_version' and make sure there is only one entry. It's tricky to know what the value should be. e4f984cd01d4 is the most recent alembic_version. If you have a table named 'log', then you should be at cd014c0d61a0 or older.

Again, I'm really sorry this happened. It's probably easier just to delete your database and have Mycodo generate a new one if you're experiencing this database issue and don't know or want to deal with modifying it manually. Especially because it's almost impossible to know what version you're supposed to be at unless you look at the exact structure of the database itself. It's imperative that the correct version be there to enable proper upgrading.

berserktron3k, 2017-01-21:
Sorry to piggy back on to this, but my DHT22 isn't working at all after the 4.1.7 upgrade. From the command prompt, I can get it to read, so I know the sensor is still good. I tried deleting mycodo.db, and reran the install to recreate it, and it didn't change anything. I also checked if gpiod is in cron, it is, and it is running (the process shows in htop). Any ideas?

Drakthule, 2017-01-21:
welp, i reinstalled 4.1.7 and still the daemon dont want to start itself, dht22s are still giving readings of 0 as well. guess ill just sit tight till a newer version pops out to see if anything changes

kizniche, 2017-01-21:
@Drakthule, Thank you for commenting. You're the first person to provide feedback after I released v4.1.7, which was an attempt at fixing the issue. User feedback is crucial to resolving these issues, so if I don't hear anything, I have no idea if it worked or not.

Could you provide a little more information ?

  - What daemon are you referring to? gpoiod, mycodo?
  - Can you provide logs and any relevant errors?

kizniche, 2017-01-21:
I found one of my DHT22s! I got it hooked up and found the issue. I'm making a small release to fix the issues for everyone, in just a few minutes after I test some more.

kizniche, 2017-01-21:
Found another issue. Tracking it down before I make the release.

kizniche, 2017-01-21:
I just released v.4.1.8, but just a moment ago I realized I never pushed the key file for fixing the web UI upgrade system. So, you will most likely have to issue the following command in a terminal to perform an upgrade:

```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_mycodo_release.sh upgrade```

kizniche, 2017-01-21:
The latest release should fix the DHT22 issues, and fix a similar issue with the DHT11. My DHT22 is now working at least. If anyone that has had issues with theirs can test and report back when you can, it would be appreciated. Thanks.

isoflux, 2017-01-22:
Mycodo V.4.1.8 (clean installation)  fixed all the issues mentioned above.    I have tested with 2xDHT22, 3xRelays,  and PID (raise and lower relays).  It works right out of the box.

Here is  /var/log/mycodo/mycodo.log:
2017-01-22 03:49:44,715 - mycodo - INFO - Mycodo v4.1.8 starting
2017-01-22 03:49:44,930 - mycodo - INFO - Anonymous statistics enabled
2017-01-22 03:49:45,411 - mycodo.relay - INFO - Relays Initialized
2017-01-22 03:49:45,412 - mycodo.relay - INFO - Relay controller activated in 203.7 ms
2017-01-22 03:49:45,413 - mycodo - INFO - All activated timer controllers started
2017-01-22 03:49:45,414 - mycodo - INFO - All activated sensor controllers started
2017-01-22 03:49:45,415 - mycodo - INFO - All activated PID controllers started
2017-01-22 03:49:45,416 - mycodo - INFO - All activated LCD controllers started
2017-01-22 03:49:45,420 - mycodo - INFO - Mycodo v4.1.8 started in 0.705 seconds
2017-01-22 03:49:45,426 - mycodo - INFO - 25.392 MB ram in use
2017-01-22 05:49:24,926 - mycodo.sensor_4iFWqMY1 - INFO - Activated in 174.0 ms
2017-01-22 05:49:29,099 - mycodo.sensor_Fm2fuof0 - INFO - Activated in 166.0 ms
2017-01-22 06:10:48,614 - mycodo.pid_bfQD9e8x - INFO - Activated in 109.2 ms.

Thank you @kizniche .

isoflux, 2017-01-22:
um.. I found out later on that Mycodo daemon didn't start automatically every time after reboot.  I need to start it manually with; 

`sudo service mycodo restart`

After that, everything works just fine.

kizniche, 2017-01-22:
Is it starting now after a fresh install or is this a current issue?

isoflux, 2017-01-22:
Daemon started perfectly after fresh installation.  Once I rebooted the RPi, daemon wouldn't start.  Actually, I found this issue since I first tried Mycodo v.4.1.5 as I posted above.

kizniche, 2017-01-22:
You're actually the first person ever to say systemctl isn't automatically starting the daemon.

kizniche, 2017-01-22:
Now, if the daemon doesn't properly shut down, there's a lock file in /var/lock/mycodo.pid (or something similar), that if it doesn't get deleted, the daemon can't start up again until it's deleted (to prevent more than one instance of the daemon).

isoflux, 2017-01-22:
Thanks @kizniche  for prompt reply as always.  I did test, both hard and normal shutdown.  At this time, the daemon started perfectly both cases.  Um.. I will keep an eye on this issue.  I will report back if I found the issue again.
Cheers,

kizniche, 2017-01-22:
I think that sometimes the computer shuts down faster than the daemon can and the lock file stays. But, the /var/lock directory I believe is temporary (ram), so it should be wiped every reboot, unless you have a different configuration (I did at one time and it caused problems periodically).

Drakthule, 2017-01-28:
4.1.9 is running without a hitch, thank you for all your effort kyle

Myke974, 2017-01-28:
@Drathule : thank you for the return after update : this decided me to hit
the *update *button ;)

@Kyle : good job mate :)

closed 2017-01-29 by kizniche