kizniche/Mycodo #175: RPi CoreVoltage/clock frequency display - feature request

MurderLabz, 2017-01-21:
Do others think it would be useful to display core voltage or clock frequency in the WebUI?

			  S Y S T E M    V O L T A G E S  

	The Core voltage is:		1.3125V
	The sdram Core voltage is:	1.2000V
	The sdram I/O voltage is:	1.2000V
	The sdram PHY voltage is:	1.2250V

			  C L O C K    F R E Q U E N C I E S  

     	arm   frequency(45)=900000000 Hz    pwm	  frequency(25)=0 Hz
     	core  frequency(1)=250000000 Hz	    emmc  frequency(47)=250000000 Hz
     	h264  frequency(28)=250000000 Hz    pixel frequency(29)=25200000 Hz
     	isp   frequency(42)=250000000 Hz    vec	  frequency(10)=0 Hz
     	v3d   frequency(43)=250000000 Hz    hdmi  frequency(9)=163682000 Hz
     	uart  frequency(22)=47999000 Hz	    dpi	  frequency(4)=0 Hz

`# --- FREQ: Display current clock frequencies
FREQ () {
   echo -e "\n\t\t\t`tput smso`  C L O C K    F R E Q U E N C I E S  `tput rmso`\n"
   for src in arm core h264 isp v3d uart pwm emmc pixel vec hdmi dpi ;do
      echo -e "\t$src\t$(vcgencmd measure_clock $src) Hz"
   done | pr --indent=5 -r -t -2 -e3
   echo
}

# --- Main Procedure

# Diagnose the Bourne shell which is not supported
if [ `ps | tail -n 4 | sed -E '2,$d;s/.* (.*)/\1/'` = "sh" ]; then
   echo
   echo "*** Oops, the Bourne shell is not supported"
   echo
   exit 86
fi

# Make sure that `bc` is installed
flagbc=`which bc`
if [ -z $flagbc ]; then
   echo Installing bc # {small}
   sudo apt-get -y install bc
fi
# Display voltages
echo -e "\n\t\t\t`tput smso`  S Y S T E M    V O L T A G E S  `tput rmso`"
echo -e "\n\tThe Core voltage is:\t\t\c"
/opt/vc/bin/vcgencmd measure_volts core|awk -F "=" '{print $2}'
echo -e "\tThe sdram Core voltage is:\t\c"
/opt/vc/bin/vcgencmd measure_volts sdram_c|awk -F "=" '{print $2}'
echo -e "\tThe sdram I/O voltage is:\t\c"
/opt/vc/bin/vcgencmd measure_volts sdram_i|awk -F "=" '{print $2}'
echo -e "\tThe sdram PHY voltage is:\t\c"
/opt/vc/bin/vcgencmd measure_volts sdram_p|awk -F "=" '{print $2 "\n"}'

# Display frequencies
FREQ
exit`

kizniche, 2017-01-21:
I never really had a use for this information. What would one theoretically use it for?

kizniche, 2017-01-21:
Also, what is 'bc'? I see it installed in the script but I don't see it used.

MurderLabz, 2017-01-21:
bc(1)                                                                 General Commands Manual                                                                 bc(1)

NAME
       bc - An arbitrary precision calculator language

SYNTAX
       bc [ -hlwsqv ] [long-options] [  file ... ]

DESCRIPTION
       bc  is  a  language  that supports arbitrary precision numbers with interactive execution of statements.  There are some similarities in the syntax to the C
       programming language.  A standard math library is available by command line option.  If requested, the math library is defined before processing any  files.
       bc  starts  by  processing  code  from all the files listed on the command line in the order listed.  After all files have been processed, bc reads from the
       standard input.  All code is executed as it is read.  (If a file contains a command to halt the processor, bc will never read from the standard input.)

       This version of bc contains several extensions beyond traditional bc implementations and the POSIX draft standard.  Command line  options  can  cause  these
       extensions to print a warning or to be rejected.  This document describes the language accepted by this processor.  Extensions will be identified as such.

kizniche, 2017-01-21:
right, but where is it used in the script? I don't see it.

MurderLabz, 2017-01-21:
What I am really after is watching voltage usage. As sensors an Relays are added when to know you would need to start using another PS - (BC) I hacked that out of another script
[full.txt](https://github.com/kizniche/Mycodo/files/721434/full.txt)
 let me post the org script

kizniche, 2017-01-21:
This is the line that uses it from full.txt, line 48:

```TEMPB4OVER=$(echo "${OVRTMP:-70}-${TEMPf}"|bc -l)                               # Calculate the number of degrees before over-temp condition```

Seems you took that out and bc is no longer a requirement.

MurderLabz, 2017-01-21:
I think for 'TEMPB4OVER=$(echo "${OVRTMP:-70}-${TEMPf}"|bc -l)  '

MurderLabz, 2017-01-21:
Sorry, I should of yanked that out

kizniche, 2017-01-21:
Just making sure I'm not missing something

kizniche, 2017-01-21:
Would the core voltage change? wouldn't the Pi stop operating if there were an under-voltage situation?

MurderLabz, 2017-01-21:
Ah.. you are correct.. "The core voltage is regulated and won't go down until the battery resembles a pancake" https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=166404
I guess this would be of no use..

MurderLabz, 2017-01-21:
Lets close and let me think this more thoroughly...

closed 2017-01-21 by MurderLabz

kizniche, 2017-01-21:
Yeah, if it's a niche area that very few people are likely to use, I'm hesitant to add it when those people can just execute their own code. But, if it helps with diagnosing issues or something that's directly related to Mycodo/Pi, and it's more than a niche market, I'm all for adding. Thanks for the suggestion.