kizniche/Mycodo #176: DHT 11 no reading at WEB UI

reymart081678, 2017-01-22:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.8
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved

   - Observed that after 4.1.8 upgrade the sensor reading of DHT11 is not working properly.
      Using manual test shows correct reading. ( test_gpio_DHTx.py )
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
Daemon.log
 2017-01-22 09:13:24,783 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-22 09:13:24,786 - mycodo.sensor-LQsoApah - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-22 09:13:33,087 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-22 09:13:33,090 - mycodo.sensor-znxp2rK5 - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-22 09:13:41,786 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-22 09:13:41,805 - mycodo.sensor-LQsoApah - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-22 09:13:48,137 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-22 09:13:48,141 - mycodo.sensor-znxp2rK5 - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-22 09:13:58,887 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-22 09:13:58,890 - mycodo.sensor-LQsoApah - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-22 09:14:03,159 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-22 09:14:03,162 - mycodo.sensor-znxp2rK5 - ERROR - Could not read sensor. Ensure it's connected properly and detected.

Manual Test:
pi@mushroom:~/Mycodo/mycodo/tests/manual_tests $ sudo python ./test_gpio_DHTx.py -g 27 DHT11
Temperature: 18.0
Humidity: 81.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 79.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 80.0
Temperature: 18.0
Humidity: 80.0
pi@mushroom:~/Mycodo/mycodo/tests/manual_tests $ date
Sun Jan 22 09:13:30 SGT 2017

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-01-22:
What version did you upgrade from?

Was the DHT11 sensor working in the previous version?

Can you check the upgrade log for errors: /var/log/mycodo/mycodoupgrade.log

reymart081678, 2017-01-22:
Hi Kyle,
    Mycodo 4.1.7.  I executed the manual  upgrade.

  Found existing installation: decorator 4.0.10
    Uninstalling decorator-4.0.10:
      Successfully uninstalled decorator-4.0.10
Successfully installed alembic-0.8.10 argparse-1.4.0 decorator-4.0.11
#### Upgrading database ####
INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.
INFO  [alembic.runtime.migration] Running upgrade cd014c0d61a0 -> e4f984cd01d4, Remove log from displayorder and remove log table
#### Compiling Translations ####
compiling catalog /home/pi/Mycodo/mycodo/mycodo_flask/translations/es/LC_MESSAGES/messages.po to /home/pi/Mycodo/mycodo/mycodo_flask/translations/es/LC_MESSAGES/messages.mo
compiling catalog /home/pi/Mycodo/mycodo/mycodo_flask/translations/fr/LC_MESSAGES/messages.po to /home/pi/Mycodo/mycodo/mycodo_flask/translations/fr/LC_MESSAGES/messages.mo
#### Removing statistics file ####
rm: cannot remove '/home/pi/Mycodo/databases/statistics.csv': No such file or directory
#### Updating crontab entries ####
Removing crontab entry ...
Appending to crontab:
-----------------------------------------------
@reboot /usr/local/bin/pigpiod &-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh mycodo --remove

#### Setting permissions ####
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
#### Reloading systemctl, Mycodo daemon, and apache2 ####
Restarting apache2 (via systemctl): apache2.service.
Done.
Upgrade completed successfully without errors.
sh: 1: /home/pi/Mycodo/mycodo/../mycodo/scripts/bin/bash: not found

reymart081678, 2017-01-22:
Sorry click the wrong button

kizniche, 2017-01-22:
The upgrade log you just pasted was from your previous upgrade, not the most recent. The line:

```sh: 1: /home/pi/Mycodo/mycodo/../mycodo/scripts/bin/bash: not found```

Was the error from when you clicked upgrade from the web interface. Your manual upgrade command does not append the log output to the upgrade file. Do you happen to have the terminal still open from which you performed the latest upgrade in order to copy the latest upgrade status output?

reymart081678, 2017-01-22:
Hi kyle,
   I do not have the copy of the upgrade last night.  Done some additional test.
-  Disabled/deactivated DHT 11 sensor ( GPIO 27)
-   Activate DHT22 sensor and use GPIO 27 pin.
-  Both Manual and Web shows the same reading.
- Seems like DHT 11 code is not working.

Please see below screen shot:
[image]

kizniche, 2017-01-22:
Can you run the following code with this command ```sudo python test_script.py```:

```python
# coding=utf-8
import os
import sys
import time
import pigpio

class DHT11Sensor:
    def __init__(self, gpio, power=None):
        self.pi = pigpio.pi()
        self.gpio = gpio
        self.power = power
        self.high_tick = 0
        self.bit = 40
        self.either_edge_cb = None
        self._dew_point = 0.0
        self._humidity = 0.0
        self._temperature = 0.0

    def next(self):
        if self.read():  # raised an error
            raise StopIteration  # required
        return dict(humidity=float('{0:.2f}'.format(self._humidity)),
                    temperature=float('{0:.2f}'.format(self._temperature)))

    @property
    def humidity(self):
        if not self._humidity:  # update if needed
            self.read()
        return self._humidity

    @property
    def temperature(self):
        if not self._temperature:  # update if needed
            self.read()
        return self._temperature

    def get_measurement(self):
        self._humidity = 0.0
        self._temperature = 0.0
        try:
            if self.power is not None:
                print("Turning on sensor at GPIO {pin}...".format(
                    pin=self.gpio))
                self.pi.write(self.power, 1)  # Switch sensor on.
                time.sleep(2)
            try:
                self.setup()
            except:
                print(
                    'DHT11 could not initialize. Check if gpiod is running.')
            self.pi.write(self.gpio, pigpio.LOW)
            time.sleep(0.017)  # 17 ms
            self.pi.set_mode(self.gpio, pigpio.INPUT)
            self.pi.set_watchdog(self.gpio, 200)
            time.sleep(0.2)
        except Exception as e:
            print("{cls} raised an exception when taking a reading: "
                  "{err}".format(cls=type(self).__name__, err=e))
        finally:
            self.close()

    def read(self):
        try:
            self.get_measurement()
            # self_humidity and self._temperature are set in self._edge_rise()
            if self._humidity != 0 or self._temperature != 0:
                return  # success - no errors
            print("{cls}: Could not acquire a measurement".format(
                cls=type(self).__name__))
        except Exception as e:
            print("{cls} raised an exception when taking a reading: "
                  "{err}".format(cls=type(self).__name__, err=e))
        return 1

    def setup(self):
        self.pi.set_pull_up_down(self.gpio, pigpio.PUD_OFF)
        self.pi.set_watchdog(self.gpio, 0)
        self.register_callbacks()

    def register_callbacks(self):
        self.either_edge_cb = self.pi.callback(self.gpio,
                                               pigpio.EITHER_EDGE,
                                               self.either_edge_callback)

    def either_edge_callback(self, gpio, level, tick):
        level_handlers = {
            pigpio.FALLING_EDGE: self._edge_fall,
            pigpio.RISING_EDGE: self._edge_rise,
            pigpio.EITHER_EDGE: self._edge_either
        }
        handler = level_handlers[level]
        diff = pigpio.tickDiff(self.high_tick, tick)
        handler(tick, diff)

    def _edge_rise(self, tick, diff):
        val = 0
        if diff >= 50:
            val = 1
        if diff >= 200:  # Bad bit?
            self.checksum = 256  # Force bad checksum
        if self.bit >= 40:  # Message complete
            self.bit = 40
        elif self.bit >= 32:  # In checksum byte
            self.checksum = (self.checksum << 1) + val
            if self.bit == 39:
                # 40th bit received
                self.pi.set_watchdog(self.gpio, 0)
                total = self._humidity + self._temperature
                # is checksum ok ?
                if not (total & 255) == self.checksum:
                    raise Exception
        elif 16 <= self.bit < 24:  # in temperature byte
            self._temperature = (self._temperature << 1) + val
        elif 0 <= self.bit < 8:  # in humidity byte
            self._humidity = (self._humidity << 1) + val
        self.bit += 1

    def _edge_fall(self, tick, diff):
        self.high_tick = tick
        if diff <= 250000:
            return
        self.bit = -2
        self.checksum = 0
        self._temperature = 0
        self._humidity = 0

    def _edge_either(self, tick, diff):
        self.pi.set_watchdog(self.gpio, 0)

    def close(self):
        self.pi.set_watchdog(self.gpio, 0)
        if self.either_edge_cb:
            self.either_edge_cb.cancel()
            self.either_edge_cb = None

if __name__ == '__main__':
    if not os.geteuid() == 0:
        sys.exit("Script must be executed as root")

    pin = 27

    dht = DHT11Sensor(pin)
    if dht.read():
        print("could not get a reading")
    else:
        print("Temp: {temp}, Hum: {hum}".format(
            temp=dht.temperature, hum=dht.humidity))
```

kizniche, 2017-01-22:
Quick question, because it begs to be asked... Have you done a complete power down and startup? I mean shutdown and plug removed from the wall and reinserted 15 seconds later? Often the DHT sensors need to have power physically unplugged for them to begin operating properly.

kizniche, 2017-01-22:
Also, if someone reading this has a DHT11 and can verify if theirs is working or not in v.4.1.8 would be helpful.

The difference between dht11.py from [v.4.1.7 and v4.1.8](https://github.com/kizniche/Mycodo/compare/v4.1.7...v4.1.8) is minimal and should not be causing an issue.

reymart081678, 2017-01-22:
i just use the restart from the web.
Please see result of test_script.py
pi@mushroom:~/Mycodo/mycodo/tests/manual_tests $ sudo python test_script.py
Temp: 17, Hum: 95
pi@mushroom:~/Mycodo/mycodo/tests/manual_tests $

I tweak the dht11.py to adjust the bit comparison.
   Instead of 34 from the original to 33   and 16 to 15
The reading is okay now.

  0..7  8..15  16..23 14..32 33..40
        elif self.bit >= **33**:  # In checksum byte
            self.checksum = (self.checksum << 1) + val
            if self.bit == 39:
                # 40th bit received
                self.pi.set_watchdog(self.gpio, 0)
                total = self._humidity + self._temperature
                # is checksum ok ?
                if not (total & 255) == self.checksum:
                    raise Exception
        elif **15** <= self.bit < 24:  # in temperature byte
            self._temperature = (self._temperature << 1) + val
 
The readings are now the same.
[image]

kizniche, 2017-01-22:
Just to be clear, you ran my unedited test script and the measurements were successful?

reymart081678, 2017-01-22:
Hi Kyle,
    Yes. just copied the test script  you sent and works okay.

kizniche, 2017-01-22:
Okay, because that doesn't make any sense, because it is the exact code (removed docstrings, and minor changes to run from one script) from v4.1.8. *scratches head*

kizniche, 2017-01-22:
I edited the test script, above. See if it will still run correctly.

reymart081678, 2017-01-22:
I think the dht11.py code bit comparison is the culprit. Maybe it is from the last version 4.1.5 and just observed it today when you added some sanity check at 4.1.8 ( no readings after 4.1.8 upgrade).

[image]

kizniche, 2017-01-22:
It's not. I'm not sure where you got that solution, but The test script is unedited and it works. I want to stick to the original code before making these types of modifications to it.

MurderLabz, 2017-01-22:
Both my DHT11 and DHT22 are working with 4.1.8

kizniche, 2017-01-22:
@MurderLabz, thanks!

@reymart081678, I'm suspecting something weird is going on with your system or sensor(s). Can you install Mycodo again after a fresh install of Raspbian as a test?

kizniche, 2017-01-22:
Also, have you tried changing the pin your sensor is connected to?

reymart081678, 2017-01-22:
okay kyle. I'll just monitor the response of the adjusted dht11.py code.

reymart081678, 2017-01-22:
I'll pull out a new sd card to do fresh re install

MurderLabz, 2017-01-22:
Mycodo is dependent on Adafruit_Python_DHT, correct? 

pi@testpi:~/Adafruit_Python_DHT $ cd build/
pi@testpi:~/Adafruit_Python_DHT/build $ ls
bdist.linux-armv7l  lib.linux-armv7l-2.7  temp.linux-armv7l-2.7
pi@testpi:~/Adafruit_Python_DHT/build $

kizniche, 2017-01-22:
Nope. Only a few apt packages and what's in install/requirements.txt

kizniche, 2017-01-22:
@reymart081678, What is your sensor read period (seconds) set to?

reymart081678, 2017-01-22:
Hi Kyle,
    17 seconds for DHT 11 Low, DHT 11 reference is 15 seconds.

kizniche, 2017-01-22:
Okay. There's a minimum limit that the readings are too close to each other and can cause issues, but I *think* 17 should be fine. It's at least longer than the default of 15 seconds.

reymart081678, 2017-01-22:
I make sure they are not the same. I'll try to play with the timing. 
[image]

reymart081678, 2017-01-23:
Hi Kyle,
   Just an update. I did the fresh install of mycodo 4.1.8.  and activated only 1 DHT11 sensor and relays.
But got two issue.

1. The sensor reading is okay (12 midnight) for several hours and stop getting data pass 12 nn.
    
[image]

Log Info of mycodo daemon

2017-01-23 20:17:19,295 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 20:17:37,361 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 20:17:37,367 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 20:17:38,309 - mycodo.pid_R6PLGjjZ - WARNING - No data returned from influxdb
2017-01-23 20:17:55,376 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 20:17:55,379 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 20:17:55,868 - mycodo - INFO - Received command to terminate daemon
2017-01-23 20:17:56,939 - mycodo - INFO - All LCD controllers stopped
2017-01-23 20:17:56,992 - mycodo.sensor_0DBhmnkU - INFO - Deactivated in 54.5 ms
2017-01-23 20:17:57,019 - mycodo.sensor_IX0A1YVS - INFO - Deactivated in 82.1 ms
2017-01-23 20:17:59,257 - mycodo.timer_KDmtnKTf - INFO - Deactivated in 2318.4 ms
2017-01-23 20:17:59,246 - mycodo.timer_JmNrDeCR - INFO - Deactivated in 2307.4 ms
2017-01-23 20:17:59,287 - mycodo.timer_1x2wrpBl - INFO - Deactivated in 2347.6 ms
2017-01-23 20:17:59,308 - mycodo.pid_R6PLGjjZ - INFO - Deactivated in 3265.8 ms
2017-01-23 20:17:59,328 - mycodo.timer_GyoLDwo2 - INFO - Deactivated in 2389.2 ms
2017-01-23 20:17:59,335 - mycodo - INFO - All Timer controllers stopped
2017-01-23 20:17:59,338 - mycodo - INFO - All Sensor controllers stopped
2017-01-23 20:17:59,342 - mycodo - INFO - All PID controllers stopped
2017-01-23 20:17:59,368 - mycodo - INFO - Mycodo terminated in 3.500 seconds

2. Using admin to restart the raspberrypi, but the daemon did not start similar to issue #174.
    -- Please see below log info for further debug.
        --  Before i issue service start , i captured the contents of /var/lock.
              - there is not mycodo.pid file present.

After Restart:

pi@raspberryMushroom:/var/lock $ ls -lrta
total 4
drwxr-xr-x  2 root     root  40 Jan 23 20:18 subsys
drwxrwxr-x  2 root     root  40 Jan 23 20:18 lockdev
-rw-r--r--  1 root     root  11 Jan 23 20:18 asound.state.lock
drwxr-xr-x 20 root     root 700 Jan 23 20:18 ..
drwxrwxrwt  5 root     root 120 Jan 23 20:18 .
drwxr-xr-x  2 www-data root  40 Jan 23 20:18 apache2
pi@raspberryMushroom:/var/lock $ ls -lrta apache2
total 0
drwxrwxrwt 5 root     root 120 Jan 23 20:18 ..
drwxr-xr-x 2 www-data root  40 Jan 23 20:18 .
pi@raspberryMushroom:/var/lock $ ls -lrta lockdev
total 0
drwxrwxr-x 2 root root  40 Jan 23 20:18 .
drwxrwxrwt 5 root root 120 Jan 23 20:18 ..
pi@raspberryMushroom:/var/lock $ ls -lrta subsys
total 0
drwxr-xr-x 2 root root  40 Jan 23 20:18 .
drwxrwxrwt 5 root root 120 Jan 23 20:18 ..
pi@raspberryMushroom:/var/lock $ 

After start mycodo using service:

pi@raspberryMushroom:/var $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  1.4G   13G  11% /
devtmpfs        182M     0  182M   0% /dev
tmpfs           186M     0  186M   0% /dev/shm
tmpfs           186M  4.4M  182M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           186M     0  186M   0% /sys/fs/cgroup
/dev/mmcblk0p1   63M   21M   43M  34% /boot
pi@raspberryMushroom:/var $ sudo service mycodo restart
pi@raspberryMushroom:/var $ ls -lrta /var/lock
lrwxrwxrwx 1 root root 9 Nov 26 01:19 /var/lock -> /run/lock
pi@raspberryMushroom:/var $ cd /var/lock
pi@raspberryMushroom:/var/lock $ ls -lrta
total 8
drwxr-xr-x  2 root     root  40 Jan 23 20:18 subsys
drwxrwxr-x  2 root     root  40 Jan 23 20:18 lockdev
-rw-r--r--  1 root     root  11 Jan 23 20:18 asound.state.lock
drwxr-xr-x 20 root     root 700 Jan 23 20:18 ..
drwxr-xr-x  2 www-data root  40 Jan 23 20:18 apache2
drwxrwxrwt  5 root     root 140 Jan 23 20:29 .
-rw-r--r--  1 root     root   3 Jan 23 20:29 mycodo.pid
pi@raspberryMushroom:/var/lock $

kizniche, 2017-01-23:
@reymart081678,

1. You seem to have not included the relevant log portion around 12:00 when the sensor stopped receiving data.

2. /var/lock is a symlink of /run/lock which is tempfs, meaning all contents are destroyed upon reboot. It is normal for no lockfile to be present upon reboot and no server running.

```systemctl list-unit-files | grep enabled``` will list all enabled services.

kizniche, 2017-01-23:
Are you testing with your sensor(s) on a breadboard with very short wires to the Pi?

reymart081678, 2017-01-23:
Hi Kyle,
    Please see below log information around 12 nn. when it stopped.

2017-01-22 15:46:28,778 - mycodo - INFO - Mycodo v4.1.8 starting
2017-01-22 15:46:29,975 - mycodo - INFO - Anonymous statistics enabled
2017-01-22 15:46:33,756 - mycodo.relay - INFO - Relays Initialized
2017-01-22 15:46:33,767 - mycodo - INFO - All activated timer controllers started
2017-01-22 15:46:33,776 - mycodo.relay - INFO - Relay controller activated in 1286.0 ms
2017-01-22 15:46:33,773 - mycodo - INFO - All activated sensor controllers started
2017-01-22 15:46:33,786 - mycodo - INFO - All activated PID controllers started
2017-01-22 15:46:33,791 - mycodo - INFO - All activated LCD controllers started
2017-01-22 15:46:33,818 - mycodo - INFO - Mycodo v4.1.8 started in 5.039 seconds
2017-01-22 15:46:33,865 - mycodo - INFO - 25.792 MB ram in use
2017-01-22 15:52:35,427 - mycodo.sensor_0DBhmnkU - INFO - Activated in 766.2 ms
2017-01-22 15:53:38,438 - mycodo.sensor_IX0A1YVS - INFO - Activated in 733.4 ms
2017-01-22 16:02:45,016 - mycodo.timer_JmNrDeCR - INFO - Activated in 175.6 ms
2017-01-22 16:02:58,632 - mycodo.timer_W2YfDvh0 - INFO - Activated in 229.6 ms
2017-01-22 16:11:38,413 - mycodo.pid_R6PLGjjZ - INFO - Activated in 1117.4 ms
2017-01-22 16:16:39,517 - mycodo.relay - WARNING - Relay yLKIJNpN (HumidifierFa) is already on.
2017-01-22 16:25:52,609 - mycodo.timer_W2YfDvh0 - INFO - Deactivated in 688.4 ms
2017-01-22 16:26:10,915 - mycodo.timer_JmNrDeCR - INFO - Deactivated in 680.7 ms
2017-01-22 16:27:42,615 - mycodo.timer_JmNrDeCR - INFO - Activated in 599.7 ms
2017-01-22 16:31:21,775 - mycodo.timer_1x2wrpBl - INFO - Activated in 236.0 ms
2017-01-23 00:33:56,823 - mycodo.timer_GyoLDwo2 - INFO - Activated in 227.3 ms
2017-01-23 00:36:56,733 - mycodo.timer_KDmtnKTf - INFO - Activated in 751.9 ms
2017-01-23 12:12:40,231 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 12:12:40,306 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 12:12:58,221 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 12:12:58,229 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 12:13:16,261 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 12:13:16,264 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 12:13:34,301 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 12:13:34,303 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 12:13:38,295 - mycodo.pid_R6PLGjjZ - WARNING - No data returned from influxdb
2017-01-23 12:13:52,353 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-23 12:13:52,360 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 12:14:08,428 - mycodo.pid_R6PLGjjZ - WARNING - No data returned from influxdb

kizniche, 2017-01-23:
That log essentially says nothing.. No indication of what the issue is.

kizniche, 2017-01-23:
Does deactivating/activating the sensor resolve the issue?

reymart081678, 2017-01-23:
Hi Kyle,
    No it is not mounted on a breadboard. The wire less than 1 meter. I added filter capacitor (0.1uF) to vcc = 3.3v and pull up resistor 10Kohms.

kizniche, 2017-01-23:
Capacitor to VCC and _what_?

reymart081678, 2017-01-23:
After activating it it runs for 1 hour and started to error out again. VCC of DHT 11.

kizniche, 2017-01-23:
A capacitor has two leads. If you connected one to VCC, what is the other connected to?

Is this your only DHT11 sensor on the Pi?

kizniche, 2017-01-23:
Also, that still doesn't answer "Does deactivating/activating the sensor resolve the issue?"

reymart081678, 2017-01-23:
What i did for experimentation is to use the DHT22.py code and modified to strip off the decimal readings and use only the integer part for humidity and temperature.

Please see below response of the sensor DHT11 using DHT22, but i would still monitor it further.

The Time 22:45  is when i reset the pi using web.
   - DHT11.py is used.

Time 23:15 it started to get no reading
    
Time 23:30 i deactivated the sensor and create a copy of DHT22.py as DHT11.py
    - change the code to return only the integer value
        total = self.hH + self.hL + self.tH + self.tL
                if (total & 255) == self.CS:  # Is checksum ok?
                    self._humidity = (self.hH << 1) + val
                    if self.tH & 128:  # Negative temperature.
                        mult = -0.1
                        self.tH &= 127
                    else:
                        mult = 0.1
                    self._temperature = (self.tH << 1) + val

[image]

kizniche, 2017-01-23:
Looks like the code is malfunctioning if the humidity maxes out at ~88 % like it appears in your graph.

kizniche, 2017-01-23:
It also looks like you removed the mult factor in the temperature, so it won't display negative values properly.

kizniche, 2017-01-23:
Also, if you can format code as code (three backtics at start end end), it makes it a lot easier to read.

reymart081678, 2017-01-23:
Yes. it is the maximum of DHT11. around 90%. As so far i only got 1 error around 1 AM but still good until now.
  Yes i remove the negative part since DHT11 also does not have negative reading.

Okay noted.

2017-01-23 22:07:26,332 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.
2017-01-23 22:07:31,552 - mycodo.pid_R6PLGjjZ - WARNING - No data returned from influxdb
2017-01-23 22:07:33,504 - mycodo.pid_R6PLGjjZ - INFO - Deactivated in 1164.6 ms
2017-01-23 22:07:34,240 - mycodo.sensor_IX0A1YVS - INFO - Deactivated in 65.1 ms
2017-01-23 22:20:24,107 - mycodo.sensor_IX0A1YVS - INFO - Activated in 945.2 ms
2017-01-23 22:45:01,599 - mycodo.pid_R6PLGjjZ - INFO - Activated in 909.5 ms
2017-01-24 01:00:03,301 - mycodo.sensors.dht11 - ERROR - DHT11Sensor: Could not acquire a measurement
2017-01-24 01:00:03,307 - mycodo.sensor_IX0A1YVS - ERROR - Could not read sensor. Ensure it's connected properly and detected.

kizniche, 2017-01-23:
Okay, so it appears deactivating and activating clears the issue. Can you confirm this? I can only assume form inference form the log.

reymart081678, 2017-01-23:
Yes. based on the response of the sensor.

kizniche, 2017-01-23:
Why does the temperature max out at 21C?

reymart081678, 2017-01-23:
Not sure. But i can try to activate another DHT11 which mounted near the raspberrypi 2 panel controller.

kizniche, 2017-01-23:
These are just observations that indicate your code edits have unintended consequences.

kizniche, 2017-01-23:
Here's a potential solution:

Copy these variables to the beginning of setup():

```
        self.high_tick = 0
        self.bit = 40
        self.either_edge_cb = None
```

and set them all to None in __init__.

reymart081678, 2017-01-23:
Thanks kyle. I'll edit them now.

kizniche, 2017-01-23:
If deactivating/activating resolves the issue, then there's an initialization bug somewhere, some objects mat not be reset properly when setup() is called, so moving the only objects that change from /_/_init/_/_ to setup() is the only logical solution I can think of.

reymart081678, 2017-01-23:
Hi Kyle,
    Just to confirm. 

```    def setup(self):
        """
        Clears the internal gpio pull-up/down resistor.
        Kills any watchdogs.
        Setup callbacks
        """
        self.high_tick = 0
        self.bit = 40
        self.either_edge_cb = None
        self.pi.set_pull_up_down(self.gpio, pigpio.PUD_OFF)
        self.pi.set_watchdog(self.gpio, 0)  # Kill any watchdogs.
        self.register_callbacks()
```
```        # Power cycle if timeout > MAX_TIMEOUTS
        self.no_response = 0
        self.MAX_NO_RESPONSE = 2

        self.tov = None
        self.high_tick = None
        self.bit = None
        self.either_edge_cb = None
```

kizniche, 2017-01-23:
Correct. And setup() should be:

```python
    def setup(self):
        """
        Clears the internal gpio pull-up/down resistor.
        Kills any watchdogs.
        Setup callbacks
        """
        self.high_tick = 0
        self.bit = 40
        self.either_edge_cb = None
        self.pi.set_pull_up_down(self.gpio, pigpio.PUD_OFF)
        self.pi.set_watchdog(self.gpio, 0)
        self.register_callbacks()
```

kizniche, 2017-01-23:
You are using the DHT22 code. I am always going to be referring to the original DHT11 code unless I specify otherwise.

reymart081678, 2017-01-23:
Hi Kyle,
   Yes i am using the DHT22 code. I reactivated the sensor again to monitor it using the latest changes.

kizniche, 2017-01-23:
Make sure you're using the latest release code with my above changes. Otherwise, any effects won't be relevant.

reymart081678, 2017-01-23:
Hi Kyle, 
   Yes. I recompiled the latest change before activating the sensor.
```
-rwxrwxr-x 1 pi     pi     10815 Jan 24 04:33 dht11.py
-rw-r--r-- 1 root   root   10034 Jan 24 04:33 dht11.pyc
drwxrwxr-x 2 mycodo mycodo  4096 Jan 24 04:33 .
```

kizniche, 2017-01-23:
Those variables we changed are something I never noticed before as being a potential issue. Now that I recognize deactivating/activating resolves the issue, it makes sense that could be what prevents the sensor module from resuming operation after an error.

kizniche, 2017-01-23:
Potentially the same fix for the DHT22, too.

reymart081678, 2017-01-23:
Hi Kyle,
    Thank you for your great help.  Hopefully it resolves the issue.

kizniche, 2017-01-24:
You're welcome! I'm curious to see if this worked :)

[149 more replies]

closed 2017-01-31 by kizniche