kizniche/Mycodo #177: Controlling home environment with gas/propane-fired and electric (30 kW) Furnaces

JL1946, 2017-01-23:
First, I would like to commend you on your work to date...

I'm sure that you are aware of the following, but for clarity before posing my question...
For either furnace, the timing is typically as follows:
**Gas/Propane fired furnace**:
     1. Thermostat calls for heat (Switches 24 Vac to the gas/propane fired Furnace control)
     2. Furnace control engages system fan and verifies fan operation
     3. Furnace Control turns on gas/propane valve and ignites the gas at the burner
     4. Thermostat stops calling for heat (24 Vac switched off)
     5. Furnace Control turns off gas to the burners and runs the fan until the heat exchanger has cooled.

**Electric Furnace**:
     1. Thermostat calls for heat (Switches 24 Vac to the Sequencers that apply power to the System Fan and coils)
     2. First Sequencer in the electric furnace turns on system fan
     3 First Sequencer also powers downstream Sequencers that connect power to the heating coils (built-in delay in each sequencer applies power in a staggered fashion)
     4. System has a bi-metallic protective device in line with each coil to open the circuit in the event of over-heating
     5. Thermostat stops calling for heat (24 Vac to sequencers switched off)
     6. Sequencers turn off the power to each coil in the reverse order (last turned on is first turned off) - System Fan is the last to be turned off

For both scenarios, the gas-fired furnace control cannot be programmed to apply a duty-cycle to the gas valve and the hysteresis of the electric furnace sequencers preclude any sort of effective duty-cycling.
As a side note, the electric-furnace sequencers could be replaced with relay contactors or solid-state switches and controlled by a PID, but I don't believe that the same can be done for gas-fired furnaces.

Given the above scenarios, what would be your suggestion or thoughts as to designing a PID or series of PIDs to maintain the set-point given that the relays cannot directly control the gas valve in the gas/propane fired furnace or the Sequencer timing in the Electric Furnace? 

thank you for your response...

kizniche, 2017-01-24:
I'm having a difficult time trying to think of a way to apply a PID to these systems.

What initially comes to mind is modeling the mycodo control after the thermostat. This would simply be two sensor conditionals, one that executes a script when the temperature is below a certain temperature, and another that executes a script when above another temperature. These scripts will turn the heater on (or let it remain on) when the temperature is too low, and when the temperature rises above a certain point, the second script will turn off (or leave off) the heater. The lower temperature conditional would be a few degrees below the upper conditional, to ensure there is a time for the environment to cool down before the heater turns on again.

I'm sure there are more complex ways to do this, perhaps by using the actual PID controller, but I don't have a good recommendation at the moment.

matatasb, 2017-01-24:
Hello JL1946,
The PID controller will give you an analog signal that will be proportional to the action that you have to apply to the sistems. There are 2 ways of using this signal:

- One way is generating a PWM signal where the time it's ON is (PID_OUTPUT/MAX_OUTPUT_PID)*PERIOD. This signal will be a digital value so you can command like you said a relay. This system is ussually used in electric furnaces (I recommend you to use a solid-state relay, a mechanical relay is not prepared to be turned on and off so often like you need in this case).
-The other way it's to use the analog signal as if. In the case of a gas/propane furnace you will have to use a  proportional valve. This kind of valve uses an analog input (0-10V or 4-20mA) to control the flow of gas, the y work like a servo-motor attached to a manual valve. 

If you already have the gas/propane furnace and It's not prepared whit the second way of control, I recommend you to use simpler mode ON/OFF with hysteresis. It's a very common way to control temperature and it will be more easy to implement than a PID controller. 

Hope this help you and that you could understand my English.

closed 2017-10-24 by kizniche