kizniche/Mycodo #178: apache 500 error when camera stream is stopped

gremlintomsk, 2017-01-23:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.8
 
#### Problem Description
Please list:
  Typical install on raspberry pi 2 with camera module. All requirements are installed
### Errors
Apache error: http://paste.ubuntu.com/23852128/

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. go to  tools\camera
 2. click start stream
 3. click stop stream

kizniche, 2017-01-23:
Thanks for the bug report. I fixed the issue and it will be included in the next release.

kizniche, 2017-01-23:
This should be fixed. Please reopen it if it's still an issue after the release is made and you are able to test the new version.

gremlintomsk, 2017-01-29:
After 4.1.9 release same problem - apache log http://paste.ubuntu.com/23886284/

kizniche, 2017-01-29:
```
[Sun Jan 29 18:01:27.265699 2017] [wsgi:error] [pid 712:tid 1945572400] 2017-01-29 18:01:27,264 Exception raised in '/camera' endpoint: max() arg is an empty sequence
[Sun Jan 29 18:01:32.240710 2017] [wsgi:error] [pid 712:tid 1962349616] [remote 78.139.196.195:30070] mod_wsgi (pid=712): Exception occurred processing WSGI script '/var/www/mycodo/mycodo_flask.wsgi'.
[Sun Jan 29 18:01:32.249738 2017] [wsgi:error] [pid 712:tid 1962349616] [remote 78.139.196.195:30070] IOError: failed to write data
[Sun Jan 29 18:01:32.418071 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786] [2017-01-29 18:01:32,399] ERROR in app: Exception on /camera [POST]
[Sun Jan 29 18:01:32.419387 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786] Traceback (most recent call last):
[Sun Jan 29 18:01:32.420153 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1982, in wsgi_app
[Sun Jan 29 18:01:32.420848 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]     response = self.full_dispatch_request()
[Sun Jan 29 18:01:32.421418 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
[Sun Jan 29 18:01:32.421957 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]     rv = self.handle_user_exception(e)
[Sun Jan 29 18:01:32.422463 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1517, in handle_user_exception
[Sun Jan 29 18:01:32.422965 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]     reraise(exc_type, exc_value, tb)
[Sun Jan 29 18:01:32.423458 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1612, in full_dispatch_request
[Sun Jan 29 18:01:32.424021 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]     rv = self.dispatch_request()
[Sun Jan 29 18:01:32.424531 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]   File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1598, in dispatch_request
[Sun Jan 29 18:01:32.425056 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]     return self.view_functions[rule.endpoint](**req.view_args)
[Sun Jan 29 18:01:32.425561 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 186, in page_camera
[Sun Jan 29 18:01:32.426331 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786]     CameraStream().terminate()
[Sun Jan 29 18:01:32.426929 2017] [wsgi:error] [pid 712:tid 1928795184] [remote 78.139.196.195:28786] TypeError: 'bool' object is not callable
```

kizniche, 2017-01-29:
This error indicates you are not running the code I pushed to fix the issue (i.e. you are running the same code as you were doing to produce the original error, hence why you would receive the same error). I said in my comment that you should upgrade your system after I make the release, then test the code again. I am fairly sure the new code will work.

gremlintomsk, 2017-01-30:
today i made fresh mycodo install from man:
`sudo apt-get install jq
cd ~
curl -s https://api.github.com/repos/kizniche/Mycodo/releases/latest | \
jq --raw-output '.tarball_url' | wget -i - -O mycodo-latest.tar.gz
mkdir Mycodo
tar xzf mycodo-latest.tar.gz -C Mycodo --strip-components=1
rm -f mycodo-latest.tar.gz
cd Mycodo/install
sudo /bin/bash ./setup.sh`
 and get same error.

kizniche, 2017-01-30:
My mistake. You are correct. I fixed the issue in the CameraStream class but never changed the function call in page_routes.py. I'll make that edit now and it will be incorporated in the next release.

I'm working on a bunch of bug fixes right now, so I may push another minor update release tonight.

kizniche, 2017-01-30:
The new release is out. I'm closing this issue. Please open it again if the issue persists.

closed 2017-01-30 by kizniche