kizniche/Mycodo #179: Desactivate relay dont work

Ericktz, 2017-01-24:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.8
 
#### Problem Description
Please list:
put OFF one relay, after to turn ON
 
### Errors
no error found in log files.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. add one relay and put in off
 2. add conditional relay, save and activate
 3. put ON relay manually
 4. try to put OFF manually
 
### Additional Notes
the problem ocurr, when is activate the conditional relay and, the relay is OFF, i put it ON, but its impossible to put in OFF again, when press its, the relay stop it for 1 ms maybe, and turn ON again  in a bucle ... the only solution, is desactivate the conditional relay, and put OFF manually.
[image]

kizniche, 2017-01-25:
What are you trying to accomplish?

It seems you are using the relay conditional statements improperly. You should not have the same relay for If and Then.

kizniche, 2017-01-25:
What you have set up is working properly.

This is your conditional statement, written in a sentence: If RelayBomba turns off, turn RelayBomba on for 120 seconds.

So, when you turn if off, your conditional will trigger it to turn back on for 120 seconds.

What I think you may be actually wanting is for the relay to cycle with it turning on for 120 seconds and turning off for (looks like) 1300 seconds. If this is the case, relay conditionals are not a way to go about this. For that, you will want to create a timer, from the Timer menu. Then, select the third option, which you can specify an on duration and an off duration.

Ericktz, 2017-01-25:
Dear, I thank you for your clarification. I was actually wrong in using the conditional relays.  What I needed can be perfectly fulfilled with the timer. You can close the thread!  sorry for the inconvenience

kizniche, 2017-01-25:
No problem. This issue you started indicates I should probably put some description or short instructions on the Relay page.

closed 2017-01-25 by kizniche