kizniche/Mycodo #181: Add feature for chaining relays

lucasvickers, 2017-01-28:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.0.26
 
#### Problem Description
Please list:

  - I would like to chain relays as follows.  
    - I have a peltier cooling source combined with two heat sinks and fans that I use to dissipate cool air inside of my environment and warm air outside of the environment.  Once the peltier turns off, I still need to run the fans for X seconds to ensure that the excess heat/cold is expelled from the cooling fans.  I see that I could set up a conditional to trigger relay Y if relay Z has turned off, but that likely would cause an infinite loop since relay Z would turn off and stay off for some time.
   - I think it would be a good feature to add some type of chaining such as "When relay Z transitions from ON to OFF, then turn on relay Y for 60 seconds"

Please let me know your thoughts!

kizniche, 2017-01-29:
> When relay Z transitions from ON to OFF, then turn on relay Y for 60 seconds

This is currently possible with the relay conditionals. Perhaps I'm misunderstanding what you desire to do.

lucasvickers, 2017-01-30:
I could just be misunderstanding how things work.

if [Relay 1] [OFF]

would this only be triggered when [Relay 1] transitions from ON -> OFF?  I was interpreting it as "Whenever [Relay 1] is OFF", as in the event could occur over and over.

kizniche, 2017-01-30:
Ah, yes, it means that "when" off. I might change the working to be more clear, or add a description to the top of the page. I really need to add more documentation all over. I'm not even sure if I mention relay conditionals in detail in the manual (help button at top right).

lucasvickers, 2017-01-30:
You do document it!  (Documentation is always the hardest part of a project)

I think just a quick snippet saying that you will only check the conditionals when the state of a relay changes would clear things up.

thanks for the quick response!

kizniche, 2017-01-30:
I'm working on adding a short description on the relay page. It will show up in the next release. Thanks for submitting this issue.

kizniche, 2017-01-30:
The relay conditional description should appear when a relay conditional is added, in the latest version (4.1.10) I just released.

closed 2017-01-30 by kizniche

Myke974, 2017-02-01:
@lucasvickers When I saw your post, I thought : I'll return to that in a few day because I'll be faced the "same" problem.
I have an ultrasonic mister that need to work with a fan so :
when ultr. mister turns ON then its Fan turns ON too
but I needed its Fan to keep running 3 secondes after its extinction to expulse the mist left in the mister receptacle.
Here is how I did it:
[image]
If humidifier turn OFF then turn Fan ON for 3s
The Fan that is already ON by the first conditionnal will stay ON only for 3 seconds.

@kizniche Is this the right way to do it ?

kizniche, 2017-02-01:
If it works, it works. Try and see.