kizniche/Mycodo #182: Graph Doesn't work after upgrading to 4.1.9

berserktron3k, 2017-01-29:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.9
 
#### Problem Description
Please list: When clicking Graph -> General Graph, get:
Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
Internal Server Error

The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1, Click Graph
 2. step 2, click General Graph
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

Myke974, 2017-01-29:
I don't have this problems on 2 R.pi since the update and I'm using graphs
a lot since 24hours.
Perhaps you can try to :
1] clear cookies
2] another brownser

kizniche, 2017-01-30:
It appears you may be a special spot (not a good one) with regard to your config database (~/Mycodo/databases/mycodo.db). I added a rough check to see if your database is out of date and unable to be upgraded by the normal means, and display an error at the top of every page. If you don't see this error at the top of every other page (error warning, so if you have error warning notifications turned off in the general configuration, Admin->Configure, turn it on to see), then it may be you have a different database error and will need to regenerate your database.

To test, you can rename your database to something like mycodo.db-backup, then reboot. The database will be regenerated. See if you can go to the general graph page then. If so, then it was a database issue and you'll have to build your setting over again with the new database.

berserktron3k, 2017-01-30:
Renaming the database and rebooting only caused my whole Mycodo install to
crash. A quick reinstall, however, resolved the issue (had to set up
sensors, graphs, etc. again).

Brady Hummel

kizniche, 2017-01-30:
Sorry about that. I'm not sure how the database bug got introduced. But it *should* be smooth sailing from here on out!

closed 2017-01-30 by kizniche

berserktron3k, 2017-01-30:
No worries at all. As always, thanks for your great work on this!

Brady Hummel

kizniche, 2017-01-30:
I do still consider this experimental software ;)

kizniche, 2017-01-30:
...aka, how I learned Python programming.