kizniche/Mycodo #184: Make circuit diagrams for all sensors/devices

kizniche, 2017-01-31:
And make them easily accessible, in both the manual and on their respective configuration pages.

kizniche, 2017-01-31:
There should also be a directory containing the working files used to create the images/svg files, for easy editing by the community if updates are needed.

Theoi-Meteoroi, 2017-10-23:
Please let me know how I can help.

kizniche, 2017-10-23:
If you look at the [Diagrams](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#diagrams) section of the manual, I have a few examples. These files are stored in the [manual_images](https://github.com/kizniche/Mycodo/tree/master/manual_images) directory. I haven't tried to make this all-encompassing yet, but it's a good base to build off of. Most of the images were just harvested from the internet, but it would be nice for them all to be uniform in design, and cover at least all the complex circuits.

pukkita, 2017-10-28:
Will look for the diagram, pics, etc for the MOS Module and will document the PWM sensing wiring, voltage divider, etc.

Any advised electronic circuit design program?

Theoi-Meteoroi, 2017-11-16:
I created this on circuitlab.com -
[image]

Seemed the cheapest way to play.  I often assemble digital stuff without a schematic. But analog is just too easy to screw up. And I have some boards I sort of regret for not having a schematic. The debug took longer than was convenient.

kizniche, 2017-11-18:
This is great! I'll get it added to the manual soon. Thanks

pukkita, 2017-11-18:
What should we aim for when adjusting the trimpot?

Theoi-Meteoroi, 2017-12-13:
I have something new to add, the gpio I2C I suggested cannot use the standard mux dtoverlay to obtain bus devices in /dev. I think your mux code probably works, however I like to use the linux driver and have the buses in /dev, so all the channel switching happens in the driver.  A side effect is all devices on those channels show up on the scans you get on the status page with i2cdetect.

I created a github repo with a tested device tree overlay.   Feel free to point to it or even copy it into the manual.  I think it makes buffering those GPIO pins a lot easier and you get the scanned devices on the webpage status.  I tried to make the instructions easy and obvious. Please let me know if I can improve them.

[https://github.com/Theoi-Meteoroi/GPIO-pca9548](url)

My 4 wire fans arrived and I'll get back to validating that schematic.  The trimpot is to trim offset in the opamp output to adjust the range in output voltage.  I may change some of the component values after some testing. I started with what I had in my parts bins that were close but not what I calculated.

kizniche, 2017-12-13:
That's great! And it's nicer to see exactly what's connected without having to scan each channel. Thanks for that. I'll add it to the manual.

Theoi-Meteoroi, 2017-12-26:
After exploring options, it becomes clear that USB is the way to add serial sensors to a Pi. I've been playing with a SC16is750 for UART and well, it becomes a pit of wasted time to implement it seems.  You can even avoid messing around with the native serial setup. That leaves bluetooth with a proper UART on a Pi3. And USB just works. You can program some of the bridge devices (CP2102 types) to make them unique, if you have more than one. The configuration suggestion to make in the manual is to create a udev rule to make the device Mycodo sees consistent across reboots and detach, reattach events.  The only requirement is some unique USB attribute (serial number or whatever) to be a unique device if you use more than one of a type.  The clue is from [http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/](url)