kizniche/Mycodo #190: Request for tester with an LCD

kizniche, 2017-02-03:
I'm looking for someone with an LCD to test something for me.

The test is very simple:

1. Set one of your LCD lines to display a Relay Time
2. Make sure the relay has been turned on and off at least once for this to work.
3. Reply here with whether it displays correctly or gives an error on the LCD. Also include which Mycodo version you are running.

The help is much appreciated. Thanks.

jj2473, 2017-02-03:
Just sent you an email to your account, appears to work fine on 4.1.8

Thanks,
Jim

kizniche, 2017-02-03:
Thank you, @jj2473

Quick update to the test, I believe you will need to have Mycodo be near the latest version 4.1.15, because I believe this is affected by having the latest version of influxdb (1.2.0).

jj2473, 2017-02-03:
Ok give me a couple to power on Incubator, that is more current but was having issue with relay so I shut it down. Replacing with solid state this weekend.

kizniche, 2017-02-03:
I actually found the issue, so I won't need any more testing. I was able to get the error to be produced on my own LCDs, so I can test on my own. Thank you for the help.

closed 2017-02-03 by kizniche