kizniche/Mycodo #191: 'Live' Page returning no data. Delete ALL sensors to resolve (temporarily).

Ryobidelacroix, 2017-02-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.15
 
#### Problem Description
Please list:
 - All sensors not returning any data on 'Live' page. ALL sensors must be deleted and the re added one by one in order to read data. 

 
### Errors

Mycodo Daemon Log attached as .txt file
HTTP Log attached as a .txt file
[HTTP_Log.txt](https://github.com/kizniche/Mycodo/files/750339/HTTP_Log.txt)
[Mycodo_Log.txt](https://github.com/kizniche/Mycodo/files/750334/Mycodo_Log.txt)

### Steps to Reproduce the issue:
How can this issue be reproduced?
- This issue has only occurred if I have left my device powered on for extended period of time (>10 hrs)  without a reboot. Rebooting after sensor data loss does not resolve issue. 

 
### Additional Notes
This was a clean install of Mycodo (4.1.15) on a recently reformatted SD Card running Raspbian Jesse (full). Issue reproduced with a minimal system (i.e., one sensor, no relays, no LCD etc etc). 

I've been following this project for a while now. It's great to see how far this system has come. Keep up the great work! :)

kizniche, 2017-02-03:
What Pi version? I suspect Zero or 1

kizniche, 2017-02-03:
Here's my reasoning...

From the log, it looks like the mycodo daemon was not properly shutting down. There are no shutdown messages. Also, these errors are only appearing directly after starting up, and are typical of when influxdb hasn't yet started up, causing an expected ```ConnectionError```. Usually only the slower Pis have this issue, because they can't start influxdb fast enough at bootup to handle the incoming connections when the mycodo daemon starts up.

This doesn't indicate an issue with Mycodo. In fact, it's expected. Your issue is something else. I'm not sure what it is, but this is the first I've heard of it.

Ryobidelacroix, 2017-02-04:
It's a Pi 3B. 

I'll attempt another reinstall and see how it goes :)

kizniche, 2017-02-12:
Also, what sensors were you using?

closed 2017-02-21 by Ryobidelacroix