kizniche/Mycodo #192: Prevent refresh when turning relays on/off on Relay Page

kizniche, 2017-02-07:
Current behavior: When the "On" button is pressed for a relay on the Relay Page, the page refreshes and the relay turns on.

Desired behavior: When the "On" button is pressed for a relay on the Relay Page, the relay turns on without a page refresh.

I've attempted to do this multiple ways using Ajax/jQuery code I found, without success. I believe it's because I'm using wtforms that it's more complicated than it seems.

hideouspenguinboy, 2017-05-03:
I conversation that *might* help?

**hideouspenguinboy**-
Another randomly generated question: Can any of you help with this bug? https://github.com/kizniche/Mycodo/issues/192 (edited)
I know some of you are coders, though I don't know what languages you know.
This is the program I run my environmental box with and I'd like to help the author out.

**andrew** 
I know a bit about flask
I don’t think the issue is related to wtforms
its the way the route is configured
https://github.com/kizniche/Mycodo/blob/master/mycodo/mycodo_flask/page_routes.py#L683 (edited)
there needs to be a route for just the refresh, that doesn’t trigger a redirect. then you can call that route using jquery
if you wanna play around with it, try either dropping the redirect in that method (which may have other consequences) or create a new route that simply performs the refresh
and have the js call that instead of the page route

kizniche, 2017-05-07:
Issue #70 initially addressed this. More recently, I made an attempt with [Release 5.0.24](https://github.com/kizniche/Mycodo/releases/tag/v5.0.24) to implement this:

### Features

  - Add toastr and ajax support for submitting forms without refreshing the page (currently only used with relay On/Off/Duration buttons) ([#70](https://github.com/kizniche/mycodo/issues/70))

However, I quickly learned that due to the multiplicity of the forms, using wtforms, I was unable to resolve the issue and reverted the changes in [Release 5.0.27](https://github.com/kizniche/Mycodo/releases/tag/v5.0.27) (latest):

### Bugfixes

  - Revert to old relay form submission method (ajax method broken)

closed 2017-10-28 by kizniche