kizniche/Mycodo #193: Using a USB camera?

Ericktz, 2017-02-07:
Hi, well this is an only question or suggestion if is not possible. Can I use a USB camera instead of a Picamera???

kizniche, 2017-02-07:
It depends on what you would like to do with it. I'm open to suggestions. Do you have a specific use in mind?

Ericktz, 2017-02-07:
I would think for the moment only as a follow-up for an indoor crop, or as an outdoor security camera. Not everyone has a Pi camera, and the USB camera also it is more accessible and cheaper.

kizniche, 2017-02-07:
The nice thing about the Pi camera is it's guaranteed to work because the hardware is standardized. Mycodo uses the python module, [picamera](https://github.com/waveform80/picamera).

USB cameras are pretty variable in how to acquire images from them. I haven't really explored using them with Mycodo, but I would give it a try. If there's a universal python module that can detect/acquire from USB cameras, this would be rather easy to integrate into Mycodo. I'm not sure if this exists or not, I haven't really looked.

As of how, if you want to do time-lapses, the easiest way is to write a script and add it to cron.

From within Mycodo, you can trigger still image captures from relay/sensor/PID conditional statements by creating a script that captures an image, then use the 'execute command' option in the conditional. So, when your conditional is triggered, the script is executed and the image acquired.

I believe there is also some good software for the Pi that can turn it into a security camera, and I believe it allows the use of USB cameras. These software that allow USB cameras may also be a good place to look how they acquire images from the USB camera, for possible use within Mycodo.

kizniche, 2017-02-10:
I just tested my webcam with [opencv](https://github.com/skvark/opencv-python), and I was able to get an image right away! Very promising!

kizniche, 2017-02-10:
You can a new script I pushed, [test_camera_capture_still.py](https://github.com/kizniche/Mycodo/blob/44e15713fcd14fc75973f18c5eaaab2a03b71bd7/mycodo/tests/manual_tests/test_camera_capture_still.py), by running the following:

```
sudo apt-get install python-opencv
python test_camera_capture_still.py
```

An image 'camera_image_[timestamp].bmp' will be created if it successfully detects a camera.

If you have multiple cameras connected to your Pi, you can change the device by altering [Line 29](https://github.com/kizniche/Mycodo/blob/44e15713fcd14fc75973f18c5eaaab2a03b71bd7/mycodo/tests/manual_tests/test_camera_capture_still.py#L29) to a different integer greater than 0:

```python
 cap = cv2.VideoCapture(0)
```

Ericktz, 2017-02-10:
thank Kyle for trying this... it is work, an image was created with the name you wrote, but I give the following errror:

pi@raspi:~$ python test_camera_capture_still.py 
HIGHGUI ERROR: V4L2: Unable to get property <unknown property string>(0) - Invalid argument
('Test: ', -1.0)
HIGHGUI ERROR: V4L2: Unable to get property <unknown property string>(2) - Invalid argument
('Ratio: ', -1.0)
HIGHGUI ERROR: V4L2: Unable to get property <unknown property string>(5) - Invalid argument
('Frame Rate: ', -1.0)
('Height: ', 640.0)
('Width: ', 480.0)
('Brightness: ', 0.4577777683734894)
('Contrast: ', 0.5)
('Saturation: ', 0.41499999165534973)
HIGHGUI ERROR: V4L2: Unable to get property Hue(9963779) - Invalid argument
('Hue: ', 0.0)
HIGHGUI ERROR: V4L2: Unable to get property Gain(9963795) - Invalid argument
HIGHGUI ERROR: V4L: Gain control in V4L is not supported
('Gain: ', -1.0)
HIGHGUI ERROR: V4L2: Unable to get property Exposure(9963793) - Invalid argument
HIGHGUI ERROR: V4L: Exposure control in V4L is not supported
('Exposure: ', -1.0)

kizniche, 2017-02-12:
I've got a whole new camera system working in the dev-5.0 branch, with USB cameras working with opencv. When I release 5.0.0, you'll be able to test it out.

Myke974, 2017-02-12:
Excellent ! Can't wait ! :D

closed 2017-03-16 by kizniche