kizniche/Mycodo #194: SQL injection areas found

etiology, 2017-02-12:
## Mycodo Issue Report:
  - Specific Mycodo Version: Dev-5.0
 
#### Problem Description
Please list:
   [Issue Example](https://github.com/kizniche/Mycodo/blob/c3d9671d546816168de3bdd5f1e2069c4f0ce3fe/mycodo/mycodo_flask/general_routes.py#L268-L274)
   The pattern of taking arguments from GET/POST parameters and passing it directly into a SQL statement opens the app up to SQL injections.  Raw SQL statements should **never** be used.
  Flask SQLalchemy models will handle them correctly and also protect from sql injections but you have to use the sqlalchemy objects and NEVER raw sql.  

Raw SQL is hard to debug, often open to exploit, and too rigid to use.  SQLAlchemy models allow you to switch SQL backends and not care about what flavor of SQL you are using.  If you hard code in SQL statements then you are married to a particular sql database.  You can't switch without rewriting the statements.

etiology, 2017-02-12:
Wait I see that this is influxDB but it just looks like sql.

kizniche, 2017-02-12:
Good point. The ```sensor_measure``` can be compared to the keys of [this dictionary](https://github.com/kizniche/Mycodo/blob/a5921bac462356dccb978f0a9acf0f90ae586dd8/mycodo/config.py#L36-L49), the ```sensor_id``` can be queried to the sensor table for existence, and the sensor_period can be checked for being an integer. Does that resolve the issues with injection attacks?

kizniche, 2017-02-12:
Or is there a better way to do this that encapsulates other benefits beyond just attack protection?

etiology, 2017-02-12:
If this was SQL it wouldn't solve the issue.  However since it's Influx I have no idea.  I feel it's something that needs to be looked into because there is zero sanitation current on the inputs.  The lookup helps so I would make sure that ```sensor_measure``` is in the ```dictionary``` of acceptable things.

Ultimately encapsulating it would keep things DRY and be easier to maintain going forward.  If influx changes the api or you find a bug in a statement then you have to look all over the place to change that.  However wrapping these statements in a query method on the influxdb object makes it nice and easy to find and allows you to fix all of the statements in one place.

kizniche, 2017-02-12:
Aha! Yes, I can do that. Let me work on that. Thanks!

closed 2017-03-25 by etiology