kizniche/Mycodo #196: General graph showing 500 Internal Server Error

pewi81, 2017-02-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 4.1.16
 
#### Problem Description
Please list: 
When trying to open General Graph I get:
**500 Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.**

Asynchronous Graph works flawless
 
### Errors
500 Internal Server Error 

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Try to access General Graph 
 2. step 1
 3. etc
 
### Additional Notes
This happened after an udate but I cant remember from which version it was.
I tried a few tips from the installation page but nothing works.

Myke974, 2017-02-19:
Can you please show your
`/var/log/mycodo/mycodoupdate.log`

pewi81, 2017-02-19:
Sorted it out by deleting the database!

closed 2017-02-19 by pewi81

kizniche, 2017-02-21:
Good to hear! That would have been my suggestion.