kizniche/Mycodo #197: bme280?

Drakthule, 2017-02-27:
i was wondering i dont seem to have a humidity reading from my new bme280, is this normal?

Myke974, 2017-02-27:
Like this ?
[image]
check the blue line 🔵 
it seems I got the same illness !
I never saw a variation in its R.H. ! in weeks ! 😮

Drakthule, 2017-02-27:
wow i feel really stupid, i could never find it in the live page but low and behold it is listed under graphs thanks mike974

kizniche, 2017-02-27:
This is a known bug. I forgot to put the humidity line in the live page when I added the sensor to Mycodo. All parts of the system should recognize the measurement from the sensor, it's just not displaying on the live page. This will be fixed in the next version release.

Since you have this sensor, can you verify you are receiving normal humidity measurements?

I ask because @myke974 has a very unresponsive humidity reading.

Myke974, 2017-02-27:
😩 yes I have !

Drakthule, 2017-02-28:
I can verify, ive nothing to test it against for accuracy but it does seem to be working

kizniche, 2017-02-28:
So it is at least a little more variable than @myke974's graph image? His basically shows humidity flatlining (it's actually just changing **very** slowly).

Drakthule, 2017-02-28:
[image]
hmmmm im not so sure the humidity is always like that in my room, now im not so sure

kizniche, 2017-02-28:
Just cupping your hand over the sensor should raise it up near 100%. Humidity should swing easily within a range of 20 to 50 % in a day.

Drakthule, 2017-02-28:
ive been meaning to put the bme in my tank, let me crimp together a cable for it and ill stick it into the tank, ill know for sure if it works as the tank is much more humid to accomdate my drosera and nepenthes

Drakthule, 2017-02-28:
perhaps its to soon to know for sure but after making a cable and putting the sensor in the tank right next to the dht22 the temp readings are within a degree faranheit however the humidity is off by 10 degrees, the dht 22 is fluxuating as expected and the bme280 doesnt seem to fluxuate much if at all and is reading in the 77% humidity area same as it was outside of the tank.

kizniche, 2017-10-18:
I'm going through old issues and trying to get investigations going for things I think are fixable. Are you able to use any other code (such as from the manufacturer or online forums, preferably in Python) that give more responsive or more accurate readings? If so, and it's more accurate than the code currently in Mycodo, I can replace it with the working code.

Theoi-Meteoroi, 2017-10-21:
I'm using the BME280 with version 5.2.5.  I'm not having any issues with the readings or presentation in graphs.  This issue has had me puzzled as I looked for problems with the data, yet found none.

kizniche, 2017-10-24:
I believe this issue was due to a faulty sensor. Feel free to reopen or reply if there are any new developments. I don't have this sensor, so I cannot test myself.

closed 2017-10-24 by kizniche